roleoroleo/yi-hack-Allwinner-v2 #852: [y623] Unexpected motion recording behaviours on WIFI interruptions

AndreasBVB, 2024-04-05:
Hi thank you very much for making this project available to us. It has been great with testing so far but I am running into an issue when there is a WIFI interruption.

The camera:
- motion recording continues to work after WIFI disconnection
- motion recording stops working after WIFI is recovered from disconnection

The HA integration:
- after camera's WIFI recovery from disconnection, the motion sensor is stuck at "detected", even though the MQTT messages operate normally in the background (remains off when no motion as expected)

My unit is Yi Pro 2K Home YFUS 12.0.51 on y623. Kindly suggest.

AndreasBVB, 2024-04-05:
Just wondering: a quick notice/message on WIFI reconnection would be helpful enough because it can be captured by HA and trigger a camera reset through the curl command.

roleoroleo, 2024-04-07:
A question: if a new motion is detected, does the cam restore its normal behavior?
Or the sensor no longer works.

So the problem is that the motion stop is sent while the wifi is not connected.
Correct?

AndreasBVB, 2024-04-07:
When there are new motions after WIFI recovery:
- MQTT receives new motion messages
- HA motion binary sensor is still stuck at "detected" and never turns off
- camera recording not working anymore upon new motions

roleoroleo, 2024-04-07:
Please, check with a different mqtt clien (e.g. MQTT Explorer) if  a new message is sent to the broker when a motion is detected.

AndreasBVB, 2024-04-07:
I can confirm the MQTT messages are normal after WIFI recovery, even visible within HA-MQTT integration
[image]

AndreasBVB, 2024-04-07:
Or do you want me to check if HA MQTT messages are fake? They sync up well with my motions so probably not.

roleoroleo, 2024-04-07:
I think that the messages in ha are ok.
The cam should recover properly after a new motion detection.
About recording I don't know how to fix the problem because this feature is handled by yi processes.
Could you check if the process mp4record is running?

AndreasBVB, 2024-04-07:
The mp4record process is apparently there according to the `top` command but looks to me like a zombie process after WIFI recovery. The normal CPU consumption is about 2.5%+ depending on recording or not but after WIFI recovery its CPU usage is constantly 0.4% or 0.2%.

Any possibility to reboot the camera after WIFI recovery or send some MQTT signals like last WIFI connect timestamp for HA reboot camera automation?

AndreasBVB, 2024-04-07:
Need to correct myself:
Under normal conditions mp4record also consumes under 0.5% when not recording, and above 2.5% when recording.

After WIFI recovery mp4record CPU usage is constantly under 0.5% regardless of the motion sensor status, so probably still alive but not triggered.

roleoroleo, 2024-04-07:
This is weird.
If mqtt messages are sent, the system (rmm process) detects the motion and send the ipc messages.
But mp4record don't handle this message.
You could try (just a try) to restart mp4record after a wifi disconnection.

> Any possibility to reboot the camera after WIFI recovery or send some MQTT signals like last WIFI connect timestamp for HA reboot camera automation?

If the problem is mp4record, you can call a web service and restart it.
Or you can reboot the cam.
There are no mqtt messages for this, at the moment. Just web services.

AndreasBVB, 2024-04-07:
So what is the recommended method to restart mp4record?

Also I am looking for the signal of reconnected WIFI hoping for an automatic reset/reboot after the interruption. Manually I can certainly do that but I may not be always checking if recording is working.

roleoroleo, 2024-04-08:
> So what is the recommended method to restart mp4record?

Check the wiki:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#service-start-and-stop

AndreasBVB, 2024-04-10:
Thanks. After WIFI recovery, when the recording is dead, `cgi-bin/service.sh?name=mp4record&action=status` shows mp4record is "started".

I tried:
`cgi-bin/service.sh?name=mp4record&action=stop`
`cgi-bin/service.sh?name=mp4record&action=start`

mp4record still shows as "started" but the recording is not restored. I have to reset the camera to bring it back to normal.

roleoroleo, 2024-04-10:
I think it's a bug of the yi firmware.
You can check it removing temporarily the sdcard and testing the system without the hack.

AndreasBVB, 2024-04-10:
Just tried with an older 1080p model with the same WIFI recovery interruption: factory system/firmware has no problem after WIFI reconnect and motion recording keeps working.

roleoroleo, 2024-04-10:
This could confirm that the problem is related to the yi firmware.
When you can access to the cam with the app, check if a yi firmware update is available.

AndreasBVB, 2024-04-10:
Cam survived WIFI disconnect/recovery with factory firmware, but not with hack. Maybe still something with the hack?
Firmware is up-to-date in Yi app. I'll create some HA ping logic. Thanks :)

roleoroleo, 2024-04-10:
So, please, let me summarize:
- y623 doesn't work when the hack is running
- y623 works when the hack is not running
- y291ga works in both situations

Is it correct?

AndreasBVB, 2024-04-10:
y291ga the same as y623:
- doesn't work when the hack is running
- works when the hack is not running

roleoroleo, 2024-04-10:
Wow, it's a terrible problem to debug.
I will try to understand the reason.

AndreasBVB, 2024-04-10:
Thanks a lot for the work 🙏

roleoroleo, 2024-04-10:
How do you simulate the wifi disconnection?
Is there a method to replicate the problem.
I tried to disconnect the cam from the router web page.
After 10 seconds the cam reconnected but the motion continues to work.
Any suggestion?

AndreasBVB, 2024-04-10:
I simply rebooted my router from router management. That's my most common situation for WIFI interruption.

AndreasBVB, 2024-07-11:
Hi can we keep this issue open? It is still there and frequent PING to many cameras overloads router quite a bit. Thanks!

closed 2024-11-24 by github-actions[bot]