roleoroleo/yi-hack-Allwinner-v2 #853: Recording sometimes stops, rmm and mqttv4 killed

Dotatay, 2024-04-05:
Sometimes the camera stops recording videos. Then the camera needs to be restarted and everything works well again.

After the error and a reboot, I ran the command "ps". These processes were no longer present during the error:
1276 root     42436 S    ./rmm
1724 root       880 S    mqttv4

My Camera Dome Guard r30gb YRS 9.0.05
yi-hack-Allwinner-v2 0.3.1

roleoroleo, 2024-04-07:
It's a known problem.
The cam runs out of memory and the kernel kills some processes.
Did you enable swap file?
Disable all unneeded features.

Dotatay, 2024-04-07:
Were there any major changes affecting resource consumption between versions 0.2.4 and 0.3.1?
With version 0.2.4 I had about 20000 KB of memory free. With version 0.3.1 it’s only 12000 KB.

Swap: 15
MQTT is disabled

Active features:
Disable Cloud
Recording without Cloud
RTSP low aac
Enable/disable time OSD
NTPD
HTTPD

As an automatic solution to keep the camera active (reboot) after an error, I’m currently using this script and the cron entry:
```
#!/bin/sh

if ! ps | grep "./rmm" | grep -qv grep; then
    reboot
fi
```

```
*/5 * * * * /tmp/sd/yi-hack/bin/checkrmm.sh
```

roleoroleo, 2024-04-07:
> Were there any major changes affecting resource consumption between versions 0.2.4 and 0.3.1?

Too many changes. It's not simple to know what change is responsible for this memory consumption.

About restart, wd_rtsp.sh already contains a check to restart the system when rmm is not running.

Dotatay, 2024-04-08:
Thank you very much for your help.

During some testing I had disabled the RTSP server and was already wondering... 
Even without my extra script, the camera restarts automatically after crashes.

closed 2024-04-08 by Dotatay