roleoroleo/yi-hack-Allwinner-v2 #854: Yi Dome Camera U 2K has stopped working

robotm-team, 2024-04-07:
I uploaded the hack two days ago and it worked without any problem.
Today I tried to connect it to the HomeAssistant server as rtsp (with the WebRTC plugin) but unfortunately it didn't work.
I enabled the onvif server in the camera and after restarting the camera no longer wants to upload the yi-hack.
It behaves as if there is no sd card. During the reboot you hear the announcement : "welcome to ..." and then "waiting for connection" and nothing further happens.
I tried this solution : https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/821 but it doesn't do anything.
Below is the log from the card (I set DEBUG_LOG=yes) so you can see that there is an attempt to boot, but why unsuccessful?

Starting system.sh
Make /etc writable
Activating swap file
Set swappiness to 15
Configuring cloudAPI
Starting yi processes

  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  454 root         0 SW   [irq/302-sunxi-m]
  456 root         0 SW   [irq/166-sdc0 cd]
  457 root         0 SW   [irq/303-sunxi-m]
  461 root         0 SW   [kworker/0:2]
  514 root         0 SW<  [bioset]
  515 root         0 SW   [mmcqd/0]
  534 root         0 SW<  [kworker/0:1H]
  547 root         0 SWN  [jffs2_gcd_mtd4]
  579 root       980 S    /usr/sbin/telnetd
  582 root         0 SW   [kworker/0:3]
  590 root         0 SW   [kworker/u2:2]
  591 root         0 SW   [ksdioirqd/mmc1]
  598 root         0 SW   [RTW_XMIT_THREAD]
  599 root         0 SW   [RTW_CMD_THREAD]
  600 root         0 SW   [RTWHALXT]
  610 root      1260 S    ./dispatch
  614 root       988 S    sh /tmp/sd/yi-hack/script/system.sh
 1222 root         0 Z    [init]
 1249 root       980 R    ps

              total        used        free      shared  buff/cache   available
Mem:          60904       10668         412          64       49824       49148
[... 133 lines cut ...]
  610 root      1260 S    ./dispatch
  614 root      1000 S    sh /tmp/sd/yi-hack/script/system.sh
 1271 root     19736 R    ./rmm
 1325 root       768 S    ./mp4record
 1326 root       952 S    ./cloud
 1327 root      1680 S    ./p2p_tnp
 1328 root      2344 S    ./oss
 1329 root      7584 S    ./rtmp
 1330 root       740 S    ./watch_process
 1351 root       964 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1362 root       844 S    pure-ftpd (SERVER)
 1376 root       892 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sb
 1379 root       880 S    mqttv4
 1407 root       964 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/u
 1432 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 1467 root       676 S    ipc2file
 1475 root       792 S    onvif_notify_server --conf_file /tmp/onvif_simple_se
 1532 root       964 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1750 root         0 Z    [init]
 1783 root       980 S    sleep 1
 1835 root       980 R    ps

              total        used        free      shared  buff/cache   available
Mem:          60904       22380        1420         188       37104       37312
Swap:         65532           0       65532

roleoroleo, 2024-04-07:
The log is ok, the cam started properly.
If you hear the welcome message, it means that the cam didn't save properly wifi credentials.
Please, try to pair again the cam.

robotm-team, 2024-04-07:
Thank you for your reply!
What does "Try pairing the camera again." mean. ?
Initially, right out of the box I simply inserted the sd card with yi-hack (h51ga_0.3.1) and it worked.
I did not pair it with any application etc. The first boot was already done with your "firmware".
Sorry, but I have no experience with such cameras.
This one is my first ...

roleoroleo, 2024-04-08:
How did you set wifi credentials? Did you use configure_wifi procedure?
Try again, please.

> PS. Additional question:
> Will the camera work properly ONLY in lan network WITHOUT internet access ?

There is no final answer to this question.
It depends on how internet is filtered. Other users have noticed problems using the camera behind a firewall.
For example, the cam could not save mp4 records.
It depends...

robotm-team, 2024-04-08:
> How did you set wifi credentials? Did you use configure_wifi procedure?
> > Try again, please.

That's exactly what I did. I started with point 4 in the instructions
I have tried many times.
I have downloaded the file from scratch, changed SD cards. 
I have even already created an additional wifi with a simple name and password like 123aBc890.
Nothing changes: the camera starts and after a long while I hear these announcements. The camera does not connect to the wifi. I also tried resetting it with that button near the SD card slot. No results.

robbinjanssen, 2024-04-08:
I think this is related to #814, I've had similar problems with this camera. For some reason it has now been working for the past few days.

What I did:
1. Use the yi home app to setup the camera
2. insert SD card with the hack
3. after a few reboots the camera asked me to connect to wifi again
4. I did, reformatted the SD card a few times
5. Steps 3 and 4 have been repeated multiple times.
6. Now it has been working for a while, can't really tell why.

robotm-team, 2024-04-11:
Well, I did some tests today. The results are strange to say the least ...

I tested it on the version "yi-hack-Allwinner-v2 0.3.2"

The camera after restart announces that "welcome to ..."

Then it says that it is waiting for a connection. 

This suggests that the hack has not loaded (in my opinion) and the original firmware is running.

And now a strange thing:

I start the YiHome app and go through several steps until the QR code is generated.

The camera reads the code, announces that it has connected and ... yi-hack starts up.

It completely skips the "Factory.done\configure_wifi.cfg" (or "Factory\configure_wifi.cfg") entry - it doesn't retrieve the wifi network information from it.

I am completely dumbfounded. The first time , when it worked the camera "said nothing". - it booted up without any announcement etc.

And now, every reboot I have to show it the QR code with the network info.

I must be doing something wrong....

roleoroleo, 2024-04-12:
> This suggests that the hack has not loaded (in my opinion) and the original firmware is running.

This is not true.
The hack doesn't overwrite Yi feature and when the main Yi process detects a missing wifi configuration, it starts with the wizard.

> It completely skips the "Factory.done\configure_wifi.cfg" (or "Factory\configure_wifi.cfg") entry - it doesn't retrieve the wifi network information from it.

This configure_wifi.cfg is used only the 1st time, when you install the hack.
As you can see in the code, if the cam is already hacked, the script doesn't run:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/sdhack/h51ga/Factory/config.sh

The weird thing is that the cam doesn't save the credentials when the wizard completes.
I remember a bug in the Yi firmware but I don't remember what model. Could you check if a Yi firmware update is available?

So, finally you can try these steps:
- check a fw update for the cam through the app and re-pair the cam
- try this method https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)

If you want to debug the problem, make a dump of the partition:
`hexdump -C -n 156 /dev/mtdblock7`
You should read your ssid starting at offset 0x1c and your password at 0x5c.

robotm-team, 2024-04-12:
Well.
I made this hexdump.
It's weird (or more accurately, I don't know what I'm looking at...) because it's a conglomeration of the 2 different ssid and passwords I used for configuration:

I starred some characters for principle :-) :

00000000  01 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010  00 00 00 00 01 00 00 00  01 00 00 00 54 65 53 74  |............TeSt|
00000020  59 00 61 50 69 53 21 21  21 00 00 00 00 00 00 00  |Y.aPiS!!!.......|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000050  00 00 00 00 00 00 00 00  00 00 00 00 31 32 33 61  |............123a|
00000060  73 64 38 39 30 00 34 73  71 42 6d 23 47 40 00 00  |sd890.4sqBm#G@..|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

From "configure_wifi.cfg"
= Target network =
wifi_ssid=NigdyNaPiS!!!
wifi_psk=*****=%&M"4sqBm#G@

=Temporary network=
wifi_ssid=TeStY
wifi_psk=123asd890

I haven't tried anything else yet.

roleoroleo, 2024-04-12:
Is this after app pairing or after configure_wifi.sh?

robotm-team, 2024-04-12:
I haven't done anything since yesterday.
As described: 
I start the camera > "welcome to ..." > "waiting for connection" > I scan the code from YiHome > yi-hack works.
Since yesterday, after several reboots and attempts, the camera works after executing the sequence above.
And in this state (today) I read this mtdblock7.

PS. I don't know if this makes a difference, but through the YiHome app I gave it the SSID "TeStY". And in the yi-hack interface I only changed to this target SSID.

robbinjanssen, 2024-04-17:
Same for me, after a while it just started working 🤷🏻 No more "waiting for connection" etc.

robotm-team, 2024-04-28:
Ok. I had some time and played around.
1- Every time I restart the camera I have to show it the qr code with the wifi network data. even though the entry in /dev/mtdblock7 is correct.
2- What you suggested : "https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)" causes the camera to get up and connect without any problem but after 10 seconds (approximately) it reboots and so on and so forth .... After deleting this entry (startup.sh) see point 1 ...
3- Camera on/off - shutdown works fine. The dome closes etc. however, when switched on, the camera does not return to its previous position (even different) and the dome does not open.
4- I cannot control the PTZ in any way via MQTT - it does not respond ...

In summary - I have NO idea what is wrong with it ....

Translated with DeepL.com (free version)

roleoroleo, 2024-05-03:
> It's weird (or more accurately, I don't know what I'm looking at...) because it's a conglomeration of the 2 different ssid and passwords I used for configuration:

ssid and password are ok, the char "00" is the string terminator. So your current data are:
wifi_ssid=TeStY
wifi_psk=123asd890

> Ok. I had some time and played around. 1- Every time I restart the camera I have to show it the qr code with the wifi network data. even though the entry in /dev/mtdblock7 is correct. 2- What you [...]

I will resume my h51ga and test it.

roleoroleo, 2024-05-10:
I tested my y51ga 9.0.21.08_202007270941 and I was able to complete the procedure.
The cam was previously reset with the button.

1 - I formatted a new sd card and copied the last version of the hack (h51ga_0.3.2.tgz).
2 - I started the cam and hacked it, but the cam speaks and want to connect to my wifi network.
3 - After 2 minutes, I switched off the cam and I removed the sd card.
4 - I added the files from sdwifi.tar.gz archive and I configured configure_wifi.cfg according to my access point credentials.
5 - I restarted the cam.
6 - The cam booted and started the wifi wizard "Waiting connection" ...
7 - The cam restarted again after 1 minute.
8 - The cam connected to my ap.
9 - I logged in with putty and I removed startup.sh and configure_wifi.* from /tmp/sd/yi-hack

I tried to reboot the cam 3-4 times and it works correctly.

What's your Yi fw version?
I recommend you check your power supply: adapter and cable.

robotm-team, 2024-07-11:
Hello again. I've had a big break due to an overload of work but I'm back for help.
I bought a new SD card but everything looked the same as before.
 I changed the power supply and cable. Still nothing new...
Now I bought a new camera. Same model. But before I load something into it I will make sure:
Do I have to connect it to the app first or not?
Do I do the first start already inserted sd card with the hack uploaded or "in the blank"? 
Do I follow the procedure in the wiki or rather from the message above? 
I'll check over the weekend if it's my fault or if I had some weird/broken unit.
Greetings.
And thank you very much for your patience.

roleoroleo, 2024-07-13:
> Do I have to connect it to the app first or not?

It's not necessary but it's recommended.

> Do I do the first start already inserted sd card with the hack uploaded or "in the blank"?

Try to complete the wifi wizard without using the sd card.
When the procedure is completed and the cam is stable, reboot with the card and the hack.

closed 2024-11-11 by github-actions[bot]