roleoroleo/yi-hack-Allwinner-v2 #855: Yi Home 1080p - RFUS 9.0.19* - y21ga - Camera not start after first reboot

bulletyhasor, 2024-04-08:
Hi all! I'm new over here. I bought some new Yi cameras 1080p (exact model described bellow). I installed yi-hack 0.3.1 and I think I found a bug anywhere that can be bothering more than one people. 

My steps:
- Firstly, I installed the camera/s withing the Yi app, just in case I pretend to use it. All OK 
- Later, I proceed with the installation of latest yi-hack release (0.3.1) and all went just fine. Insert the SD and camera works just perfect. Easy and quickly.
- I play around a bit with the configuration and start experiencing some issues. I have 2 cameras, and without any logic, if I apply the changes rebooting the camera with the reboot button in the yi-cam web sometimes the camera do it ok and others it just reboot and do nothing. Instead, if I reboot unplugging the camera of the power its worst, allways do nothing. **Actually, not nothing but the camera blink as a normal start and turn off the led (as I configured). So I start thinking I'm not in a reboot loop**, it just not connects to wifi. I sniffed the network and no signals of the camera, no ping, no binding to wifi, etc..

I don't understand very well the hole process of wifi connection and SSID process and so, but **after some research about it I notice that my camera loose some part of the SSID of the wifi. This behavior is because a "_" character in the SSID.** 
I dont understand why sometimes the camera reboot ok with the web button but in any case, **I suppose there is any problem with the configure_wifi.sh that writes the SSID to the mtdblock7.** I don't really know how to fix it in code.

Easy solution: **(For me as workaround) I created a new wifi SSID with the stored one and the camera connects. Bingo!**

**Some outputs about the issue:**
- My SSID: **H-B-535_X**.  (I know it could be easy ;-) )
- Cam1 data stored: **Notice** the SSID stored is missing the final "X".
```shell
root@Yi_1:~# cat /tmp/wpa_supplicant.conf
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="H-B-535_"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="blablabla"
}

root@Yi_1:~# cat /dev/mtdblock7
H-B-535_blablablaaRzXE5u6A9WcC7cxlslprTaHunmEzSvm                                '      02544d534131364714242ee0d900e801
...
..
.
``` 
- Cam2 data stored: **Notice** the SSID stored in wpa_supplicant differs from mtdblock7
```shell
root@Yi_2:~# cat /tmp/wpa_supplicant.conf
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="H-B-535_X"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="blablabla"
}

root@Yi_2:~# cat /dev/mtdblock7
H-B-535_blablablaQNSqmcrh0HtJSpwxYS2EwtwuZtHVZ6C                                '      035344534c3332478028e984cc00eb5f
...
..
.
``` 

More info: When its happends, after reboot, I cant reset the camera with the button at the back and reconnect the camera with the yi-app and all works again until I reboot and it happends again.

**Cameras info:**
Model: Yi camera Home 1080p
RFUS: 
FW: 9.0.19.12_202102241808
Yi hack version: 0.3.1

bulletyhasor, 2024-04-11:
Hi, @roleoroleo 

**sure this look fairly the same case I suffered.** The only difference I see is that I don't event try to install the Yi-hack without linking the Yi app (using the wifi settings in Yi-hack only). 
Can it be hacked and later on connected with the reset button to the Yi app?. I've two more new cameras to add, I can try it, I didn't have time to play with it yet.

Today both cameras have the same output even in the wpa_supplicant.conf, I think I made some changes via the camera web site and restart, but I don't remember very well. So I just put one camera output.

Here the hexdump output :

```
shell
root@Yi_1:~# cat /tmp/wpa_supplicant.conf 
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="H-B-535_"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="blablabla"
}

root@Yi_1:~# hexdump -C /dev/mtdblock7 
00000000  01 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010  00 00 00 00 01 00 00 00  00 00 00 00 48 2d 42 2d  |............H-B-|
00000020  35 33 35 5f 00 00 00 00  00 00 00 00 00 00 00 00  |535_............|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000050  00 00 00 00 00 00 00 00  00 00 00 00 30 6c 34 66  |............blab|
00000060  XX XX XX XX XX XX XX XX  00 00 00 00 00 00 00 00  |labla........|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000090  00 00 00 00 00 00 00 00  00 00 00 00 61 52 7a 58  |............aRzX|
000000a0  45 35 75 36 41 39 57 63  43 37 63 78 6c 73 6c 70  |E5u6A9WcC7cxlslp|
000000b0  72 54 61 48 75 6e 6d 45  7a 53 76 6d 00 00 00 00  |rTaHunmEzSvm....|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000d0  00 00 00 00 00 00 00 00  00 00 00 00 12 27 00 00  |.............'..|
000000e0  00 00 00 00 30 32 35 34  34 64 35 33 34 31 33 31  |....02544d534131|
000000f0  33 36 34 37 31 34 32 34  32 65 65 30 64 39 30 30  |364714242ee0d900|
00000100  65 38 30 31 0a 00 00 00  00 00 00 00 00 00 00 00  |e801............|
00000110  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 20 1c 00 00  |............ ...|
00000130  0a 00 00 00 00 00 00 00  c8 0f 00 00 7d 02 00 00  |............}...|
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001a0  0a 00 00 00 01 00 00 00  00 00 00 00 80 51 01 00  |.............Q..|
000001b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  02 00 00 00 02 00 00 00  01 00 00 00 00 00 00 00  |................|
000001e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000210  00 00 00 00 00 00 00 00  00 00 00 00 55 51 50 54  |............UQPT|
00000220  50 55 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |PU..............|
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000250  00 00 00 00 00 00 00 00  00 00 00 00 4d 4d 46 42  |............MMFB|
00000260  4a 50 4c 44 49 45 45 4b  4b 4b 48 4b 4f 4d 46 41  |JPLDIEEKKKHKOMFA|
00000270  50 4b 45 46 47 46 43 48  46 42 48 4b 48 4b 45 4d  |PKEFGFCHFBHKHKEM|
00000280  4c 42 4a 4e 49 4a 42 4a  43 45 42 41 50 43 49 41  |LBJNIJBJCEBAPCIA|
00000290  42 4e 4e 44 50 46 43 50  42 50 4b 4b 4f 4c 4b 44  |BNNDPFCPBPKKOLKD|
000002a0  44 45 50 49 4d 50 44 46  41 46 44 47 42 49 00 00  |DEPIMPDFAFDGBI..|
000002b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002d0  00 00 00 00 00 00 00 00  00 00 00 00 f3 5c 16 66  |.............\.f|
000002e0  30 30 45 55 30 30 30 30  30 30 30 30 30 30 30 30  |00EU000000000000|
000002f0  30 30 30 30 00 00 00 00  00 00 00 00 00 00 00 00  |0000............|
00000300  64 00 00 00 01 00 00 00  64 00 00 00 01 00 00 00  |d.......d.......|
00000310  02 00 00 00 00 01 00 00  03 00 00 00 01 00 00 00  |................|
00000320  00 00 00 00 01 00 00 00  50 00 00 00 00 00 00 00  |........P.......|
00000330  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000640  00 00 00 00 54 4e 50 55  53 41 50 2d 37 33 30 32  |....TNPUSAP-7302|
00000650  33 36 2d 54 4a 45 5a 42  00 00 00 00 00 00 00 00  |36-TJEZB........|
00000660  00 00 00 00 52 46 55 53  59 34 34 43 32 33 57 51  |....RFUSY44C23WQ|
00000670  53 34 32 33 30 38 32 39  00 00 00 00 00 00 00 00  |S4230829........|
00000680  00 00 00 00 52 68 77 31  6f 33 61 42 64 77 4f 50  |....Rhw1o3aBdwOP|
00000690  54 69 32 49 00 00 00 00  00 00 00 00 00 00 00 00  |Ti2I............|
000006a0  00 00 00 00 30 30 30 30  30 30 30 30 00 00 00 00  |....00000000....|
000006b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000006c0  00 00 00 00 46 46 46 46  46 46 46 46 46 46 46 46  |....FFFFFFFFFFFF|
000006d0  46 46 46 46 00 00 00 00  00 00 00 00 00 00 00 00  |FFFF............|
000006e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00010000
```

roleoroleo, 2024-04-11:
I tested the shell script inside my cam with your ssid name.
It works properly.
So please, try to run the script manually.

Create a text file /tmp/configure_wifi.cfg with your data:
```
# Don't use backslash char \
# Don't use double quote " around ssid and psk
# Space chars are allowed
wifi_ssid=H-B-535_X
wifi_psk=blablabla
```
And run the script `/tmp/sd/yi-hack/script/configure_wifi.sh`

closed 2024-08-10 by github-actions[bot]