roleoroleo/yi-hack-Allwinner-v2 #857: Manual time not sticking after reboot

gne5, 2024-04-10:
Wondering if this can be changed to not reset after a reboot?

If the camera is not connected to the internet and I set the date & time via telnet with: [date -s "DDMMYY HH:MM:SS"] then it resets after reboot. And I get random reboots sometimes when it switches to and from IR night mode.

I tried enabling my router to act as a NTP time server and changed the address in the camera settings to the router IP but still no auto timing.

y21ga - v0.3.1

roleoroleo, 2024-04-11:
The cam has no battery and it's not able to keep the time after a reboot.
But, with ntp it should work.
Are you sure your ntp is working properly?

gne5, 2024-04-11:
It was an error in my routing, my bad. It's working now from the routers NTP server.

Thanks for the work. One small request is to add batch delete for videos.

closed 2024-04-11 by gne5

roleoroleo, 2024-04-12:
> One small request is to add batch delete for videos.

There are 2 methods for this:
- configure a free space value
[image]
- use the web service https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#delete-directories