roleoroleo/yi-hack-Allwinner-v2 #858: ONVIF configuration not saving on 0.3.2 in Firefox

fnordson, 2024-04-13:
Firmaware Version | 0.3.2
Base Version | 9.0.46.00_202110081405
Model Suffix | r35gb
Firefox | 124.0.2
Windows | 10

I have ONVIF enabled on the "Configurations" page, however if I go to page=configurations_onvif from there, all options set there are not being saved.  
It looks like the "save"-button doesn't do anything on Firefox. 
It did work on Vivaldi though.

TungstenE2, 2024-04-14:
btw, to which ONVIF Profiles do 'High' and 'Low' match? Would be good to use official namings.

[image]

roleoroleo, 2024-04-14:
Did you clear the cache of the browser?
There are many changes in the js file.

> btw, to which ONVIF Profiles do 'High' and 'Low' match? Would be good to use official namings.

Please explain me better the question.

fnordson, 2024-04-14:
> Did you clear the cache of the browser? There are many changes in the js file.

I did. 
It didn't help.

roleoroleo, 2024-04-15:
Could you try to debug with the development tools of the browser.
Open network tab and click "save configuration" button.
Is the response "error: false"?

TungstenE2, 2024-04-20:
> Did you clear the cache of the browser? There are many changes in the js file.
> 
> > btw, to which ONVIF Profiles do 'High' and 'Low' match? Would be good to use official namings.
> 
> Please explain me better the question.

See also https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/863

You name the profiles High and Low, but official Profiles are named A, C, D, G, M, S and T, So how do they match?

roleoroleo, 2024-04-21:
There is a misunderstanding about the terms. When I write "ONVIF Profile" I mean the media profile: the content returned by "GetProfiles" function.
In our case there are two profiles: a high res profile and a low res profile.
The profile you mentioned is the set of feature supported by the cam and is fixed to "S" (although in the latest version there are some implementations related to the T profile).

fnordson, 2024-05-01:
> Could you try to debug with the development tools of the browser. Open network tab and click "save configuration" button. Is the response "error: false"?

(Sorry for the late response, I've been traveling)

I tried that now and it worked. 
It works now altogether. Maybe it was the Firefox update to 125.0.3.

Thank you for your work :)

closed 2024-05-01 by fnordson

TungstenE2, 2024-05-27:
> There is a misunderstanding about the terms. When I write "ONVIF Profile" I mean the media profile: the content returned by "GetProfiles" function. In our case there are two profiles: a high res [...]

@roleoroleo 
To be honest, this is really confusing. There should only be one term when it comes to ONVIF Profile. If you use the term other than the official term, you better choose another wording/term.

roleoroleo, 2024-05-27:
I'm sorry but it's not my choice.
Here the description of Profile S: https://www.onvif.org/profiles/profile-s/
Here the description of the media profiles: https://www.onvif.org/specs/srv/media/ONVIF-Media-Service-Spec.pdf
returned by GetProfiles and GetProfile methods.
Maybe I could add the word "Media" before "profile" to make it clearer.