roleoroleo/yi-hack-Allwinner-v2 #869: Backup files for y623 (firmware 12.0.51.01_202303091901)

sebastianha, 2024-04-28:
Can someone provide me the factory / original backup files for a y623 camera with firmware 12.0.51.01_202303091901?

I have 3 cameras here and block 4, 6 and 7 differ. I thought at least block 4 should be the same (backup partition).

sebastianha, 2024-04-29:
Thanks, there block 4 is identical to one of my cams.

EEtinkerer, 2024-04-30:
> Thanks, there block 4 is identical to one of my cams.

want me to dump my other y623's?

sebastianha, 2024-04-30:
Would be interesting it you don't mind.
I wonder why they are different. I would have guessed that they are all the same.

EEtinkerer, 2024-04-30:
[Uploading backupArduinoDoorbell.zip…]()

sebastianha, 2024-04-30:
> Uploading backupArduinoDoorbell.zip…

I think this failed :)

sebastianha, 2024-04-30:
Interesting. These are the same but different from the first one you posted:

```
384afe584adc9ef2287b9a67def3bcc3  backup/mtdblock4.bin
4edd320d35cc6b3bf7394a4b7dd7efad  backup_doorbell/mtdblock4.bin
4edd320d35cc6b3bf7394a4b7dd7efad  backup_garage/mtdblock4.bin

a54cc6bae3ef44d4c1dafe6def5167ce  backup_1/mtdblock4.bin
384afe584adc9ef2287b9a67def3bcc3  backup_2/mtdblock4.bin
a868b950d4199434a3072095210edd62  backup_3/mtdblock4.bin
```

the lower 3 ones are mine where the "backup_2" is the same as your "backup" (the first one)

EEtinkerer, 2024-04-30:
Or theyre the same backup and i F'd up naming my files and I need to go pull my custom doorbell cams off the wall, are they completely identical? I used two nearly the same camera when i made my doorbell cam set ups

sebastianha, 2024-04-30:
Block 6&7 are always different, but 4 (and all others) are identical.

roleoroleo, 2024-05-03:
The flash depends on the Yi fw version.
If the fw changes, partitions can be different.

sebastianha, 2024-05-04:
Firmware is the same on all my 3 cameras

Vincent835, 2024-05-21:
My backup from Yi Home 2K PRO - YY623 - YFUSY94 - Model YYS.61121 v12.0.51.01_202303091901
[backup Yi Home 2K PRO - YY623 - YFUSY94 - Model YYS.61121.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15396193/backup.Yi.Home.2K.PRO.-.YY623.-.YFUSY94.-.Model.YYS.61121.zip)

closed 2024-09-19 by github-actions[bot]