roleoroleo/yi-hack-Allwinner-v2 #871: y21ga camera issues

poseidogithub, 2024-04-30:
Hello, 

I have this camera: 

y29ga_mb_v2.1
2020/04/07
realtek 8189ftv k91ky31 gkj9c
winband 25064jvsiq 2026

https://user-images.githubusercontent.com/7256096/111965191-c31cd780-8af5-11eb-9ccd-659eab8a008d.jpg
https://user-images.githubusercontent.com/7256096/111965196-c3b56e00-8af5-11eb-91ec-3dddf1ed0ba0.jpg
(pictures stolen from https://github.com/roleoroleo/yi-hack-MStar/issues/331 thanks @marcel89l :) ) 

according to the post, I should use this firmware: https://github.com/roleoroleo/yi-hack-Allwinner-v2 and y21ga  version.

According to the instructions, I should download it from release page, so I donwload this: https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.6/y21ga_0.2.6.tgz
I do a format on FAT32 of a 8gb sd card, copy the files to the sdcard and first issue... according to the instructions I should have a configure_wifi.cfg.ori  on the Factory folder: 

4 _(Optional) If you want to set wifi credentials, rename the file Factory/configure_wifi.cfg.ori to Factory/configure_wifi.cfg and edit the file with your username and password._

but on that folder I only have config.sh and factory_test.sh

anyhowm I put the card on my brand new camera and I only get "waiting to connect" audio message multiple times...

what I'm doing wrong?

thanks!

sebastianha, 2024-05-01:
I had the same problem after I pressed the reset button on the camera. The hack was still installed but the WiFi credentials no longer available. As the hack was still active on the camera it did not re-install it and therefore did not re-configure wifi.

For me it helped to "unbrick" the camera, so to reset it to factory and then start all over with the hack.

poseidogithub, 2024-05-01:
Tha la, but that's not exactly my problem, I can not install the hack on
the first place ..

sebastianha, 2024-05-01:
The "hack" is a small modification in the original firmware of the camera. If this is in place the wifi connect procedure is not being executed when I understand the code correctly. So when at some point the hack has been installed with wrong wifi credentials it will not fix this.

Try the following: copy the hack to the SD card and change name + edit the wifi config. Then edit the "config.sh" script, add the following lines *before* "### Disable the hack for next reboot"

```
    ### Set wireless credentials if configure_wifi.cfg exists
    if [ -e /tmp/sd/Factory/configure_wifi.cfg ]; then
        /tmp/sd/Factory/configure_wifi.sh
    fi
```
(copied from above)

This will make sure your wifi will definitely be configured. The file should look something like:

```
[...]
    cp /tmp/init.sh /backup/init.sh
    rm /tmp/init.sh

    ### Set wireless credentials if configure_wifi.cfg exists
    if [ -e /tmp/sd/Factory/configure_wifi.cfg ]; then
        /tmp/sd/Factory/configure_wifi.sh
    fi
fi

### Set wireless credentials if configure_wifi.cfg exists
if [ -e /tmp/sd/Factory/configure_wifi.cfg ]; then
    /tmp/sd/Factory/configure_wifi.sh
fi

### Disable the hack for next reboot
echo "### Disabling hack for next reboot"
if [ -e /tmp/sd/Factory.done ]; then
    rm -rf /tmp/sd/Factory.done
fi
if [ -e /tmp/sd/Factory ]; then
    mv /tmp/sd/Factory /tmp/sd/Factory.done
fi

sync
sync
sync

reboot
```

roleoroleo, 2024-05-07:
In addition to the problem reported by @sebastianha 

@poseidogithub
To choose the correct version of the hack, you need the Yi fw version.
According to your information I don't know which model is. Probably a y29ga but you need to check the Yi fw version,

closed 2024-09-05 by github-actions[bot]