roleoroleo/yi-hack-Allwinner-v2 #876: y211ga with 12.0.37.0_202211241030 firmware can't hack or connect wifi

djRicky, 2024-05-09:
Hi @roleoroleo thanks for your great job. I have a camera y211ga with the same 12.0.37.0_202211241030 firmware. The camera was updated using original Y! app. And now after a reset I can't connect anymore to any wifi. Always getting Wifi connection error. Then I try with the hacks y211ga_0.3.1.tgz and y211ga_0.3.2.tgz and the backup is done automatically but nothing more. Setting up the wifi with configure_wifi.cfg doesn't work and using QR code still getting the same error. I don't know how to downgrade or fix this error. When I apply the hack, the camera boot at first time, create the backup and reboot normally. The hack never create the LOG folder.

_Originally posted by @djRicky in https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535#issuecomment-2101180727_

djRicky, 2024-05-09:
[backup](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535#issuecomment-2101191564)
[board pictures](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535#issuecomment-2102577254)

djRicky, 2024-05-09:
I can downgrade to 9.0.36 but I still have the wifi error. Even after apply the hack.
But now I get a log.
Here are some:
[log_pwd_wrong.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15268278/log_pwd_wrong.tar.gz)
[log_pwd_wrong2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15268279/log_pwd_wrong2.tar.gz)

roleoroleo, 2024-05-10:
Try again to pair the cam with the original Yi app.
Don't use a 5 GHz wifi, bring the cam near your access point and try again.
If your AP uses wpa3, try to disable it.

djRicky, 2024-05-10:
Tried with:
-Reset several times
-8 different power supply (3 original Y! and 1 new)
-Different USB cables (original new too)
-5 different routers, wifi, SSID
-Last wifi try, old 54G router with WPA2, WPA and TKIP, AES combination.
-Y! app with QR code hacked or restoring
-Hacked with WiFi settings (permanent blinking Blue led after booting)

The problem start with firmware 12.0.37.0_202211241030. I have 5 cameras with this problem.
In this test, I downgraded to 9.0.36.00_202011101611 using only mtdblock2.bin 3 and 4 from [y211ga_9.0.36.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535#issuecomment-1327754657) for creating rootfs_y211ga, home_y211ga and backup_y211ga.

Where are stored the wifi drivers?
The downgrade restore old wifi drivers?
If the wifi drivers are stored in other partition than 2,3 or 4, I think that maybe the drivers in firmware 12.0.37.0_202211241030 are broken.

djRicky, 2024-05-10:
Lastest pwd_wrong log with 0 bytes of size but reset log has some info:
[log_start_reset.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15275495/log_start_reset.tar.gz)

roleoroleo, 2024-05-10:
> Where are stored the wifi drivers?

mtdblock3 and 4.

> The downgrade restore old wifi drivers?

Yes.

> If the wifi drivers are stored in other partition than 2,3 or 4, I think that maybe the drivers in firmware 12.0.37.0_202211241030 are broken.

Maybe.

djRicky, 2024-05-10:
Thanks for you answer. 
I am currently blocked, I don't know what to do anymore. I went back to version 12.0.37 and everything is the same.
If I think wrong, it seems to me that the Kami people made some update that blocks devices for those of us who do not pay the monthly service.

roleoroleo, 2024-05-12:
Maybe a hardware failure.

djRicky, 2024-08-12:
> Maybe a hardware failure.

How can I check the hardware? Do you have any script to do that? 

I'm thinking about a hypothesis: Could using a different charger than the original damage only some component of the wifi circuit? (transmitter or receiver) What about the quick chargers that have a 9 volt stage and some up to 12v?

djRicky, 2024-08-12:
> Maybe a hardware failure.

Hi, 

How can I check the hardware? Do you have any script to do so?

I'm thinking about a hypothesis: Could using a different charger than the original damage only some component of the wifi circuit? (transmitter or receiver) What about the quick chargers that have a 9 volt stage and some up to 12v?

closed 2024-12-12 by github-actions[bot]