roleoroleo/yi-hack-Allwinner-v2 #878: Video/Stream stops for a few seconds after motion detection

DivineJimmi, 2024-05-14:
Hi, i am using a Yi Dome U (2K).

Whenever motion is detected, the camera stops for a few seconds and is not updating its stream. After the motion detection, it sends an MP4 file to my FTP (Shinobi in my case) and the recorded video does pause too for a few seconds after motion is detected.

This happens:
Camera streams fine
I walk into the camera, triggering the motion detection
The camera streams for about 3 seconds fine
then the camera view stops for about 5 seconds
Then, when the view is streaming again i walk off the camera
A few seconds later the motion-detection-file gets uploaded

The uploaded file shows the same 5-second stop as the preview.
I think that the camera has some internal struggle to stream/record at the same time, right after it detected motion. Maybe not enough processing power? 

What am i doing wrong? I am already streaming in low-res RTSP. Is there any setting i can tune, to eliminate this behavior? It would be very nice to have a non-stopping video on my FTP-server, in case something bad happens in my home, i dont want my camera to stop for some seconds.

roleoroleo, 2024-05-15:
I think this problem depends on the cpu/memory usage.
When a motion is detected, the cam tries to take a high resolution snapshot and uses a lot of resourses.
Try to:
- disable unneeded  features
- force low resolution snapshots
- disable the snapshot creation when an event is triggered (leave blank "Topic Suffix for jpeg image")

DivineJimmi, 2024-05-15:
Thank you. Disabling the snapshot function solved my problem. I dont need this function anyway.

closed 2024-09-13 by github-actions[bot]