roleoroleo/yi-hack-Allwinner-v2 #882: Wifi doesn't connect on YI Cam Home Pro 2K ZFUS/y623

GrvtTera, 2024-05-17:
Hello,

Thanks @roleoroleo for your work!
i've an issue on my new (and firts) yi cam home pro 2k. i've prepare the configure_wifi.cfg with my wifi credentials inside, every steps seems getting good, but when i power on my cam i've got the blue led flashing in continue.

I've maybe forgot a step?

Can you help me please?

Thx.

closed 2024-05-17 by GrvtTera

GrvtTera, 2024-09-03:
Hello @vaboston i've successfuly connect my cam on my wifi, i had few problems with the process and i tried several times and it finally worked for me. I don't have a reason for my previous issue. So retry the entire process. Good luck and have a good day.

vaboston, 2024-09-05:
Thanks @vaboston, I used the app to set wifi, it works but when it restarts, I have to go back through the application, not ideal but it unblocked me