roleoroleo/yi-hack-Allwinner-v2 #883: Wifi password error for my last 8th camera

gl2007, 2024-05-17:
I have posted couple of issues I faced and how I got them to work in last 2 days. However, I am not able to get my last 8th camera to work no matter how much I tried. 

The Yi Home Camera with serial BFUSxxxxxx and firmware 9.0.19.* is refusing to connect to Wifi despite resetting it and gives "Wifi password error" when in fact it is the correct password and same used for all other 7 ones I have. I thought maybe my spectrum router has banned it or blacklisted it. So, I went into "My Spectrum" app and removed the disconnected device (it had connected once I believe yesterday after I left it sitting for a long time). Now, it just gives up with that error "Wifi password error".

I managed to get the output of the "config.sh" and "lower_half_init.sh" for 2 scenarios, 
1. booting with SD card with hack inserted (files without _0.txt)
2. booting into regular mode after resetting and then inserting the SD card with the hack (files with _0.txt) . 

I have attached both these outputs here.

Let me know if you see anything which can indicate hardware failure or anything else. I see many "crash" logs in there.
[hackoutput_0.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15358624/hackoutput_0.txt)
[config-output_0.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15358625/config-output_0.txt)
[config-output.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15358626/config-output.txt)
[hackoutput.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15358627/hackoutput.txt)

roleoroleo, 2024-05-20:
Strange messages in the 1st file:
```
rmm.c(msg_proc-2225)[06:42:49.679]:got RMM_SPEAK_WAIT/backup/tools/wpa_supplicant: line 3: syntax error: unexpected end of file (expecting ")")
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
```
And a lot of messages related to wifi pwd problems.

Is this cam the same model?
Did you try to upgrade to 12.x.x.x?

gl2007, 2024-05-22:
Thanks for responding. Yes, model/firmware version is the same as remaining 7, got them as 2 packs of 4 when I bought them couple of years back. This same hack had worked back them and even worked once night before I made this post.

Have not updated any firmware, still stock on 9.0.19* for all of these.

So question is why only this camera? Maybe a hardware issue has occurred? And why the modules seem to be crashing?

roleoroleo, 2024-05-25:
Probably it's a hardware failure.
If the problem is related to a corrupted firmware you could try to restore mtdblock4 from a working cam.
Then, if it still doesn't work, mtdblock3 as well.
But don't try this method with other partitions.

gl2007, 2024-05-25:
I see, I was wondering the same. I am willing to try the partition restore option to see if it can bring it back. However, so you have instructions on how to do it? I have flashed firmwares into android mini PCs earlier but just want to be safe with precise instructions :)

roleoroleo, 2024-05-25:
The procedure is simple:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

gl2007, 2024-08-26:
Issue was resolved automatically with no actions from me. Looks like my router is involved in some way that I can't figure out?

closed 2024-12-25 by github-actions[bot]