roleoroleo/yi-hack-Allwinner-v2 #885: Two Yi Dome Camera U disconnect from WIFI every few hours randomly

elmaswebon, 2024-05-24:
Hi there!

I have two YHS 5020 cameras working fine with versión 0.32.

The problem is that every several hours, sometimes more, sometimes less, one of them disconnects from the WIFI and I lose RTSP and web access. Over time the other one usually disconnects as well.

If I restart the WIFI (only the WIFI, not the router), they reconnect immediately. If I disconnect and reconnect the cameras (without restarting the WIFI), they reconnect without problem too.

They are connected directly to the router. I have tried connecting them to an AP and the same thing happens.

I have tried deactivating services (FTP, SSH), and also activating their connection to the cloud, but nothing works, they always keep disconnecting.

I'm looking for a log on the SD that I can consult in case it gives me more information about what's happening, because the camera continues to work, only the WIFI is disconnected, but I can't find any log file.

I also have a Yi Home 1080p, also with 0.32, connected to the same router, same WIFI, same configuration and it does not disconnect.

Is there a log that I can consult or something that I can contribute here to analyze what is happening?

Thanks in advance.

elmaswebon, 2024-05-25:
Well, it seems that disabling MQTT has solved the problem. For some reason, having the Home Assistant integration and MQTT activated seems to be problematic.

I will update the situation again in a few days to see if it remains stable.

roleoroleo, 2024-05-25:
Does the cam reboot when it disconnects from the router?
Or simply it lose the wifi?
Check the uptime.

elmaswebon, 2024-05-25:
> Does the cam reboot when it disconnects from the router? Or simply it lose the wifi? Check the uptime.

Ir simply reconnects instantly without rebooting.

It seems that this time took a little longer, but one of them got disconnected again.

I´ve just reset the wifi, the disconnected camera has connected again and the uptime is 10 hours:

[image]

roleoroleo, 2024-05-25:
There is no log about wifi connections.
But you should find some useful information in the standard output of the yi processes.
For example, try to kill dispatch and restart it:
```
cd /home/app
killall dispatch
./dispatch
```
and check the output when the cam disconnects.

elmaswebon, 2024-05-26:
Ok, I see that executing this restarts the camera and leaves the command executed monitoring. So what you propose is to leave a terminal open with this command and when it disconnects from the Wi-Fi see if anything is seen in the last lines?

elmaswebon, 2024-05-26:
Another interesting thing is that I have the cameras' crontab configured to restart every day at 4AM.

Last night one of them was disconnected again, I left it disconnected to see if with the 4AM restart it would reconnect and this morning I saw that it was still disconnected. I restarted the WIFI and saw that the camera had an uptime of 4.5 hours, that is, it had restarted at 4AM, but it was not connected to the WIFI...

roleoroleo, 2024-05-27:
> Ok, I see that executing this restarts the camera and leaves the command executed monitoring. So what you propose is to leave a terminal open with this command and when it disconnects from the Wi-Fi see if anything is seen in the last lines?

This command shouldn't restart the camera. If reboot happens, add `killall watch_process` before `killall dispatch`

Dotatay, 2024-05-27:
When executing the dispatch command, the PTZ only starts turning likely in the same way as it does during a restart.

Since updating to version 0.3.2, I have also issues with the WIFI connection. My router/AP shows constant disconnects and reconnections of the camera. This making it difficult to access the webinterface of the hack.

I have created a dispatch-log (lines such as cost and heartbeat are removed):

```
dispatch.c(check_wifi_connect-1816)[18:12:39.509]:signal_quality(58)

dispatch.c(check_wifi_connect-1816)[18:22:41.705]:signal_quality(59)

dispatch.c(do_mq_processcat: can't open '/proc/timeout_cnt': No such file or directory
ash: 100: unknown operand
ash: 0: unknown operand
killall: p2p_tnp: no process killed
ash: 1: unknown operand

dispatch.c(do_monitor_wifi-1955)[18:29:35.803]:wifi disconnected, now reconnect wifi

dispatch.c(do_monitor_wifi-1965)[18:29:45.519]:wpa connected , now dhcp
killall: udhcpc: no process killed
udhcpc: started, v1.27.2
udhcpc: sending discover
udhcpc: sending select for 192.168.0.33
udhcpc: lease of 192.168.0.33 obtained, lease time 864000

dispatch.c(do_monitor_wifi-1967)[18:29:45.892]:wpa connected , now dhcp finish

dispatch.c(check_wifi_connect-1722)[18:29:46.941]:bssid(xx:xx:xx:xx:xx:xx)

dispatch.c(check_wifi_connect-1744)[18:29:46.941]:ip(192.168.0.33)

dispatch.c(check_wifi_connect-1755)[18:29:46.949]:gw(192.168.0.1)

dispatch.c(check_wifi_connect-1774)[18:29:48.059]:mask(255.255.255.0)

dispatch.c(check_wifi_connect-1784)[18:29:48.090]:mac(xx:xx:xx:xx:xx:xx)

dispatch.c(check_wifi_connect-1798)[18:29:48.115]:signal_quality(58)
cmd(rm /tmp/sd/log_wifi_connected.tar.gz;rm /tmp/sd/wifi_connected -r;mkdir /tmp/sd/wifi_connected;ps > /tmp/sd/wifi_connected/ps1.txt;cp /home/homever /tmp/sd/wifi_connected;lsmod > /tmp/sd/wifi_connected/lsmod.txt;ls -laSR /home > /tmp/sd/wifi_connected/file.txt;cp /tmp/wpa_supplicant.conf  /tmp/sd/wifi_connected;cp /tmp/log.txt /tmp/sd/wifi_connected;cp /tmp/log_oss.txt /tmp/sd/wifi_connected;df /home > /tmp/sd/wifi_connected/df.txt;df / >> /tmp/sd/wifi_connected/df.txt;tar -cf /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/wifi_connected;mkdir /tmp/sd/log; mv /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/log;rm /tmp/sd/wifi_connected -r &rm: can't remove '/tmp/sd/log_wifi_connected.tar.gz': No such file or directory
rm: can't remove '/tmp/sd/wifi_connected': No such file or directory
cp: can't stat '/tmp/log.txt': No such file or directory
cp: can't stat '/tmp/log_oss.txt': No such file or directory
tar: removing leading '/' from member names
mkdir: can't create directory '/tmp/sd/log': File exists
)

dispatch.c(check_wifi_connect-1816)[18:32:43.602]:signal_quality(58)

g_dispatch_info.mmap_info->ptz_x_runtime =cat: can't open '/proc/timeout_cnt': No such file or directory
ash: 100: unknown operand
ash: 0: unknown operand
killall: p2p_tnp: no process killed
killall: cloud: no process killed
ash: 1: unknown operand
killall: udhcpc: no process killed
udhcpc: started, v1.27.2
udhcpc: sending discover
udhcpc: sending select for 192.168.0.33
udhcpc: sending select for 192.168.0.33
udhcpc: lease of 192.168.0.33 obtained, lease time 864000
13 2

dispatch.c(do_monitor_wifi-1955)[18:41:14.445]:wifi disconnected, now reconnect wifi

dispatch.c(do_monitor_wifi-1965)[18:41:24.050]:wpa connected , now dhcp

dispatch.c(do_monitor_wifi-1967)[18:41:27.679]:wpa connected , now dhcp finish

dispatch.c(check_wifi_connect-1722)[18:41:28.732]:bssid(xx:xx:xx:xx:xx:xx)
[... 125 lines cut ...]
dispatch.c(do_monitor_wifi-1965)[20:47:40.466]:wpa connected , now dhcp

dispatch.c(do_monitor_wifi-1967)[20:47:49.825]:wpa connected , now dhcp finish

dispatch.c(check_wifi_connect-1722)[20:48:13.692]:bssid(xx:xx:xx:xx:xx:xx)

dispatch.c(check_wifi_connect-1744)[20:48:13.692]:ip(192.168.0.33)

dispatch.c(check_wifi_connect-1755)[20:48:13.700]:gw(192.168.0.1)

dispatch.c(check_wifi_connect-1774)[20:48:14.768]:mask(255.255.255.0)

dispatch.c(check_wifi_connect-1784)[20:48:14.794]:mac(xx:xx:xx:xx:xx:xx)

dispatch.c(check_wifi_connect-1798)[20:48:14.821]:signal_quality(59)
cmd(rm /tmp/sd/log_wifi_connected.tar.gz;rm /tmp/sd/wifi_connected -r;mkdir /tmp/sd/wifi_connected;ps > /tmp/sd/wifi_connected/ps1.txt;cp /home/homever /tmp/sd/wifi_connected;lsmod > /tmp/sd/wifi_connected/lsmod.txt;ls -laSR /home > /tmp/sd/wifi_connected/file.txt;cp /tmp/wpa_supplicant.conf  /tmp/sd/wifi_connected;cp /tmp/log.txt /tmp/sd/wifi_connected;cp /tmp/log_oss.txt /tmp/sd/wifi_connected;df /home > /tmp/sd/wifi_connected/df.txt;df / >> /tmp/sd/wifi_connected/df.txt;tar -cf /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/wifi_connected;mkdir /tmp/sd/log; mv /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/log;rm /tmp/sd/wifi_connected -r &rm: can't remove '/tmp/sd/log_wifi_connected.tar.gz': No such file or directory
rm: can't remove '/tmp/sd/wifi_connected': No such file or directory
cp: can't stat '/tmp/log.txt': No such file or directory
cp: can't stat '/tmp/log_oss.txt': No such file or directory
tar: removing leading '/' from member names
mkdir: can't create directory '/tmp/sd/log': File exists
)

dispatch.c(check_wifi_connect-1816)[20:53:14.300]:signal_quality(58)
```

roleoroleo, 2024-05-28:
The log doesn't show errors related to the wifi disconnection reason.
Suddenly, this log happens:
`dispatch.c(do_monitor_wifi-1955)[20:46:24.640]:wifi disconnected, now reconnect wifi`
and I'm not able to understand why. I haven't got the sources of this Yi binary.

Could you check if the same error occurs with hack 0.3.1.
In this case I can check if I "moved something" around this scripts...

elmaswebon, 2024-05-28:
I think it also happened with 0.3.1 but I´ve just returned back to 0.3.1 both cameras and will report how it works.

elmaswebon, 2024-05-29:
So, after going back to 0.3.1, I´m having the same disconnections.

Here´s the output of the dispatch command of both cameras.

[dispatch_entrada.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15479182/dispatch_entrada.log)
[dispatch_salon.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15479185/dispatch_salon.log)

As you said, the cameras don´t disconnect after executing the dispatch command, they only recalibrate the PTZ again (and, by the way, after executing it the PTZ controls won´t respond again after restarting the camera).

Dotatay, 2024-05-29:
Sorry for the confusion. The disconnects also happen with version 0.3.1. I've been trying to remember what settings I changed and have done some testing...

After turning off the "Disable Cloud" option (DISABLE_CLOUD=no), I haven't had any disconnects in the last few hours.

The downside is that the RTSP connection doesn't work anymore:

> [tcp @ 0x7f70002d90] Connection to tcp://192.168.0.33:554?timeout=0 failed: Connection refused
> rtsp://xxx:xxx@192.168.0.33/ch0_1.h264: Connection refused

Also, continuous video recording doesn't work anymore ("Save video when a motion is detected" = disabled to enable "video will be always saved").

However, with ipc_read I can still see that motion and sound are being detected.

This may be because the camera can only access the local network at the moment and is blocked from the Internet.

Dotatay, 2024-05-30:
In the meantime, I’m testing this script. This could reduce the downtime to ~10 secounds:
```
while true; do if wpa_cli -i wlan0 status | grep -q "wpa_state=INTERFACE_DISABLED"; then wpa_cli -i wlan0 reconnect; fi; sleep 10; done > /dev/null 2>&1 &
```

roleoroleo, 2024-05-31:
@Dotatay 
Please, explain me better.
You set DISABLE_CLOUD=no and configured the firewall to block the traffic.
So, the cam is not able to connect to the server.
Is it correct?
If yes, please run a tcpdump in the cam: I need to know which connections the cam is trying.

Dotatay, 2024-06-02:
Yes, for testing purposes, the cloud was active (system.conf: DISABLE_CLOUD=no) and my router/firewall blocked the camera from accessing the internet. Unfortunately, even with this setting, the camera still loses the WIFI connection, but less frequently. Maybe this is just a coincidence?

These are all the addresses that were blocked last night:
```
8.208.8.243
47.254.89.110 (Router Service?)
47.90.209.117 (Router Service?)
96.152.158.18
110.89.254.47
117.209.90.47
158.254.117.161
161.117.254.158 (Router Service?)
api.eu.xiaoyi.com
ec2-52-57-227-127.eu-central-1.compute.amazonaws.com
ec2-18-158-152-96.eu-central-1.compute.amazonaws.com
224.0.0.250
243.8.208.8
```

I then captured a disconnection of the camera with tcpdump. Before this, the blocked attempts to connect to '8.208.8.243.53' was without problems for several hours:
```
03:19:41.558024 IP cam-yir30gb.fritz.box.43278 > 8.208.8.243.53: 30117+ A? api.eu.xiaoyi.com. (35)
03:19:47.551972 EAPOL key (3) v2, len 95
03:19:47.553010 EAPOL key (3) v1, len 117
03:19:47.577293 EAPOL key (3) v2, len 159
03:19:47.577744 EAPOL key (3) v1, len 95
03:19:47.840173 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 54:xx:xx:xx:xx:xx (oui Unknown), length 300
03:19:47.857904 IP 192-168-0-1.fritz.box.67 > cam-yir30gb.fritz.box.68: BOOTP/DHCP, Reply, length 548
03:19:47.867682 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 54:xx:xx:xx:xx:xx (oui Unknown), length 308
03:19:48.050286 IP 192-168-0-1.fritz.box.67 > cam-yir30gb.fritz.box.68: BOOTP/DHCP, Reply, length 548
03:19:49.255716 ARP, Request who-has 192-168-0-1.fritz.box (Broadcast) tell cam-yir30gb.fritz.box, length 28
03:19:49.261226 ARP, Reply 192-168-0-1.fritz.box is-at xx:xx:xx:xx:xx:xx (oui Unknown), length 28
03:19:51.615692 ARP, Request who-has 192-168-0-1.fritz.box tell cam-yir30gb.fritz.box, length 28
03:19:51.621069 ARP, Reply 192-168-0-1.fritz.box is-at xx:xx:xx:xx:xx:xx (oui Unknown), length 28
...

07:49:43.809290 IP cam-yir30gb.fritz.box.34689 > 8.208.8.243.53: 12384+ A? api.eu.xiaoyi.com. (35)
07:49:46.328316 EAPOL key (3) v2, len 95
...

23:39:38.386990 IP ec2-52-57-227-127.eu-central-1.compute.amazonaws.com.80 > cam-yir30gb.fritz.box.36290: Flags [.], ack 116, win 453, options [nop,nop,TS val 1752475639 ecr 62131], length 0
23:39:38.695282 EAPOL key (3) v2, len 127
...
```

I also tried to get information about the disconnects by using wpa_cli. I think a part of the problem is, that the wlan0 interface is temporarily unavailable:
`Failed to connect to non-global ctrl_ifname: wlan0`

Dotatay, 2024-06-03:
Yesterday I disabled RTSP for testing purposes, but I reactivated the Disable Cloud option. Within 24 hours I had only one disconnection. I also noticed that the output of 'pc_read is now delayed by at least 10-20 seconds.

elmaswebon, 2024-06-03:
So, a little more information.

I´ve been closely monitoring both cameras and found something.

Both connected at the same time, doing the same, but suddenly one of them starts sending/receiving a lot more information than the other one.

Shortly after, this camera loses access via http, but the rtsp keeps working. Some minutes later, it loses rtsp too and there´s nothing I can do to comunicate with it.

The AP, however, continues to show the camera conected even though it has been disconnected for about 30 minutes:

[image]

I´ve tried connecting them directly to the router and to two different APs. With fixed IP in the camera, fixing it in my DHCP server or leaving the DHCP assign a different one each time. Same result every time...

Dotatay, 2024-06-03:
@elmaswebon 
Could you please disable RTSP on your cameras and then reboot them? If this doesn’t reduce the disconnects for you, our issues might be different and I might be in the wrong place.

elmaswebon, 2024-06-04:
> @elmaswebon Could you please disable RTSP on your cameras and then reboot them? If this doesn’t reduce the disconnects for you, our issues might be different and I might be in the wrong place.

I'm away from home until the weekend and I need to have both cameras sending the stream to my Home Assistant, but as soon as I get back, I'll try it and let you know.

roleoroleo, 2024-06-07:
DISABLE_CLOUD=yes
should ensure the same results you obtain with an external firewall.
All known addresses are blacklisted at system level: router command for IP addresses and /etc/hosts for aliases.
Could you try if the behavior is the same?

elmaswebon, 2024-06-08:
So, my conclusions so far.

- No matter what I change in the settings, the cameras randomly disconnect from the WIFI

- I do not see an obvious change in the disconnection time by using one or other configuration options

- If I connect the cameras directly to the router's WIFI, once they disconnect from the WIFI, they do not reconnect. If I connect the cameras to a separate AP, the cameras also disconnect, but after a while they reconnect and work normally again

- Since I don't have logs, I can't know exactly what is happening, but before the complete disconnection of the WIFI, it seems that the services are falling little by little. That is, at a given moment I have RTSP but no access via HTTP, for example. Or I can have HTTP but not RTSP

In any case, at least having them connected to the AP and returning to normal automatically, it is not a big problem for me, just a minor problem and a partial loss of recordings that it would be nice to be able to solve, although given what I have seen, I think there will be no simple solution.

Dotatay, 2024-06-11:
@elmaswebon 
Too bad it doesn't work for you. Perhaps you can get some useful information from this script. Add it as cron oneshot after boot or to startup.sh:
```
while true; do if ! wpa_cli -i wlan0 status | grep -q "wpa_state=COMPLETED"; then echo -e "$(date): CONNECTION LOST\n\n$(wpa_cli -i wlan0 status)\n\n$(wpa_cli interface)\n\n$(ps)\n-----\n\n" >> /tmp/sd/yi-hack/wlanlog.txt; fi; sleep 1; done &
```
As soon as the connection is lost wpa_cli status (available interfaces) and all running processes are written to wlanlog.txt.

@roleoroleo 
Ok, I will try it and check the camera connection attempts to yi-servers with "DISABLE_CLOUD=yes". Should I create a new issue because it's different from elmaswebon issue?

closed 2024-10-17 by github-actions[bot]