roleoroleo/yi-hack-Allwinner-v2 #891: Device Error on RESTful API

KayGundhardt, 2024-05-28:
Hello everybody!

Tried to set up a RESTful command within HomeAssistant to turn the webcams recording feature on and off (see below). As these are my first baby steps into API commands any help or tips would be kindly appreciated - thank you! ❤️

[This here is this repository’s link to its API documentation](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#set-configuration) which tells me to change the configuration taking into account:

>```
>Set configuration
>
>http://IP_CAM/cgi-bin/set_configs.sh?conf=CONF
>```
>
>|CONF | action | 
>|--- | --- |
>|system  | Update system configuration |  
>|(…)|(…) |
>
>Need POST method with a json file.

The relevant system setting and its payload within would be:

```
{
…,
"REC_WITHOUT_CLOUD":"yes",
…
}
```

And my RESTful service command from HA to the cam is:

```
rest_command:
  cam_eg_surveillance_on:
    url: "http://username:pass@0.0.0.0:8080/cgi-bin/set_configs.sh?conf=system"
    method: POST
    payload: "REC_WITHOUT_CLOUD=yes"
    content_type: 'application/json; charset=utf-8'
  cam_eg_surveillance_off:
    url: "http://username:pass@0.0.0.0:8080/cgi-bin/set_configs.sh?conf=system"
    method: POST
    content_type: 'application/json; charset=utf-8'
    payload: "REC_WITHOUT_CLOUD=no"
```

But when I call the service I get only 

```
Error: Client error occurred when calling resource "http://username:pass@0.0.0.0:8080/cgi-bin/set_configs.sh?conf=system"
Result:
params:
  domain: rest_command
  service: cam_eg_surveillance_off
  service_data: {}
  target: {}
running_script: false
```

Any ideas ? :(

KayGundhardt, 2024-05-28:
Actually found out that 

https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/www/httpd/cgi-bin/set_configs.sh

does not include a manipulation of my desired keys value (REC_WITHOUT_CLOUD) although the

get_configs.sh

script lists this key. Is it possible to add the possibility to change this value into 

set_configs.sh

Would be great! :)

roleoroleo, 2024-05-28:
It should work using set_configs.sh
Try with this content:
{"REC_WITHOUT_CLOUD":"yes"}

To test it, try a browser extension like this one:
[image]

KayGundhardt, 2024-05-28:
Solution is coming!!
:partying_face:

Correct syntax must be like this:

```
rest_command:
  cam_eg_surveillance_on:
    url: "http://0.0.0.0/cgi-bin/set_configs.sh?conf=system"
    username: “username“
    password: “password“
    method: POST
    payload: ‘{"REC_WITHOUT_CLOUD“:“yes"}‘
    content_type: 'application/json; charset=utf-8'
  cam_eg_surveillance_off:
    url: "http://0.0.0.0/cgi-bin/set_configs.sh?conf=system"
    username: “username“
    password: “password“
    method: POST
    content_type: 'application/json; charset=utf-8'
    payload: ‘{"REC_WITHOUT_CLOUD“:“no"}‘
```

As you can see I pulled out the username and pass from the url and especially changed payload formatting!

Working like a charm now!

closed 2024-05-28 by KayGundhardt