roleoroleo/yi-hack-Allwinner-v2 #892: Yi Pro 2K Home - High stream corrupts after two simultaneous connections

Quedale, 2024-05-29:
Good day!

First, thanks for this great project!

I've had issues with the High stream profile dropping frames (especially important i-frame) when more than one client is connected.
All clients no longer receive I-frames, which doesn't allow the stream to show the entire picture. It only shows a grey screen with partial updates.

[image]

I understand that the camera has limited resources, but the current behavior doesn't seem functional, since it may corrupt an existing stream fed to an NVR.
If the camera simply doesn't have enough resource to serve more than one client, I would expect it to refuse additional connections to keep the existing stream healthy.

The Low profile still works fine even after the High profile was corrupted.

Here's the camera's resources while two clients are simultaneously connected:
```
Mem: 57956K used, 2956K free, 1124K shrd, 192K buff, 4176K cached
CPU: 38.9% usr 15.2% sys  0.0% nic 43.3% idle  0.0% io  0.0% irq  2.4% sirq
Load average: 1.46 1.37 1.31 2/108 16988
```

Let me know if I can provide any relevant logs to help diagnose the issue.

Thanks!

roleoroleo, 2024-05-29:
As you can read here
http://www.live555.com/liveMedia/faq.html#scalability
there is no limit in the code about concurrent connections.

But 1 connection at a time is too little, with my cam I tested 5 connections (high res) without problem with these stats:
```
Mem: 53384K used, 7528K free, 1148K shrd, 656K buff, 6868K cached
CPU: 31.3% usr 10.8% sys  0.0% nic 57.7% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 1.18 1.16 1.18 3/120 32205
```

Probably is related to the size of the I-Frame, like the problem fixed in #846.
You should take a log of the RTSP server daemon.
```
killall wd_rtsp.sh
killall rRTSPServer
rRTSPServer -m y623 -r both -a no -p 554 -d 15
```

Quedale, 2024-05-29:
I'm glad to hear that it's supposed to work.

The first time I ran the rRTSPServer, it failed. (Probably not relevant)
Regardless, here's the log of the failure:
[Failed start.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15486102/Failed.start.txt)

Here's about 10 seconds worth of log, while two clients are connected:
[rRTSPServer_logs.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15486107/rRTSPServer_logs.txt)

Thanks or this prompt response!

roleoroleo, 2024-05-31:
> The first time I ran the rRTSPServer, it failed. (Probably not relevant) Regardless, here's the log of the failure: [Failed start.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15486102/Failed.start.txt)

This behavior is normal: you killed the daemon while a client connection was running.
You have to wait for the system to free up network resources.

> Here's about 10 seconds worth of log, while two clients are connected: [rRTSPServer_logs.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15486107/rRTSPServer_logs.txt)

> EDIT: Looking at the logs, I would guess it broke around '1716993824219'. Where `start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0` started to show.

This line is when the daemon starts to deliver frames, so it's not a problem.
In the previous lines, the daemon was running but no clients was connected yet.

There is only a problem between lines 1716993828570 and 1716993829758: there are 23 lost frames after an I-Frame.
The lost frames are related to the delivery phase, there are no errors in the capture phase.
I don't know why. It could be a buffer problem (the size of the buffer is 393216 bytes) or a performance issue.
The peculiarity of this model is that I-Frames are very big. In the models I own these are about 150 KB.

Quedale, 2024-09-18:
I'm sorry, I've been busy for a long while.

Is there anything I can do to help determine the root cause?

It's not exactly essential to my surveillance network, but it would be nice to improve its stability.

roleoroleo, 2024-09-20:
Please, try this beta:
https://github.com/user-attachments/files/17078115/y623_0.3.2.tgz

Quedale, 2024-09-21:
I upgraded using [these](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Manual-firmware-upgrade) steps.

The upgrade went fine, but ONVIF and RTSP credentials no longer works.
HTTP access still works.

I tried removing the password entirely and rebooted the camera, but I still get Unauthorized response.

Quedale, 2024-09-21:
I would like to make a correction, hadn't removed the credentials entire, only the password and left the user. (Still didn't work)

I tried a fresh install and I had a different result.
I'm now authorized to access ONVIF service, but the rtsp server isn't started.
I tried all RTSP server program options, but always reverts to the standard one.

I set credentials again so that I can access SSH, but at that point I started getting Unautorized again just like after I did a manual upgrade.

rRTSPServer Logs:
```
1726877833165: the size of the buffer is 917872
1726877833166: capture - mapping file fshare_frame_buf, size 917872, to b6cf3000
1726877833167: capture - closing the file fshare_frame_buf
1726877833168: capture - frame header size = 28
1726877833169: capture - starting capture main loop
1726877833170: capture - buf_idx_end == buf_idx_end_prev
1726877833181: capture - buf_idx_end == buf_idx_end_prev
1726877833191: capture - buf_idx_end == buf_idx_end_prev
1726877833201: capture - buf_idx_end == buf_idx_end_prev
1726877833211: capture - buf_idx_end == buf_idx_end_prev
1726877833221: capture - buf_idx_end == buf_idx_end_prev
1726877833231: capture - buf_idx_end == buf_idx_end_prev
1726877833242: capture - buf_idx_end == buf_idx_end_prev
1726877833252: capture - buf_idx_end == buf_idx_end_prev
1726877833262: capture - buf_idx_end == buf_idx_end_prev
```

roleoroleo, 2024-09-21:
Same errorSame problem described here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/909

I'm trying to fix it.

Quedale, 2024-09-21:
Thanks! 
Looking forward to it!

No rush on my account. I'm just happy that it's being worked on.

Quedale, 2024-11-18:
I tried the latest firmware 0.3.3 and there's some good and some bad news.

The good news is that the original issue seems resolved!!
My camera now supports more than one simultaneous HIGH streams.
Although there seems to be a considerable amount of ghosting happening even with a single client but its tolerable. (Previous picture details are still slightly visible after the picture changed)

Only the standard live555 actually works.

The bad news is that some other functionalities broke.

ONVIF Authentication didn't work after a manual upgrade, but was fixed with a factory reset.
SSH access is always denied.

Camera snapshots aren't loading. When I tried the snapshot URLs in the browser, I get the following:
`The image “http://xxxx/cgi-bin/snapshot.sh?res=high&watermark=yes” cannot be displayed, because it contains errors.`
I also noticed that the screenshot is actually working for a very short period of time after booting.

I would like to provide additional details, but I'm not sure how without SSH access.

Thanks!!

roleoroleo, 2024-11-18:
Probably something was wrong in the update procedure.
I tested the hack with several cams (including a y623) and it worked properly.
Please, format again the sd card and try to install 0.3.3 from scratch.

Quedale, 2024-11-18:
Good news is that the snapshots works now and we can throw the ONVIF authentication issue under a failed upgrade as well.

Unfortunately, SSH and FTP still give me authentication failed errors.
For the purpose of testing, I'm using the same password as HTTPD which works fine.

Quedale, 2024-11-18:
Omg I just had to use "root" as that user.
I'm an imbecile.

Sorry about that!

I think we can close this! (Victory!!)

I will open something new if I encounter something else.

Thanks for your help!!