roleoroleo/yi-hack-Allwinner-v2 #893: h30ga and r35gb on 0.3.0 version do not find 0.3.2 online update

morenod, 2024-05-31:
When trying to upgrade h30ga and r35gb, they say:

[image]

But you click on upgrade firmware and:

[image]

maxen0815, 2024-05-31:
I had a simmilar issue and upgraded to 0.3.2 manually - now it seems to work again.
[image]

Upload File via FTP and follow the instructions to install the update manually:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Manual-firmware-upgrade

djiesr, 2024-06-01:
Same here... I have many cameras, if I can didn't have to set a FTP on each , that can be a good thing. Hope to update to 3.3 from 3.0. Thanks

roleoroleo, 2024-06-01:
There was a bug in the previous version of busybox.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/779
You have to update with the manual procedure.

djiesr, 2024-06-01:
ok.
I try turn on the FTP Serve, but I got:
421 Can't change directory to /home/ftp/ [/]
I miss something?

morenod, 2024-06-03:
Upgrade completed using the manual procedure, thanks!!

closed 2024-06-03 by morenod