roleoroleo/yi-hack-Allwinner-v2 #894: Two timestamps within snapshot - any way to remove one?

maxen0815, 2024-05-31:
Hi there,

I currently do see two timestamps within the snapshots on a newly hacked (Kami) camera.
My old devices don't show this duplicate. Any way to remove the original Kami timestamp?

[image]

roleoroleo, 2024-06-01:
How are you calling this snapshot?
Onvif? Web service?

maxen0815, 2024-06-01:
Shows up in web interface snapshot as well as in ftp uploaded ones.
The ONVIF and RTSP streams only shows left upper timestamp.

[image]

roleoroleo, 2024-06-01:
The upper left osd is made by Yi processes and you can't remove it.
The lower right osd is made by the hack and can be removed with the option watermark=no:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#snapshot
With onvif there is a specific option:
[image]

maxen0815, 2024-06-11:
Thanks for your reply. So I can close this one.

closed 2024-06-11 by maxen0815