roleoroleo/yi-hack-Allwinner-v2 #895: FTP uploads sometimes 1min without context

DivineJimmi, 2024-06-01:
Hey, i use yi-hack for 5 Yi Dome U 2K models. It works really good, most of the time.

But what i discovered, what makes no sense for me:

Sometimes the camera pushes an .mp4 file to the FTP server, which is exactly 1min long (sometimes 59secs) without any motion happening. 
When i walk past the camera, the motion detection works fine and uploads a video for the length of the motion. (sometimes 23 secs, sometimes 4mins, whatever.) This does work really good. But why do i get 1min long clips out of nowhere?

roleoroleo, 2024-06-01:
Is the clip near another mp4 file?
I mean: if the motion is still active the cam records a series of mp4 files.
The first one starts from the beginning of the event until the end of the minute, the second one starts from the following minute and lasts 60 seconds.
For example: the event is at 14:22:37
The first file will be 22M37S23.mp4
The second file will be 23M00S60.mp4

Is this your case?

DivineJimmi, 2024-06-01:
Yes, this is exactly my case. Never realized it before.

DivineJimmi, 2024-06-01:
So theres probably no way to stop this, since its supposed to do that?

I also have 2 other problems i want to address:

I am uploading my files to GDrive. Is there any way to rename the folder structure?
I live in germany, so "2024Y06M01D13H" is not so neat for me to look at. Something like "01.06.2024 13H" or something similar is better for me.

Also is my time not working properly. Its always 2 hours behind my real time. When it is 16:00 in germany, the file shows 14:00, while the overlay in the video shows 16:00.
I already changed my timezone to "CET-1CEST,M3.5.0,M10.5.0/3" and my NTP server to "de.pool.ntp.org" in the configurations page. Nothing seems to work. What did i do wrong?

roleoroleo, 2024-06-01:
To change the name structure you need to edit the script and change it for your needs.
There isn't a ready to use method.
About the name of the files, unfortunately the cam works in gmt and it's not possible to change this behavior.
You could try to translate the time in the same script converting the date in time_t (seconds from 01-01-70), adding your offset and converting back to a human readable date.
I already (partially) wrote a code for this: check the function translateFULLFNWithTz:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/db31ed8bb74eb022140817eb881aac1b4fcbe7e0/src/static/static/yi-hack/script/ftppush.sh#L152

closed 2024-09-30 by github-actions[bot]