roleoroleo/yi-hack-Allwinner-v2 #903: Yi 1080p Home SFUS 9.053.00_202107131614

Jorge0518, 2024-07-05:
Hello,

I am attempting to get this hack working on this camera. I got this camera from Cove Security a while back and was not 100% sure of the model. The motherboard says Y291GA_MB_V1.0, so I attempted to use the "Yi 1080p Home | QFUS-RRFUS | 9.0.35* | y291ga" hack, since it's the closest firmware I could find from all of the available YI-hacks. It did not work. Is there anything I can provide to get a hack for this firmware version going? Any further information would be greatly appreciated. I have attached some pictures and a firmware backup obtained from the SD card.

[image]
[image]
[image]
[backup.zip](https://github.com/user-attachments/files/16103491/backup.zip)

roleoroleo, 2024-07-13:
Based on fw version (9.0.53), your cam is not currently supported.
I will check your backup file to understand if I can support it.

roleoroleo, 2024-07-13:
Please, try this beta:
[y211ba_0.3.2.tgz](https://github.com/user-attachments/files/16202755/y211ba_0.3.2.tgz)

Check if stream and snapshots work.

Jorge0518, 2024-07-13:
Sweet! I was able to get high- and low-res snapshots, but no video or audio streams. I tried the Alternative RTSP option as well, and it did not work. Is there any network configuration I need to do on my end? thank you!

roleoroleo, 2024-07-13:
Good news!
I will add the support with the next release.

VirtualEnder, 2024-07-18:
I have the same SFUS camera, but a later version of the firmware. The file you uploaded above doesn't seem to work. I'm on 12.2.0.2_202304071306. Here is the backup file from mine. 
[backup.zip](https://github.com/user-attachments/files/16289775/backup.zip)

roleoroleo, 2024-07-23:
@VirtualEnder 
Your cam is a y20ga.
Check allwinner project.

roleoroleo, 2024-09-20:
@Jorge0518
Could you send me a copy of the video buffer /dev/shm/fshare_frame_buf ?
I need to check the h264 frame.

Jorge0518, 2024-09-25:
@roleoroleo Here you go! I don't know much of anything, but what's on it? How does it help? And how do I see what's in it? :)
[fshare_frame_buf.zip](https://github.com/user-attachments/files/17139447/fshare_frame_buf.zip)

roleoroleo, 2024-10-04:
This file it's a shared memory object used Yi processes to exchange video data.
It contains video and audio frames inside a custom structure.
Please, try these files:
[rRTSPServer.gz](https://github.com/user-attachments/files/17256991/rRTSPServer.gz)
[h264grabber.gz](https://github.com/user-attachments/files/17257009/h264grabber.gz)
[imggrabber.gz](https://github.com/user-attachments/files/17257037/imggrabber.gz)

closed 2025-02-03 by github-actions[bot]