roleoroleo/yi-hack-Allwinner-v2 #904: Camera stops every ~2 days | Yi Dome Camera U (2K) - SFUS

LaQuay, 2024-07-14:
Hello,

First of all thanks for the efforts on this project. I have a Yi Dome Camera U (2K) - SFUS and I flashed it with the `h51ga_0.3.2.tgz`. 

There are two main wrong behaviors on my camera and I don't know what can happen.

1. Every X minutes the camera resets. It goes back to the default position and starts again.
2. Every X day, the camera stops working. I plug it on again and then it works for another 3 to 5 days.

The only use I do is via the default app + the RTSP video to watch it on my computer. I can see the recordings in the SD card and it has still some GBs of space.

Any hint?

Thanks!

LaQuay, 2024-07-21:
Hi again, since last time, case 2 has not happened but case 1 happens more than once per day. 

Here is the dump of the `hack_debug.log`. Any hint? Thanks!

```Starting system.sh
Make /etc writable
Activating swap file
Set swappiness to 15
Configuring cloudAPI
Starting yi processes

  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  454 root         0 SW   [irq/302-sunxi-m]
  456 root         0 SW   [irq/166-sdc0 cd]
  457 root         0 SW   [irq/303-sunxi-m]
  461 root         0 SW   [kworker/0:2]
  528 root         0 SW<  [kworker/0:1H]
  529 root         0 SW   [kworker/0:3]
  531 root         0 SW<  [bioset]
  532 root         0 SW   [mmcqd/0]
  548 root         0 SWN  [jffs2_gcd_mtd4]
  585 root       980 S    /usr/sbin/telnetd
  595 root         0 SW   [ksdioirqd/mmc1]
  602 root         0 SW   [RTW_XMIT_THREAD]
  603 root         0 SW   [RTW_CMD_THREAD]
  604 root         0 SW   [RTWHALXT]
  616 root      1260 S    ./dispatch
  620 root       988 S    sh /tmp/sd/yi-hack/script/system.sh
  624 root         0 SW   [kworker/u2:2]
 1049 root         0 Z    [init]
 1314 root       980 R    ps

              total        used        free      shared  buff/cache   available
Mem:          60904       10512       44276          68        6116       49284
Swap:         65528           0       65528

Yi processes started successfully
[... 134 lines cut ...]
 1463 root      1528 S    ./mp4record
 1464 root      1216 S    ./cloud
 1465 root      2280 S    ./p2p_tnp
 1466 root      2644 S    ./oss
 1468 root       992 S    sh /tmp/sd/yi-hack/script/system.sh
 1470 root       980 S    sleep 30
 1486 root       964 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1564 root       844 S    pure-ftpd (SERVER)
 1576 root       892 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sb
 1636 root       880 S    mqttv4
 1789 root       964 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/u
 1865 root      3692 S<   rRTSPServer -m h51ga -r high -a yes -p 554
 1873 root      1032 S    {service.sh} /bin/sh /tmp/sd/yi-hack/script/service.
 1877 root       980 S    sleep 30
 1951 root       676 S    ipc2file
 1955 root       792 S    onvif_notify_server --conf_file /tmp/onvif_simple_se
 2025 root       688 S    wsd_simple_server --pid_file /var/run/wsd_simple_ser
 2047 root       964 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 2240 root         0 Z    [init]
 2359 root       980 R    ps

              total        used        free      shared  buff/cache   available
Mem:          60904       50832         648         704        9424        8328
Swap:         65528          40       65488
```

roleoroleo, 2024-07-23:
When the cam reboots often, it's probably a memory issue.
The cam runs with very low memory free and sometimes it hangs.
The kernel kills the main process (rmm) and the watchdog restarts the cam.

1 - The problem could be the snapshot feature: it requires a lot of memory. Disable it or set it to low resolution.
2 - Disable all unnecessary processes: mqtt, onvif, etc...

If this will not fix the problem, check power and cable. Are you using the stock power supply?

LaQuay, 2024-08-15:
Hi, I have done several tests, including changing the power supply, and this still happens. Currently, I have MQTT off, snapshot in force low resolution, FTP off, telnet off... Any other test I can do? Thanks!

roleoroleo, 2024-08-15:
Try to kill all wahtcdog applications and let the cam run:
```
killall wd_rtsp.sh
killall watch_process
```

If/when the cam stops working, login with ssh and check the process list:
`ps`

LaQuay, 2024-08-15:
Done @roleoroleo , will let you know. Thanks!

LaQuay, 2024-09-15:
Hello @roleoroleo; It has been 23 days without a reboot after I killed the watchdogs. Thanks!

roleoroleo, 2024-09-15:
This is what i expected, but I'm surprised that the cam works properly.
If the watchdog restarts the cam, this is because rmm process dies.

closed 2025-01-15 by github-actions[bot]