roleoroleo/yi-hack-Allwinner-v2 #905: Camera is accessible via ssh but app is failing

Nickleby9, 2024-07-15:
Hi,
A few days ago my camera wasn't available anymore.
When I try to access it it looks like this, with the following error on the browser console:
`Error: <path> attribute d: Expected number, "…2.097898,1.5862amundomario, role…".`
[image]

I still have ssh access, so I tried looking at dmesg logs and this is what I got:
```
[    0.558622] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.559456] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.560292] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.560334] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm OD pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    0.561258] sunxi-mmc sdc1: smc 1 p1 err, cmd 1, RTO !!
[    0.561288] sunxi-mmc sdc1: sdc set ios:clk 0Hz bm PP pm OFF vdd 0 width 1 timing LEGACY(SDR12) dt B
[    0.640374] sunxi-codec-machine sndcodec: [sunxi_card_dev_probe] register card finished.
[    0.640993] NET: Registered protocol family 17
[    0.641042] NET: Registered protocol family 15
[    0.641072] Key type dns_resolver registered
[    0.641667] Registering SWP/SWPB emulation handler
[    0.646981] sunxi-rtc rtc: setting system clock to 2024-07-15 22:46:00 UTC (1721083560)
[    0.647816] snddaudio snddaudio0: codec: snd-soc-dummy, codec_dai: snd-soc-dummy-dai.
[    0.648350] snddaudio snddaudio0: snd-soc-dummy-dai <-> daudio0 mapping ok
[    0.649647] ALSA device list:
[    0.649656]   #0: sun8iw19-codec
[    0.649659]   #1: snddaudio0
[    0.649716] Warning: unable to open an initial console.
[    0.654072] VFS: Mounted root (squashfs filesystem) readonly on device 31:2.
[    0.655696] devtmpfs: mounted
[    0.655934] Freeing unused kernel memory: 108K
[    0.655938] This architecture does not have kernel memory protection.
[    0.672107] random: fast init done
[    0.693970] sunxi-mmc sdc0: sdc set ios:clk 0Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    0.694023] sunxi-mmc sdc0: Switching to 1.8V signalling voltage ok
[    0.712051] sunxi-mmc sdc0: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    0.719726] mmc0: host does not support reading read-only switch, assuming write-enable
[    0.720435] sunxi-mmc sdc0: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 4 timing LEGACY(SDR12) dt B
[    0.721263] sunxi-mmc sdc0: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 4 timing UHS-SDR104 dt B
[    0.721348] sunxi-mmc sdc0: sdc set ios:clk 150000000Hz bm PP pm ON vdd 21 width 4 timing UHS-SDR104 dt B
[    0.721492] mmc0: new ultra high speed SDR104 SDHC card at address 1388
[    0.722897] mmcblk0: mmc0:1388 USD00 14.7 GiB
[    0.724792]  mmcblk0: p1
[    1.179488] udevd[535]: could not create /run/udev: No such file or directory
[    3.106481] sunxi-wlan wlan: bus_index: 1
[    3.226115] sunxi-mmc sdc1: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 timing LEGACY(SDR12) dt B
[    3.244017] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    3.264710] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    3.267205] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    3.269088] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
[    3.272346] sunxi-mmc sdc1: card claims to support voltages below defined range
[    3.283552] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
[    3.283674] sunxi-mmc sdc1: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
[    3.283834] sunxi-mmc sdc1: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 4 timing SD-HS(SDR25) dt B
[    3.284607] mmc1: new high speed SDIO card at address 0001
[    3.298770] random: crng init done
[    3.444617] [sc3335_mipi]V4L2_IDENT_SENSOR = 0xcc1a
[    3.457021] sunxi_gpadc_init,1779, success
[    3.457829] input: sunxi-gpadc0 as /devices/virtual/input/input0
[    3.457891] sunxi_gpadc_probe success
[    5.313401] libphy: 5020000.eth: probed
[    5.313810] sunxi-gmac 5020000.eth eth0: eth0: Type(6) PHY ID 0000ffff at 31 IRQ poll (5020000.eth-0:1f)
[    5.315821] sunxi-gmac 5020000.eth eth0: Initialize hardware error
[    5.703674] cpld_periph: loading out-of-tree module taints kernel.
[    5.709632]
[    5.709632] [CPLD_PERIPH] ir_led_ctl: 0, ircut_direction: 0 audio_ctl:0
[    5.709833]
[    5.709833] [CPLD_PERIPH] cpld_init ok, [ ver=Aug 16 2021, 10:46:41 ]
[    5.709901] can't request wifi_en gpio[198]
[    5.710265] [CPLD_PERIPH] pwm init ok.
[    5.710273] [CPLD_PERIPH] timer init ok.
[    5.710277] timer resolution:4 MHZ
[    5.710280] [CPLD_PERIPH] HZ = 250
[    7.146822] hw8_xy:[0],hw12_ab_cd_ratio[0],hw23_26_x_step:[0],hw27_30_y_step:[0]
[    7.477738] Adding 65532k swap on /tmp/sd/swapfile.  Priority:-1 extents:20 across:66684k SS
[    9.255462] [CPLD_PERIPH] Timer closed.
[    9.255474] [CPLD_PERIPH] Timer opened.
[   11.322889] cedar_ve: VE real_fre=300000000
[   11.597112] snddaudio snddaudio0: codec_dai set sysclk failed
[   11.597123] snddaudio snddaudio0: codec_dai set set_pll failed.
[   11.597129] snddaudio snddaudio0: codec dai set fmt failed
[   11.597148] snddaudio snddaudio0: codec_dai set clkdiv failed
[   12.071981] -----> wait adc hw ready
[   12.120795] snddaudio snddaudio0: codec_dai set sysclk failed
[   12.120805] snddaudio snddaudio0: codec_dai set set_pll failed.
[   12.120810] snddaudio snddaudio0: codec dai set fmt failed
[   12.120829] snddaudio snddaudio0: codec_dai set clkdiv failed
[   12.191300] [sc3335_mipi]V4L2_IDENT_SENSOR = 0xcc1a
[   12.288111] into set sensor hfilp the value:1
[   12.288657] [VIN]sensor_flip_option:isp0 reset!!!
[   12.288710] into set sensor hfilp the value:1
[   12.289244] [VIN]sensor_flip_option:isp0 reset!!!
[   12.290152] cedar_ve: VE real_fre=300000000
[   12.290225] cedar_ve: VE real_fre=300000000
[   12.292928] cedar_ve: VE real_fre=300000000
[   18.432185] [CPLD_PERIPH] Timer closed.
[   19.540012] -----> wait adc hw ready
[   19.588912] snddaudio snddaudio0: codec_dai set sysclk failed
[   19.588940] snddaudio snddaudio0: codec_dai set set_pll failed.
[   19.588946] snddaudio snddaudio0: codec dai set fmt failed
[   19.588966] snddaudio snddaudio0: codec_dai set clkdiv failed
[   19.593162] [sc3335_mipi]V4L2_IDENT_SENSOR = 0xcc1a
[   19.633215] [sc3335_mipi]V4L2_IDENT_SENSOR = 0xcc1a
[   19.665329] cedar_ve: VE real_fre=300000000
[   19.756452] cedar_ve: VE real_fre=300000000
[   20.005739] cedar_ve: VE real_fre=300000000
[   45.912825]
[   45.912825] [PID_LIST] pid_list_init ok, [ ver=Aug 16 2021, 10:46:40 ]
```

I thought about running unbrick but in my backup file I can only see .conf files, not the .bin files mentioned in the wiki.
Is there a way to solve it without unbrick?
If not, is there a way to perform a backup that includes the bin files using ssh?

Thanks a lot!

Nickleby9, 2024-07-19:
I have formatted the sd card and reinstalled the hack and everything is back to normal.
I'm still wondering how to get the bin files described in the unbrick guide.

Nickleby9, 2024-07-19:
Solved by formatting the sd card and reinstall the hack

closed 2024-07-19 by Nickleby9

roleoroleo, 2024-07-23:
Good to know.