roleoroleo/yi-hack-Allwinner-v2 #907: Backchannel

kiddyfurby, 2024-07-23:
Was try to get backchannel working with go2rtc, noticed that the camera doesn't advertise that capability.
```
{
  "producers": [
    {
      "type": "RTSP active producer",
      "url": "rtsp://192.168.19.24/ch0_0.h264/",
      "remote_addr": "192.168.19.24:554",
      "user_agent": "go2rtc/1.9.2",
      "sdp": "...",
      "medias": [
        "video, recvonly, H.264 High 3.2",
        "audio, recvonly, MPEG4-GENERIC/16000"
      ],
      "receivers": [
        "96 H264, bytes=51545563, senders=1",
        "97 MPEG4-GENERIC/16000, bytes=2645133, senders=1"
      ],
      "recv": 54874236
    }
  ],
```

My other camera which has backchannel support has `audio, sendonly, PCMU/8000` in producers.medias

roleoroleo, 2024-07-23:
There is a type with G711 that I need to fix.
Try with AAC and let me know if it works.

kiddyfurby, 2024-07-24:
It's already set to AAC.

roleoroleo, 2024-07-25:
This is my output:
```
{
  "producers": [
    {
      "id": 2,
      "format_name": "rtsp",
      "protocol": "rtsp+tcp",
      "remote_addr": "192.168.0.245:554",
      "url": "rtsp://192.168.0.245/ch0_0.h264",
      "sdp": "...",
      "user_agent": "go2rtc/1.9.4",
      "medias": [
        "video, recvonly, H264",
        "audio, recvonly, PCMU/8000",
        "audio, sendonly, MPEG4-GENERIC/16000"
      ],
...
```

Check the configuration.

kiddyfurby, 2024-07-25:
Yes I am on 0.3.2
```
Firmware Version | 0.3.2
Base Version | 9.0.09.06_202103291039
Model Suffix | y29ga

```
I noticed that you're on go2rtc 1.9.4 so I upgraded just to make sure, now my json is like this:
```
{
  "producers": [
    {
      "id": 43,
      "format_name": "rtsp",
      "protocol": "rtsp+tcp",
      "remote_addr": "192.168.19.24:554",
      "url": "rtsp://192.168.19.24/ch0_0.h264",
      "sdp": "...",
      "user_agent": "go2rtc/1.9.4",
      "medias": [
        "video, recvonly, H264",
        "audio, recvonly, MPEG4-GENERIC/16000",
        "audio, recvonly, MPEG4-GENERIC/16000"
      ],
```
I have non-alt rtsp server, onvif on, rtsp audio set to aac, onvif discovery on, onvif high, backchannel aac

roleoroleo, 2024-07-26:
Better, but the 3rd media should be sendonly.

Do you have another cam that works with go2rtc?
I need to check the messages between the cam and the software.
Check this issue too:
https://github.com/roleoroleo/onvif_simple_server/issues/10#issuecomment-2249893385

kiddyfurby, 2024-07-26:
my other camera (not a allwinner-v2) that has a backchannel looks like this:
```
{
  "producers": [
    {
      "id": 5822,
      "format_name": "rtsp",
      "protocol": "rtsp+tcp",
      "remote_addr": "192.168.1.42:554",
      "url": "rtsp://192.168.1.42/Preview_01_sub",
      "sdp": "...",
      "user_agent": "go2rtc/1.9.4",
      "medias": [
        "video, recvonly, H264",
        "audio, recvonly, MPEG4-GENERIC/16000",
        "audio, sendonly, PCMU/8000"
      ]
```

roleoroleo, 2024-07-30:
Apart from the codec, it's the same ouput I receive from my rtsp.

closed 2024-11-28 by github-actions[bot]

0xBURP, 2024-11-28:
Good news. I'd love to report that with yi-hack-Allwinner-v2 0.3.3, backchannel works!
I paired it with go2rtc (1.9.4) backchannel support and sound picked up on my laptop mic would come out from the yi-hack speaker!

roleoroleo, 2024-11-30:
Could you post your configuration for other users?
Thanks.

0xBURP, 2024-12-01:
Of course!

In yi-hack-Allwinner-v2 0.3.3

1. Configurations / RTSP server = standard
2. Configurations / ONVIF / ONVIF audio decoder (backchannel) = G711
3. Reboot to apply the settings

In go2rtc 1.9.4, goto :1984 web ui, click links on your yi-hack-Allwinner-v2 0.3.3 camera

1. Scroll the the bottom and find WebRTC Magic section
2. Select video+audio+microphone = two way audio from camera
3. Click webrtc.html local WebRTC viewer

https connection to go2rtc is required.

kiddyfurby, 2024-12-01:
Of course!

In yi-hack-Allwinner-v2 0.3.3

Configurations / RTSP server = standard
Configurations / ONVIF / ONVIF audio decoder (backchannel) = G711
Reboot to apply the settings
In go2rtc 1.9.4, goto :1984 web ui, click links on your yi-hack-Allwinner-v2 0.3.3 camera

Scroll the the bottom and find WebRTC Magic section
Select video+audio+microphone = two way audio from camera
Click webrtc.html local WebRTC viewer
https connection to go2rtc is required.

roleoroleo, 2024-12-11:
Sorry, I mean go2rtc configuratrion.

kiddyfurby, 2024-12-11:
Oh, on the go2rtc side, no specific config is required for backchannel other than a RTSP camera.
The yi-hack side config are more important as AAC didn't work
```
streams:
  living:
  - rtsp://192.168.19.25/ch0_0.h264
```