roleoroleo/yi-hack-Allwinner-v2 #908: YI Dome Cam U (3K)

matzingerhh, 2024-07-24:
Hi all I hope this is the right category. I use this project for my YI cams. 
With my newest one I have an issue regarding the video streaming. 

I bought YI Dome Cam U (3k)
Serial : SFUSY4.....
Firmware 09.21.13...

I missed the 3k and took the h51ga
archive from the release and put it into the cam.

So far the hack worked. I saw the hack website and had control. While integrating the cam into frigate realized that there is no video / audio / snapshot coming from the cam. Also from hack website I am not able to get into video. 

My question is the h51ga file wrong or is there currently no working hack for my cam? 

Many thanks and kind regards
Mathias

roleoroleo, 2024-07-25:
Tell me the exact fw revision.
9....?
Maybe it's a different model.

matzingerhh, 2024-07-25:
[image]

Hi thank for having a look. 
Many thanks and kind regards

matzingerhh, 2024-07-30:
Hi I have an update. I retried the hack again with an reseted cam and new sd card. 
And now it seems that I can use the high res stream in VLC. Snapshot seems to be ok too. Frigate shows some strange errors but I will go deeper with that. 
I don't know why the first try fails but for now I would say also the dome u 3k is supported by h51ga. 
Kind regards from Germany

closed 2024-07-30 by matzingerhh