roleoroleo/yi-hack-Allwinner-v2 #909: Yi Dome Guard YRS.3019 RTSP broken on yi-hack-Allwinner-v2 0.3.2

maxromanovsky, 2024-07-26:
As others experienced, RTSP broken on yi-hack-Allwinner-v2 0.3.2 for Yi Dome Guard YRS.3019.
In my case, VLC can't connect and display the stream.

Firmware Version 0.3.2
Base Version 12.1.05.1_202302091435

As @Solarbotics [pointed out](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/484#issuecomment-1818245542), editing `<sd>/yi-hack/model_suffix` file contents to `r35gb` resolves the issue.

Maybe it's an issue with `Base Version 12.1.05.1_202302091435`?
I don't know.

I decided to create a separate issue out of https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/484 because
- that issue is an uber-issue regarding support for Yi Dome Guard YRS.3019
- that issue has been closed for quite some time

roleoroleo, 2024-07-30:
I double checkde the fw versione and 12.1.05.1_202302091435 refers to r30gb.
It could be a problem with the codec.
Could you please send me a copy of the frame buffer /dev/shm/fshare_frame_buf ?

About onvif, between 0.2.6 and 0.3.2 I completely changed the daemon. But I already tested Synology Surveillance Station with this new software and the test was successful.
Did you try to enable/disable the advanced options you mention above?
If it doesn't work, I need a network traffic between the cam and the Synology to check the messages.

roleoroleo, 2024-07-31:
Tested again now with DSM 7 and Surveillance 9.2.0-11289.
It works.
Please, explain me what's the problem.

maxromanovsky, 2024-07-31:
@roleoroleo I just went on a trip, and will check it upon return in about a week or so.

maxromanovsky, 2024-08-18:
@roleoroleo sorry it took me longer.

Here's what I did:
- Changed camera back from h264 to h265 (`printf "\02\00\00\00" | dd of=/dev/mtdblock7 bs=1 seek=780 count=4 conv=notrunc`) - as you wrote here: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/484#issuecomment-1235411673
- Took out SD card with hack v0.2.6
- Updated via app from 9.0.05.33_202009251856 to 12.1.05.1_202302091435
- Made a stock firmware backup: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)
- Installed r30gb_0.3.2 (pre-configured wifi)
- Kept everything else as is (default config), haven't modified `mtdblock7` for h264
- Video plays via YI home mobile app
- tried to connect via VLC on macOS (`rtsp://x.x.x.x/ch0_0.h264`) - `Connection failed
VLC could not connect to "x.x.x.x:554".`
- tried `nc x.x.x.x:554`: `nc: connectx to x.x.x.x port 554 (tcp) failed: Connection refused`
- `scp root@192.168.5.10:/dev/shm/fshare_frame_buf .` - attached 
[fshare_frame_buf.zip](https://github.com/user-attachments/files/16649751/fshare_frame_buf.zip)

- I can't proceed further

It seems rtsp is NOT working:

```sh
root@(none):/tmp/sd/log# ps | grep rtsp
 1923 root       992 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
10434 root       992 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
10441 root       980 S    grep rtsp
```

I've also enabled the rtsp log by uncommenting `LOG_FILE="/tmp/sd/wd_rtsp.log"` and commenting `LOG_FILE="/dev/null"` in `/yi-hack/script/wd_rtsp.sh`

```console
root@(none):/tmp/sd# cat wd_rtsp.log
2024-08-18 15:46:03 - Starting RTSP watchdog...
2024-08-18 15:46:04 - Restarting rtsp process
2024-08-18 15:46:17 - Restarting rtsp process
2024-08-18 15:46:31 - Restarting rtsp process
2024-08-18 15:46:44 - Restarting rtsp process
2024-08-18 15:46:57 - Restarting rtsp process
```

roleoroleo, 2024-08-19:
The frame buffer is ok.
Frames as expected: h264 supported by rRTSPServer.
Please, run the server with the full log:
```
killall wd_rtsp.sh
killall rRTSPServer
rRTSPServer xxx -d 15
```
where 'xxx' are your current options.

roleoroleo, 2024-08-20:
Yes, the buffer contains 2 different h264 streams: low and high res.
You can check the file looking for SPS frames (they starts with the hex sequence 00 00 00 01 67).

The log is not ok, the parser is unable to find the frames and the capture thread does not start. I don't know why, I have to check the code. But I'm afk for a while.

maxromanovsky, 2024-08-21:
I wonder why the firmware changed from h265 to h264...
No big deal, let's keep the PR open, and I will make a gentle reminder in a month or so :)
In the meantime I will also be away every other week

roleoroleo, 2024-09-07:
The length of the header is changed.
In the previous fw version was 22, your version is 26.
I have to change the parser.

roleoroleo, 2024-09-07:
But...
Are you sure it's a R30GB?
Could you send me a backup of the flash?

maxromanovsky, 2024-09-08:
@roleoroleo backup attached. I edited mtdblock7.bin and replaced wifi ssid & pwd with `x`

I am traveling, so can't take a photo of the device itself, but I'm pretty sure it's R30GB.

[Archive.zip](https://github.com/user-attachments/files/16921534/Archive.zip)

roleoroleo, 2024-09-08:
Please, try this binary:
[rRTSPServer.gz](https://github.com/user-attachments/files/16922514/rRTSPServer.gz)

maxromanovsky, 2024-09-09:
@roleoroleo than you! I will do my best to try it this Sunday. I am traveling now.

maxromanovsky, 2024-09-16:
@roleoroleo just tested. Still doesn't work (not listening on port):
```console
$ scp rRTSPServer root@192.168.5.10:/tmp/sd/yi-hack/bin/
$ ssh root@192.168.5.10

BusyBox v1.27.2 () built-in shell (ash)

mount: mounting none on /sys/kernel/debug failed: Resource busy

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------
root@(none):~# cd /tmp/sd/yi-hack/bin/
root@(none):~/bin# chmod a+x rRTSPServer
root@(none):~/bin# rRTSPServer xxx -d 15
1723999432547: the size of the buffer is 1786156
unable to find /tmp/audio_fifo, audio disabled
1723999432549: capture - mapping file fshare_frame_buf, size 1786156, to b6c40000
1723999432550: capture - closing the file fshare_frame_buf
1723999432551: capture - starting capture main loop
1723999432551: capture - frame header size = 28
1723999432552: capture - buf_idx_end == buf_idx_end_prev
1723999432562: capture - buf_idx_end == buf_idx_end_prev
1723999432573: capture - buf_idx_end == buf_idx_end_prev
1723999432583: capture - buf_idx_end == buf_idx_end_prev
1723999432593: capture - buf_idx_end == buf_idx_end_prev
1723999432604: capture - buf_idx_end == buf_idx_end_prev
1723999432614: capture - buf_idx_end == buf_idx_end_prev
1723999432624: capture - buf_idx_end == buf_idx_end_prev
1723999432634: capture - buf_idx_end == buf_idx_end_prev
```

maxromanovsky, 2024-09-18:
I updated previous comment, as I used a wrong command arguments on Monday.

roleoroleo, 2024-09-18:
Maybe the buffer is stopped.
Run 3 times this command:
`hexdump -C -n 32 /dev/shm/fshare_frame_buf`

maxromanovsky, 2024-09-19:
@roleoroleo here's the output (right after starting the camera):
```console
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000020
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000020
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000020
root@(none):~# killall wd_rtsp.sh
root@(none):~# killall rRTSPServer
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000020
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000020
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000020
```

roleoroleo, 2024-09-20:
This is the problem: the buffer is not filled...
This is strange because the buffer you posted above was ok.

If I remeber correctly it's a race condition in the startup script.
You should try to increase the sleep after the rmm command in the file /tmp/sd/yi-hack/script/system.sh:
Pay attention: there are 2 rmm lines: 1 for cloud enabled and 1 for cloud disabled.

maxromanovsky, 2024-09-21:
@roleoroleo I have tried to increase `sleep` in 2 places (via ssh)

https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/0.3.2/src/static/static/yi-hack/script/system.sh#L203 (lines `203` & `244`)

From `sleep 6` to `sleep 20`, and then to `sleep 120`.

However, after reboot result is the same.

roleoroleo, 2024-09-21:
Ok, restore this sleep and try to increase the sleep before rmm.

maxromanovsky, 2024-09-22:
@roleoroleo increased sleep From sleep 2 to sleep 20 before rrm.
Still the same result.
Thinking out loud, are there any logs from `/tmp/sd/yi-hack/script/system.sh` STDOUT/STDERR that I can share?

roleoroleo, 2024-09-24:
Yes you can enable debug
[image]
but probably is not useful.
Try to run
`ipc_cmd -x`
and check if the buffer starts.

maxromanovsky, 2024-09-24:
@roleoroleo buffer started!
```console
root@(none):~# ipc_cmd -x
root@(none):~#
root@(none):~#
root@(none):~#
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 2c 25 1b 00  0a 98 00 00 f9 2d 00 00  |....,%.......-..|
00000010  cd 48 00 00 0a 5a c9 89  64 03 00 00 00 00 00 00  |.H...Z..d.......|
00000020
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 3e 15 1b 00  0a 98 00 00 78 9a 08 00  |....>.......x...|
00000010  3a c5 08 00 0f 6a c9 89  6b 04 00 00 00 00 00 00  |:....j..k.......|
00000020
root@(none):~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 2d 38 1b 00  0a 98 00 00 ba 53 1a 00  |....-8.......S..|
00000010  8d 5b 1a 00 57 7f c9 89  c7 05 00 00 00 00 00 00  |.[..W...........|
00000020
```

And media info from VLC:
[image]

maxromanovsky, 2024-09-29:
@roleoroleo plz let me know what are the next steps once you have time.

roleoroleo, 2024-09-29:
Il will add the ipc_cmd command to system.sh.
And I will share a new version to try.

maxromanovsky, 2024-10-06:
@roleoroleo gentle reminder. please share a new script version when you have some time. Thanks!

roleoroleo, 2024-10-08:
Here it is:
[system.sh.gz](https://github.com/user-attachments/files/17289132/system.sh.gz)
Overwrite system.sh

maxromanovsky, 2024-10-09:
@roleoroleo I have tried, bit these changes didn't work (after copy & reboot).

`system.sh` has execute permissions.

And comparing with the latest release, here's the diff:
```console
diff yi/r30gb_0.3.2/yi-hack/script/system.sh system.sh
368a369,383
> MDNSD_ENABLED=0
> if [[ $(get_config HTTPD) == "yes" ]] ; then
>     mkdir -p /tmp/mdns.d
>     echo -e "type _http._tcp\nport $HTTPD_PORT\n" > /tmp/mdns.d/http.service
>     MDNSD_ENABLED=1
> fi
> if [[ $(get_config SSHD) == "yes" ]] ; then
>     mkdir -p /tmp/mdns.d
>     echo -e "type _ssh._tcp\nport 22\n" > /tmp/mdns.d/ssh.service
>     MDNSD_ENABLED=1
> fi
> if [[ "$MDNSD_ENABLED" == "1" ]] ; then
>     /tmp/sd/yi-hack/sbin/mdnsd /tmp/mdns.d
> fi
>
```

Here's what I did

```console
$ md5sum system.sh
ab290f41cffad02468b4aba7aa8a1a0b  system.sh
$ scp system.sh root@192.168.5.10:/tmp/sd/yi-hack/script/
system.sh                                                                                                                                                                                                  100%   15KB 310.9KB/s   00:00
$ ssh root@192.168.5.10

BusyBox v1.27.2 () built-in shell (ash)

mount: mounting none on /sys/kernel/debug failed: Resource busy

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@(none):~# ls -l /tmp/sd/yi-hack/script/
drwxr-xr-x    2 root     root         32768 Apr 11 18:35 blacklist
-rwxr-xr-x    1 root     root          2931 Apr 11 18:35 check_conf.sh
-rwxr-xr-x    1 root     root          1328 Apr 11 18:35 clean_records.sh
-rwxr-xr-x    1 root     root           633 Apr 11 18:35 conf2mqtt.sh
-rwxr-xr-x    1 root     root          1908 Apr 11 18:35 configure_wifi.sh
-rwxr-xr-x    1 root     root          1161 Apr 11 18:35 create_avi.sh
-rwxr-xr-x    1 root     root           551 Apr 11 18:35 env.sh
-rwxr-xr-x    1 root     root          1271 Apr 11 18:35 ethdhcp.sh
-rwxr-xr-x    1 root     root          8695 Apr 11 18:35 ftppush.sh
drwxr-xr-x    2 root     root         32768 Apr 11 18:35 mqtt_advertise
-rwxr-xr-x    1 root     root          3231 Apr 11 18:35 ptz_presets.sh
-rwxr-xr-x    1 root     root         16405 Apr 11 18:35 service.sh
-rwxr-xr-x    1 root     root         15811 Aug 18 15:46 system.sh
-rwxr-xr-x    1 root     root          5538 Apr 11 18:35 thumb.sh
-rwxr-xr-x    1 root     root          3768 Apr 11 18:35 time_lapse.sh
-rwxr-xr-x    1 root     root          4181 Aug 18 15:44 wd_rtsp.sh
-rwxr-xr-x    1 root     root          1273 Apr 11 18:35 wifidhcp.sh
root@(none):~# md5sum /tmp/sd/yi-hack/script/system.sh
ab290f41cffad02468b4aba7aa8a1a0b  /tmp/sd/yi-hack/script/system.sh
root@(none):~# reboot now
root@(none):~#
root@(none):~# Timeout, server 192.168.5.10 not responding.
```

roleoroleo, 2024-10-11:
Strange, you should find these lines at line 246:
```
        # Trick to start circular buffer filling
        start_buffer
```

maxromanovsky, 2024-10-12:
@roleoroleo just to be clear, should I just use the script from v0.3.2 and add these 2 lines?

```sh
        # Trick to start circular buffer filling
        start_buffer
```

Or do I need any of these changes?

```diff
diff yi/r30gb_0.3.2/yi-hack/script/system.sh system.sh
368a369,383
> MDNSD_ENABLED=0
> if [[ $(get_config HTTPD) == "yes" ]] ; then
>     mkdir -p /tmp/mdns.d
>     echo -e "type _http._tcp\nport $HTTPD_PORT\n" > /tmp/mdns.d/http.service
>     MDNSD_ENABLED=1
> fi
> if [[ $(get_config SSHD) == "yes" ]] ; then
>     mkdir -p /tmp/mdns.d
>     echo -e "type _ssh._tcp\nport 22\n" > /tmp/mdns.d/ssh.service
>     MDNSD_ENABLED=1
> fi
> if [[ "$MDNSD_ENABLED" == "1" ]] ; then
>     /tmp/sd/yi-hack/sbin/mdnsd /tmp/mdns.d
> fi
>
```

roleoroleo, 2024-10-12:
I downloaded the file above and it contains the new lines.
And the date of your system.sh is not equal to the date of the file I posted.
Please check again.

maxromanovsky, 2024-10-13:
@roleoroleo I just downloaded again, and it's the same file.
Moreover, these 2 lines you're pointing out are already in the release `0.3.2`:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/0.3.2/src/static/static/yi-hack/script/system.sh#L246-L247

I re-downloaded file again, uploaded it to camera, but it still doesn't work.

Please check if we're talking about the same files:
```console
$ md5sum system.sh*
ab290f41cffad02468b4aba7aa8a1a0b  system.sh
d92adf94242e7d77d9e6f8f0eaaad1e9  system.sh.gz
```

roleoroleo, 2024-10-14:
Sorry, you are right.
I removed these lines some month ago...

If the buffer doesn't start with these lines it's a race condition difficult to find.
You should try to run the command in a different position of the system.sh script.
Maybe at the end of the script.

maxromanovsky, 2024-10-15:
Okay, I will troubleshoot and let you know

maxromanovsky, 2024-11-12:
Just a heads up, it takes me longer than I expected. I'll be able to start looking at it by mid-December. Now I'm unfortunately very busy with the other things, but I didn't forget about this one ;)

maxromanovsky, 2025-01-28:
Sorry it takes me longer, lots of stuff going on. End of February would be more realistic.

maxromanovsky, 2025-03-09:
I finally got time to investigate.

- In the meantime there was a new yi firmware release `12.1.05.4_202411081549`
- There was also a new hack firmware release `0.3.3`
- Everything works fine on this combination (VLC, Synology Surveillance Station 9.2.2-11575)!

Here's my config:
- disable cloud: on
- Recording without Cloud: on
- rtsp stream: high
- rtsp audio: aac
- ONVIF configuration
    - Fault response if unknown request: off
    - Synology NVR compatibility: on
- Snapshot: off
- Snapshot for recorded video: off
- telnet: off
- enable swap: on
- set ONVIF credentials (login/password)

Hence, I'm closing this issue.
@roleoroleo thank you very much for your help!

closed 2025-03-09 by maxromanovsky