roleoroleo/yi-hack-Allwinner-v2 #912: Yi Dome Guard YRS.3521 Issues

Gillmores, 2024-08-04:
I have 4 Yi Dome Guard cameras.  They all say they are Model YRS.3521 on the bottom and look physically identical.

The first two I bought a year ago came with 12.1.47.**00** and I downgraded them to 9.0.46.00 to get the motion recording features back.  They have been working fine for a year.

The two new ones I just bought came with 12.1.47.**02** (still with the same motion detection error), so I downgraded one to the same 9.0.46.00 version as the first two, but now it refuses to boot (i.e. stuck in the reboot loop).  I can put the original 12.1.47.02 version back and it works again.  Do you know if there are different versions of "Model YRS.3521", or what else might be happening?

I "downgrade" them by just putting the *.bin and *.sh files in the 'Factory' folder on the SD card, which has always worked before...  Any ideas?

roleoroleo, 2024-08-05:
> I "downgrade" them by just putting the *.bin and *.sh files in the 'Factory' folder on the SD card, which has always worked before... 

What do you mean with this?

Gillmores, 2024-08-05:
> > I "downgrade" them by just putting the *.bin and *.sh files in the 'Factory' folder on the SD card, which has always worked before...
> 
> What do you mean with this?

I made a backup of one of my Dome Guard cameras before I let it update from 9.0.46.00 to 12.1.47.00 (I have more than just the 4 I mentioned, from several years ago).  So I have the 8 mtdblockX.bin files for version 9.0.46.00.  To "downgrade", I put those 8 .bin files in a folder called "Factory" along with a config.sh file that copies them from the sd card to /dev on the camera, and then let the camera reboot to copy the files onto the camera.  This process worked on the two cameras from a year ago, but not on the latest one.  The only difference I can see between old and new is that the newer ones came with a slightly newer firmware version (.02 vs. .00).  The model numbers are the same.

Gillmores, 2024-08-05:
If I sent you the set of mtdblockX.bin files from my 2 new cameras, could you tell me which of your mods would work with it?  Since the camera is pretty useless as it is now, I'd like to experiment with it.

roleoroleo, 2024-08-06:
Where did you get the fw 9.0.46?

roleoroleo, 2024-08-11:
There might be a mac address problem.
If you clone the whole flash, all new cams have the same mac address and, probably, they are not able to connect to the wifi ap.
I'm not sure but it could be.

closed 2024-12-10 by github-actions[bot]

Gillmores, 2024-12-12:
> There might be a mac address problem. If you clone the whole flash, all new cams have the same mac address and, probably, they are not able to connect to the wifi ap. I'm not sure but it could be.

Yes, I'm pretty sure this is the problem now.  Is there any way to edit the block that contains the MAC address to set it back to match the hardware?  I know what the MAC address of each camera is supposed to be, but I'm not sure where the bytes are stored and in which block.  I couldn't find anything that looked like a MAC address scanning the block files for ascii, so I'm guessing maybe they are stored as 6 bytes somewhere?  Do you have any idea where?

roleoroleo, 2024-12-13:
I don't know where is it.
If you look at the uboot env there 3 variables:
```
mac_addr=${mac}
wifi_mac=${wifi_mac}
bt_mac=${bt_mac}
```
For sure it's not in mtdblock6 or mtdblock7.
But, did you check if mac is really the same?