roleoroleo/yi-hack-Allwinner-v2 #916: local network IP not working but external working

gl2007, 2024-08-15:
Sometime back I was able to properly setup my Yi cameras for external access. Back then I had access to their configs via Http from my local network. Now, however, none of them are accessible via the local IP and http port number I have setup but are accessible via the my router's IP address and external port which I forward to internal port which I set up on them. I even tried changing http port to 80 (from the non-80 port) but no luck. 

Are there logs to see why they are not responding to local IP address http request? (I know local IPs don't go through router so it has to be something going wrong.)

roleoroleo, 2024-08-15:
Sorry but there are no logs to study this issue.
Try to capture and analyze the network traffic between your pc and the cam.

gl2007, 2024-08-16:
Good idea to see the traffic if it is reaching the camera. But would be good to have a configuration parameter to see logs, otherwise how to troubleshoot if the traffic is indeed reaching the camera? (I will report back in either case)

gl2007, 2024-08-26:
Issue was resolved automatically with no actions from me. Looks like my router is involved in some way that I can't figure out?

roleoroleo, 2024-09-07:
Maybe.

closed 2025-01-07 by github-actions[bot]