roleoroleo/yi-hack-Allwinner-v2 #919: Camera settings are resetted after a undefined period of time

Gifford47, 2024-08-20:
In section "camera settings" I enabled the camera as also the ai human detection and IR led.
After some time (f.e. 6hours) all the options are disabled automatically in the UI. I can reenable the options via webui or mqtt but the issue occurs continuously.

I've enabled swap, low snapshots/ stream and mqtt. No other unnecessary stuff.

What can cause this issue, any idea?

PBadz, 2024-09-01:
Exactly the same on ZFUS but in Configurations pane. All settings are rolling back to default after few reebots.

Firmware Version: 0.3.2
Base Version: 12.0.51.01_202303091901
Model Suffix: y623

roleoroleo, 2024-09-09:
Are you using the app?

Gifford47, 2024-09-09:
I don't use the app

roleoroleo, 2024-09-12:
Does the cam reboot before the settings reset?

Gifford47, 2024-09-12:
Maybe. I didn't recognise it yet. For me the reset occurs after about 24h (not exactly).

roleoroleo, 2024-09-14:
Is cloud enabled? (Disable cloud switched off)

Gifford47, 2024-09-15:
Cloud is disabled (switch is on)

Gifford47, 2024-09-19:
> Does the cam reboot before the settings reset?
> 

Now it had a reboot and resetted the settings afterwards. Might be a service that isn't started successfully?

Gifford47, 2024-10-23:
@roleoroleo any progress? Do you need any debug information?

Gifford47, 2024-11-16:
After reinstall to 0.3.3 no issue persists.

closed 2024-11-16 by Gifford47