roleoroleo/yi-hack-Allwinner-v2 #921: LFUS Downgrade Firmware

fraizzo, 2024-08-23:
Hi, i have a LFUS with 12.0.27 firmware, can i downgrade it so can i install custom firmware?

Thanks

closed 2024-08-23 by fraizzo

roleoroleo, 2024-09-07:
Your cam is not supported at the moment.
Please, send me a backup of your flash.

fraizzo, 2024-09-07:
Where can i sent you the backup?

fraizzo, 2024-09-07:
[backup cam.zip](https://github.com/user-attachments/files/16917766/backup.cam.zip)

roleoroleo, 2024-09-12:
Try this:
[h60ga_0.3.2.tgz](https://github.com/user-attachments/files/16982632/h60ga_0.3.2.tgz)

fraizzo, 2024-09-12:
i try to install this, orange light steady with sd inserted but nothing seams to happen...

fraizzo, 2024-09-12:
no... i think it's bricked, yellow led stuck and the yi app not recognize it

fraizzo, 2024-09-12:
Ok, i restore the backup and now the cam works with yi app. 

How much time to add custom firmware? Led have to blink during the flash?

roleoroleo, 2024-09-14:
If you configured the cam with the app, you don't need to configure wifi credentials.
Otherwise you can switch off "Disable cloud", configure wifi and then switch on "Disable cloud".

fraizzo, 2024-09-14:
if want to restore original firmware i have to put only backup on sd right?

roleoroleo, 2024-09-15:
Follow the unbrick procedure:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

fraizzo, 2024-09-15:
i try with factory and hacked but nothing happen when i turn on the cam

roleoroleo, 2024-09-16:
Are you using the correct model?
h60ga?

fraizzo, 2024-09-16:
> Are you using the correct model? h60ga?

Yes

roleoroleo, 2024-10-04:
@n4rrOx
The first backup is enough to restore the cam.
Bute the bootloader stores the checksum of the file that you used to restore the cam.
And you can't use it again.

You have to create a new backup file (following the unbrick procedure) with "hacked" option.
In this manner, the checksum will change.

@fraizzo
Could you share with me the file you used to restore the cam?

n4rrOx, 2024-10-04:
> @n4rrOx The first backup is enough to restore the cam. But the bootloader stores the checksum of the file that you used to restore the cam. And you can't use it again.
> 
> You have to create a new backup file (following the unbrick procedure) with "hacked" option. In this manner, the checksum will change.
> 
> @fraizzo Could you share with me the file you used to restore the cam?

Thanks for your help!
I will try probably on Tuesday or latest Thursday and will report.

fraizzo, 2024-10-04:
> Files are ok. The problem could be:
> 
> * sd card is not properly recognized
> * the cam is not a h60ga
> * the bootloader is corrupted (if the yellow light is on the bootloader is ok)

i changed 2 sd... and yes, yellow light is on

n4rrOx, 2024-10-09:
> @n4rrOx The first backup is enough to restore the cam. But the bootloader stores the checksum of the file that you used to restore the cam. And you can't use it again.
>
> You have to create a new [...]

Yesterday I had a little bit time to test....
First I tried to create the "hacked" version (only from backup partition) and installed the SD Card...unfortunately nothing happend.

After that I created another "factory version" from my first backup (but this time from all 3 partitions) and could restore the cam.
So it is working again. Thanks!

I'm not sure if it was normal but the console printed some errors with choosing the "hacked version":
> root@vm-debian:/home/vm-debian/Documents/unbrick# ./build.sh hacked
Image Name:   xiaoyi-rootfs
Created:      Tue Oct  8 17:10:33 2024
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1179648 Bytes = 1152.00 KiB = 1.12 MiB
Load Address: 00000000
Entry Point:  00000000
Image Name:   xiaoyi-home
Created:      Tue Oct  8 17:10:34 2024
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    3407872 Bytes = 3328.00 KiB = 3.25 MiB
Load Address: 00000000
Entry Point:  00000000
./mount.jffs2: line 42: modprobe: command not found
umount: h60ga/mnt: not mounted.
'### Updating h60ga/backup/init.sh
cp: cannot stat 'h60ga/backup/init.sh': No such file or directory
sed: can't read h60ga/backup/tmp_init.sh: No such file or directory
sed: can't read h60ga/backup/tmp_init.sh: No such file or directory
Image Name:   xiaoyi-backup
Created:      Tue Oct  8 17:10:35 2024
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1245184 Bytes = 1216.00 KiB = 1.19 MiB
Load Address: 00000000
Entry Point:  00000000

Is this how it should be?
If selecting the "factory version" I do not see the same errors:

> root@vm-debian:/home/vm-debian/Documents/unbrick# ./build.sh factory
Image Name:   xiaoyi-rootfs
Created:      Tue Oct  8 17:18:54 2024
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1179648 Bytes = 1152.00 KiB = 1.12 MiB
Load Address: 00000000
Entry Point:  00000000
Image Name:   xiaoyi-home
Created:      Tue Oct  8 17:18:54 2024
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    3407872 Bytes = 3328.00 KiB = 3.25 MiB
Load Address: 00000000
Entry Point:  00000000
Image Name:   xiaoyi-backup
Created:      Tue Oct  8 17:18:54 2024
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1245184 Bytes = 1216.00 KiB = 1.19 MiB
Load Address: 00000000
Entry Point:  00000000

But, cam is working again...and I think I have a procedure to get the cam back to life with switching between both versions (have each time different checksums).

Should I try to hack the cam with the version you sent above?
> Try this: [h60ga_0.3.2.tgz](https://github.com/user-attachments/files/16982632/h60ga_0.3.2.tgz)

roleoroleo, 2024-10-11:
This error is the reason for the previous problems.
modprobe command is not found.
Maybe you are using a non standard linux distro.
Or your user is not allowed to load kernel modules.

> Should I try to hack the cam with the version you sent above?

It's a beta version. It includes new features but it's nor stable.
If your current version is ok, it's not recommended to update.

n4rrOx, 2024-10-11:
I'm not sure about the linux distro... I just installed a "plain" debian in a VM.
Nethertheless, I can restore the cam with the "2 sets of backup-files" .... (the not working hacked version and after that the factory version ;-)

But: Up to now I don't got the hack to work with my cam running 12.0.27.0_202401191651.
First I tried the official release, afterwards your attached beta with always the same result: bricked cam

Could I probably assist with more info / tryouts?

roleoroleo, 2024-10-11:
The official release probably doesn't work with this fw version.
What's the status of the cam? Yellow led?
Are you able to enter with a telnet session?

n4rrOx, 2024-10-11:
> The official release probably doesn't work with this fw version. What's the status of the cam? Yellow led? 

Yes, If I apply any hack backup is done, factory folder is renamed and the came stuck with yellow LED.

> Are you able to enter with a telnet session?

With PuTTY via USB?
I think so but would need assistance because I have no knowledge about that (only basic SSH things).

n4rrOx, 2024-10-14:
Today I tried another SD Card with the same result (Sandisk and Samsung, both are 128GB).

Do you probably have a backup of h60ga with firmware 9.0.27* to test?

roleoroleo, 2024-10-14:
If the cam is blocked with the yellow led you can't reach it with telnet.
9.0.27.20_202108161046
[mtdblock4.bin.gz](https://github.com/user-attachments/files/17367713/mtdblock4.bin.gz)

n4rrOx, 2024-10-14:
> If the cam is blocked with the yellow led you can't reach it with telnet. 9.0.27.20_202108161046 [mtdblock4.bin.gz](https://github.com/user-attachments/files/17367713/mtdblock4.bin.gz)

Thanks! I will try to downgrade the cam with your file and will report.

Sorry, probably I explained it not correctly....I started the tests always with a working cam.... with my backup of fw 12.0.27.0_202401191651 I could unbrick it again. Only hack was not successfully with fw 12* ;-)

I understand that the cam should be available/work for connecting via telnet.... but unfortunately I don't have any experience with that (connecting via USB/serial, putty settings, commands etc.)
I saw in older Issues that someone built their own "communication cable" for serial connection...is this necessary or could I just connect it via USB to a notebook and open putty?

roleoroleo, 2024-10-15:
> I saw in older Issues that someone built their own "communication cable" for serial connection...is this necessary or could I just connect it via USB to a notebook and open putty?

Unfortunately you can't use the usb port to access the cam.
You should use a serial connection but it's a complicated task.

If you wanto to downgrade from 12 to 9 you need these files too:
[mtdblock2.bin.gz](https://github.com/user-attachments/files/17373762/mtdblock2.bin.gz)
[mtdblock3.bin.gz](https://github.com/user-attachments/files/17373763/mtdblock3.bin.gz)

n4rrOx, 2024-10-16:
> Unfortunately you can't use the usb port to access the cam. You should use a serial connection but it's a complicated task.
>
> If you wanto to downgrade from 12 to 9 you need these files too: [...]

Thanks for the information and the backup files.
Downgrade is done, hack could be applied, Cam is integrated in HA and works very well!

Thank you for the hard work and the superb support!