roleoroleo/yi-hack-Allwinner-v2 #924: Recordings directory name / time mismatch.

requies, 2024-08-27:
I have a problem with the recordings directory name.
Recordings time and recordings file name match.
Recordings directory name doesn't. The directory is named localtime-2h even tho the creation time is correct.

The localtime in the webcam is correct and the Timezone is correct. (rome = CET-1CEST,M3.5.0,M10.5.0/3)
The event time is set to GMT. If set to localtime the recordings directory name doesn't change while camera internal time is localtime+2h.

Is there any way to fix this ? Thank you

Firmware Version: 0.3.2
Base Version: 12.0.51.08_202403081746
Model Suffix: y623
Model: Yi Pro 2K Home

[image]

roleoroleo, 2024-09-07:
This is normal.
The process that handles records works in GMT.

requies, 2024-09-07:
> This is normal. The process that handles records works in GMT.

Is there any way to change it to lolcaltime ?

roleoroleo, 2024-09-12:
Yes but this change causes a lot of problems.

closed 2025-01-13 by github-actions[bot]

mcanteri, 2026-11-03:
For anyone landing here: I have been running a workaround for this for a
while and it works, at least on `y21ga` (fw 9.0.19.02) and `y291ga`
(fw 12.0.35.8) with yi-hack 0.3.7.

The record process (`rmm` / `mp4record`) inherits the system timezone, which
is UTC because `TIMEZONE` in `system.conf` is not applied by any script.
Exporting it before the hack starts fixes the directory names.

Add this at the top of `/tmp/sd/lower_half_init.sh` (on the SD, so it
survives reboots):

```sh
TZ_LOCALE=$(grep -w TIMEZONE /tmp/sd/yi-hack/etc/system.conf | cut -d'=' -f2-)
if [ -n "$TZ_LOCALE" ]; then
    export TZ="$TZ_LOCALE"
    echo "$TZ_LOCALE" > /etc/TZ
fi
```

To survive hack updates too, put the same block in a script on the SD root
and source it from `/backup/init.sh` (that file is not touched by the online
update).

Result: record directories are named in local time (e.g. `2026Y11M02D23H`
for 23:xx local) and the file names inside match, while the system clock
stays UTC.

Caveat: you said this change can cause problems. In my case I have seen none
since I applied it, but I only record on motion with FTP upload. If there is
a specific case I should test, I am happy to.

Note: manual recordings via `/cgi-bin/record.sh` use another path and stay in
UTC; motion recordings are correct.

roleoroleo, 2026-11-06:
If you use Yi app with this configuration, the timeline will be shifted.
If you don't use the app, I think there are no other problems.