roleoroleo/yi-hack-Allwinner-v2 #926: FTP upload not working

paloi, 2024-09-01:
I am using  yi-hack-Allwinner-v2 0.3.2 on my Yi I1 webcams since yesterday

Everything seems to be working well, **except the FTP upload.** 
I would like to upload the recordings somewhere away from home, hopefully for free or using the storage providers I already have.

It seem that free FTP services became completely off fashion once the cloud storage services became commodity 10 or 15 years ago (dropbox, google drive, MS OneDrive). Lately I try to avoid using those and I have moved to Nextcloud and some other storage provided by my Internet provider (Digi Spain). Actually none of them seem to provide an FTP Server except the ones from Digi Spain or Digi Romania.
In the Digi Spanish pages, there is nothing about FTP, but in the Romanian ones I found this translated to English:
https://storage-rcs--rds-ro.translate.goog/help/ftp?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en-US

I have 50GB of storage included with my Internet Fiber service.
If I connect to it from Windows (just from the simple file explorer), it does work, I can create folders, delete and upload files.
If I do it from Filezilla, Nemo in LinuxMint or the command line from my Linux, I can navigate through the folders, create folders, but **when I make a put to upload a file it does not work.** I get the following error in the command line:

_421 Service not available, remote server timed out. Connection closed.
27366 bytes sent in 01:00 (0.44 KiB/s)_

Basically a 27k is transferred on the  spot, but it fails to write the file.
I have tried this in a folder where I successfully managed to upload files on windows. So that account should have rights on that folder.... 

Of course, coming back to the point of this site, I have setup the FTP upload in the Yi I1 web interface like this and i**t does not work:**
FTP Host                 digistorage.es   or digistorage.net  or 188.26.213.253
FTP Directory 	/DIGIstorage/YiI1hall
FTP Username 	myemailaddress@gmail.com
FTP Password       xxxxxxxxxxxxxxxxxx

On the other hand it is not so strange that it does nto work if I do not managed to make it work manually myself in the command line, on Linux or with Filezila.

Perhaps after reading my long post you can point me out to what I am doing wrong.
btw, if I type the bin command to go into binary transfer mode, it fails the same.
Or perhaps the FTP service provided by Digi my provider in Spain does not really support what I want to do and I have to look for another one. Any advice from anyone on the right setting or a low cost FTP service suitable for uploading with the camera?

I did a test with ftp.cameraftp.com and the outcome was even more cumbersome. It manage to upload 2 videos from yesterday and nothing else. In this service a created a free trial account, which is supposed to keep your file for only 1 day ... but in my tests it doe snot get beyond 2 files and I get the same errors from the command line that I get with Digistore.

Some traces of my tests in the command line:
--------------------------------------------------
_paloi@inspiron15z:~$ ping digistorage.es
PING digistorage.es (188.26.213.253) 56(84) bytes of data.
64 bytes from static-188-26-213-253.digimobil.es (188.26.213.253): icmp_seq=1 ttl=53 time=88.4 ms
64 bytes from static-188-26-213-253.digimobil.es (188.26.213.253): icmp_seq=2 ttl=53 time=89.1 ms
64 bytes from static-188-26-213-253.digimobil.es (188.26.213.253): icmp_seq=3 ttl=53 time=64.0 ms
^C
--- digistorage.es ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 63.998/80.524/89.148/11.689 ms

paloi@inspiron15z:~$ ftp digistorage.es
Connected to digistorage.es.
220 DIGI storage FTP 1.0
Name (digistorage.es:paloi): myemail@gmail.com
331 User name ok, password required
Password: 
230 Password ok, continue
Remote system type is UNIX.
Using binary mode to transfer files.
ftp> cd DIGIstorage
250 Directory changed to /DIGIstorage
ftp> put file.log
local: file.log remote: file.log
229 Entering Extended Passive Mode (|||44068|)
150 Data transfer starting
100% |***********************************| 27366       89.99 MiB/s    00:00 ETA

421 Service not available, user interrupt. Connection closed.
27366 bytes sent in 00:31 (0.85 KiB/s)
ftp> quit

paloi@inspiron15z:~$ ftp 188.26.213.253
Connected to 188.26.213.253.
220 DIGI storage FTP 1.0
Name (188.26.213.253:paloi): myemail@gmail.com
331 User name ok, password required
Password: 
230 Password ok, continue
Remote system type is UNIX.
Using binary mode to transfer files.
ftp> cd DIGIstorage
250 Directory changed to /DIGIstorage
ftp> put file.log
local: file.log remote: file.log
229 Entering Extended Passive Mode (|||44129|)
150 Data transfer starting
100% |***********************************| 27366       49.71 MiB/s    00:00 ETA

421 Service not available, remote server timed out. Connection closed.
27366 bytes sent in 01:00 (0.44 KiB/s)_

paloi, 2024-09-01:
To make the situation with the FTP access to Digi more confusing the first time I accessed teh service from Windows I got the below screenshot when opening each folder by first time:
[image]

And from the Linux Mint file manager (nemo) it lets me create a new text file, save it, etc ...  but it does not let me upload a text file from my local folder.

roleoroleo, 2024-09-09:
I don't know why this Digi service doesn't work properly with a standard ftp client.
And yes, I know that FTP is a very old protocol. But it's simple and available in Busybox.
If you want to be sure if the transfer works in the cam, please try a temporary ftp server in your local network.

closed 2025-01-09 by github-actions[bot]