roleoroleo/yi-hack-Allwinner-v2 #929: Yi Cam 1080p Home (y211ga) looses hack after restart over web gui

i7zTr0n, 2024-09-07:
I have the problem, that after restarting the cam over the Web Gui to load a new config it starts the stock firmware again.
"Welcome to Yi Home Cam. Waiting to connect".
I managed to restore the hack after some combination of reformating the sd card and unbricking.
How can I prevent that?

roleoroleo, 2024-09-12:
Probably the cam lost wifi connection parameters.
Try to resync the cam with the app and then restart it.

UserRS232, 2024-09-23:
same here... lost wifi login after every reboot/power
resync with the app helps for the moment until no new reboot

roleoroleo, 2024-10-04:
I re-tested now my y211ga and it's ok.
Please, try to dump the flash and check if ssid and password are stored properly:
`hexdump -C -n 160 /dev/mtdblock7`

Rushmed, 2024-11-22:
Hi.

Got the the same on Yi Dome Camera U (2K) SFUS h51ga.

Ouput of `hexdump -C -n 160 /dev/mtdblock7` looks fine but that is clear as I reconnect the wifi of the cam via app after reboot to get shh access.
Cant do `hexdump -C -n 160 `/dev/mtdblock7` when cam says "waiting to connect".

roleoroleo, 2024-11-24:
I have a h51ga. I will try asap.

Rushmed, 2024-12-06:
> I have a h51ga. I will try asap.

Any news here? If needed I can try to support.

roleoroleo, 2024-12-13:
I finished the test now. And it's working properly.
Also after 5-6 reboot cycles.
I don't understand the reason of your problem.
Maybe a hardware issue.
Did you try another SD card?
What about if you remove the SD card? Does the cam survive to a reboot?
Does the problem disappear if you don't use the hack?

Rushmed, 2024-12-13:
Hi.

Thanks for testing.
I downloaded the hack files fresh from gh, took a new SD and always tested multiple reboots with following results:

- without SD; ok
- with SD without hack; ok --> SD formatted in cam
- with SD, with hack (nothing set up); ok
- with SD, with with hack and "Disable Cloud" and "Recording without Cloud" activated; ok
- with SD, with hack and everything set up; not ok ("Waiting to connect")

Also trying to siable Kernel Tunig does not help.

I dont want to stress this further. I'll place the cam on an accesible spot and try to not reboot it.

roleoroleo, 2024-12-14:
Send me the non working config.
I will copy to my cam.

Rushmed, 2024-12-16:
@roleoroleo Did you try my conf?

closed 2025-04-16 by github-actions[bot]

fran2a, 2025-12-03:
hi all, 
same problem for me with last hack 0.3.4.
Any update on this issue?
Thanks.