roleoroleo/yi-hack-Allwinner-v2 #930: Bricked yi 1080p while backing uo

Algtronic, 2024-09-09:
I connected sd card with backup scripts and now camera continuously yellow.  I can't reset or do anything. The type is BFUSY. Can you kindly give any suggestions?

roleoroleo, 2024-09-12:
The backup script cannot brick the cam.
Remove the sd card and check power and cable.

If the backup procedure completed you can try a restore.

Algtronic, 2024-09-12:
> The backup script cannot brick the cam. Remove the sd card and check power and cable.
Maybe cam was faulty at the first place I dunno.
> If the backup procedure completed you can try a restore.
I think you already understand it's not the case...

But thank you for reply.

closed 2024-09-14 by Algtronic

roleoroleo, 2024-09-14:
If you know the exact model (I mean y21ga, h52ga, ecc...) I can send you a backup copy.

Algtronic, 2024-09-21:
It reboots since first start I suppose (I tried to backup right out of the box). So no, I have no idea, really. But now I can play russian roullete by flashing random firmware. If you have one I'll be glad to try. Stakes are really low...

roleoroleo, 2024-09-21:
Too many versions...
Are you able to take a picture of the board? Usually there are labels on the PCB.

Algtronic, 2024-09-21:
Thanks for the tip! The PCB says Y20GA_MB_V1.1 2019/11/07
[image]

roleoroleo, 2024-09-21:
Try here:
https://github.com/roleoroleo/yi-hack-Allwinner

Algtronic, 2024-09-23:
I don't have any factory file for unbricking since backup never actually worked. Should I try to install modified allwinner firmware?

roleoroleo, 2024-09-24:
If the hack procedure works, it will create a backup automatically.
Otherwise I will send you a backup from another cam (same model).

Algtronic, 2024-10-03:
Unfortunately, no luck with Allwinner hack v1. I waited more than hour, but light is still orange and nothing changed in SD card folders. I assume I need factory files to proceed with unbricking

roleoroleo, 2024-10-04:
This should be enough:
[mtdblock3.bin.gz](https://github.com/user-attachments/files/17257732/mtdblock3.bin.gz)

Follow the unbrick procedure: https://github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick
Let me know if it works.

Algtronic, 2024-12-29:
It worked! Thanks, much appretiated. 
[image]

Algtronic, 2025-01-10:
I successfully installed hack v1 but encountered another problem during the process. After installation I didn't find a name of the device (but camera was connected to the router). The IP was in the list though. API page didn't open so I tried ssh. No luck with it. Then I tried telnet and it was online. I stopped experiments and reinstalled hack. Then problems were gone. I checked log file for wifi connection before wipe. It's nothing special, but I'll attach them here. Maybe that will be helpful for further development.
```
[./dispatch][1/1/0:0:10:33]: dispatch.c(main-5267) open_ptz fail[./dispatch][1/1/0:0:10:97]: dispatch.c(get_config-1251) got sn(TAjH5J4JDcYaO1hScpGmXHlJE3da6jme)[./dispatch][1/1/0:0:10:97]: dispatch.c(get_config-1252) got pwd(%password was here%)[./dispatch][1/1/0:0:10:97]: dispatch.c(get_config-1253) got ssid(%ssid was here%)[./dispatch][1/1/0:0:10:97]: dispatch.c(get_config-1254) got tnp_init_string(MMFBJPLDIEEKKKHPODFPPOEJHNCHFCGAHHEDKFIJIDBCDIBDPKIIAONIPBCLBOKOPIKMDKPMNADGACDMBC)[./dispatch][1/1/0:0:10:115]: dispatch.c(choose_server-649) in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, log_server = http://log.us.xiaoyi.com, sname = familymonitor-y20ga, dlproto = mius
[./dispatch][1/1/0:0:10:115]: dispatch.c(choose_server-649) in choose_server, region_id = 16, api_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sname = familymonitor-y20ga, dlproto = mieu
[./dispatch][1/1/0:0:10:126]: dispatch.c(main-5348) init ok, cost time(2542 ms)[./dispatch][1/1/0:0:11:148]: dispatch.c(do_monitor_wifi-1725) wifi disconnected, now reconnect wifi
[./dispatch][5/13/5:37:57:97]: dispatch.c(p_worker-3307) DISPATCH_SET_DEFAULT_TIME 1620884277[./dispatch][5/13/5:38:1:172]: dispatch.c(do_monitor_wifi-1735) wpa connected , now dhcp
[./dispatch][5/13/5:38:1:311]: dispatch.c(do_monitor_wifi-1737) wpa connected , now dhcp finish
[./dispatch][5/13/5:38:2:326]: dispatch.c(check_wifi_connect-1505) bssid(50:d2:f5:0b:35:96)[./dispatch][5/13/5:38:2:326]: dispatch.c(check_wifi_connect-1527) ip(%ip was here%)[./dispatch][5/13/5:38:2:333]: dispatch.c(check_wifi_connect-1538) gw(%gateway was here%)[./dispatch][5/13/5:38:3:370]: dispatch.c(check_wifi_connect-1547) gwmac(50:D2:F5:0B:35:94)[./dispatch][5/13/5:38:3:377]: dispatch.c(check_wifi_connect-1557) mask(255.255.255.0)[./dispatch][5/13/5:38:3:391]: dispatch.c(check_wifi_connect-1567) mac(7C:A7:B0:43:99:FB)[./dispatch][5/13/5:38:3:440]: dispatch.c(check_wifi_connect-1581) signal_quality(54)
```

roleoroleo, 2025-01-12:
If telnet is running the cam is half-hacked.
The init files are ok but something was wrong.
Please run the following commands and share the output:
```
ps
ls -laR /home
cat /home/app/init.sh
```

Algtronic, 2025-01-14:
I already hacked it by running second time from scratch. Thanks again, your firmware is tp-notch!