roleoroleo/yi-hack-Allwinner-v2 #931: Yi Pro 2K Home (YFUS) RTSP not working

aurevo24, 2024-09-10:
Hello,

I formatted the SD card and used the newest release, but the RTSP stream is not working. No HD, no low quality.

VLC shows black and no stream and same with Blue Iris.

Firmware Version | 0.3.2
Base Version | 12.0.51.08_202403081746
Model Suffix | y623
Serial Number | YFUSY94I*************

This is not the first time I have used firmware from your project, on the 1080p model it worked straight away and without any problems.

mkoehne, 2024-09-11:
I have the same problem.

SiriusLightning21, 2024-09-11:
Same problem for me

floronxt, 2024-09-11:
Workaround:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/915

aurevo24, 2024-09-12:
> Same problem for me.
> 
> Edit: I disabled the Screenshot mode and everything what I did not need. After this I enabled both streaming modes + sound and reboot the camera. Now it works for me with both streams.

I enabled RTSP Stream for both, RTSP Audio to pcm, disabled snapshot and onvif and now it is also working for me.

roleoroleo, 2024-09-12:
y623 is a problem for me.
There is something that I don't understand...

oli737, 2024-09-13:
Similar problem with this camera.

YFUSY94P.....

12.0.51.01_202303091901

Stream and Snapshot is not working after couple minutes running.

oli737, 2024-09-13:
> I enabled RTSP Stream for both, RTSP Audio to pcm, disabled snapshot and onvif and now it is also working for me.

Not working for me.

roleoroleo, 2024-09-14:
Are you using my custom integration for HA?
Check the HA registry, are there error messages?

oli737, 2024-09-14:
> Are you using my custom integration for HA? Check the HA registry, are there error messages?

2024-09-14 19:18:35.702 ERROR (stream_worker) [homeassistant.components.stream.stream.camera.yi_hack_a2_########_cam] Error from stream worker: Error opening stream (ERRORTYPE_111, Connection refused, rtsp://****:****@192.###.###.###.###/ch0_0.h264)

After reboot it works couple minutes.
#### i censored

dwl99, 2024-09-27:
I flashed my YFUS camera with firmware 0.3.2 and can access the webserver but rtsp isn't working. I've disabled snapshot and ONVIF and enabled hi & lo resolution streams. I've also tried it with and without the alternative rtsp server but still nothing. Any ideas?

MakoShark2, 2024-10-03:
Another one... YFUS 0.3.2, webserver is up but no rtsp... Tried all suggestions but not working...

@dwl99 by hard reboot you mean reset?

roleoroleo, 2024-10-04:
Please, test this beta:
[y623_0.3.2.tgz](https://github.com/user-attachments/files/17257174/y623_0.3.2.tgz)

PauloJTavares, 2024-10-05:
Thank you roleoroleo. Woks just fine.
I figured out what my issue was when I formated the SD card from within the app and reloaded the original hack. Works like a charm. Great job

scharrrfi, 2024-10-07:
> Please, test this beta: [y623_0.3.2.tgz](https://github.com/user-attachments/files/17261559/y623_0.3.2.tgz)

With this version I have no RTSP stream at all. With the original 0.3.2 version, the stream is there for about 10 to 30 seconds and then I got a timeout for 1 - 2 minutes and then the stream is back for again 10 to 30 seconds.

With the same settings, I have about 20000kb free memory with the patched (not working) version and about 8000kb free memory with the original version.

roleoroleo, 2024-10-25:
Check this post:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/753#issuecomment-2438422971

closed 2025-02-23 by github-actions[bot]