roleoroleo/yi-hack-Allwinner-v2 #932: Can you change the osd time format for rtsp?

Export33, 2024-09-12:
Hi,

i´m using the Timezone Europe/Berlin 
```
CET-1CEST,M3.5.0,M10.5.0/3
```

Is there a way to change the time-format for the rtsp-stream?
Instead of `mm/dd/yyyy` i would like to use the german format `dd/mm/yyyy` or even better `dd.mm.yyyy`.

[image]

roleoroleo, 2024-09-12:
Mmmmh
What cam?
Mine is:
[image]

Export33, 2024-09-13:
Your date also looks ok.

It´s a YI Home 2K PRO

[image]

roleoroleo, 2024-09-14:
Please check what timezone is set in the Yi app.

Export33, 2024-09-16:
I´m using your hack in order to not use the app.
Is there a ssh command to do so? The commands i found for timezone didn´t work.

`date`  shows the following
[image]

roleoroleo, 2024-09-16:
This is the pseudocode:
```
        pcVar2 = strstr((char *)(g_rmm_info._8_4_ + 0xc),"us");
        if (pcVar2 == (char *)0x0) {
          pcVar2 = strstr((char *)(g_rmm_info._8_4_ + 0xc),"eu");
          if ((pcVar2 == (char *)0x0) &&
             (pcVar2 = strstr((char *)(g_rmm_info._8_4_ + 0xc),"plt-api-de"), pcVar2 == (char *)0x0)
             ) {
            snprintf(acStack_1d0,0x80,"%04d/%02d/%02d %02d:%02d:%02d ",local_2c->tm_year + 0x76c,
                     local_2c->tm_mon + 1,local_2c->tm_mday,local_2c->tm_hour,local_2c->tm_min,
                     local_2c->tm_sec);
          }
          else {
            snprintf(acStack_1d0,0x80,"%02d/%02d/%04d %02d:%02d:%02d ",local_2c->tm_mday,
                     local_2c->tm_mon + 1,local_2c->tm_year + 0x76c,local_2c->tm_hour,
                     local_2c->tm_min,local_2c->tm_sec);
          }
        }
        else {
          snprintf(acStack_1d0,0x80,"%02d/%02d/%04d %02d:%02d:%02d ",local_2c->tm_mon + 1,
                   local_2c->tm_mday,local_2c->tm_year + 0x76c,local_2c->tm_hour,local_2c->tm_min,
                   local_2c->tm_sec);
        }
```

This means that rmm process tries to find the string eu, us or plt-api-de in the file /tmp/mmap.info starting from address 0xc.
It's a part of the server name.
So, dump the content of /tmp/mmap.info:
`hexdump -C /tmp/mmap.info`

Export33, 2024-09-17:
Sure.

```
root@Kamera:~# hexdump -C /tmp/mmap.info
00000000  10 00 00 00 00 00 00 00  03 00 00 00 68 74 74 70  |............http|
00000010  73 3a 2f 2f 61 70 69 2e  75 73 2e 78 69 61 6f 79  |s://api.us.xiaoy|
00000020  69 2e 63 6f 6d 00 00 00  00 00 00 00 00 00 00 00  |i.com...........|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000100  00 00 00 00 00 00 00 00  00 00 00 00 68 74 74 70  |............http|
00000110  3a 2f 2f 6c 6f 67 2e 75  73 2e 78 69 61 6f 79 69  |://log.us.xiaoyi|
00000120  2e 63 6f 6d 00 00 00 00  00 00 00 00 00 00 00 00  |.com............|
00000130  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000180  00 00 00 00 00 00 00 00  00 00 00 00 66 61 6d 69  |............fami|
00000190  6c 79 6d 6f 6e 69 74 6f  72 2d 79 36 32 33 00 00  |lymonitor-y623..|
000001a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001c0  00 00 00 00 00 00 00 00  00 00 00 00 6d 69 75 73  |............mius|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 00 00  00 00 00 00 48 a3 17 00  |............H...|
00000210  00 00 00 00 31 32 2e 30  2e 35 31 2e 30 31 5f 32  |....12.0.51.01_2|
00000220  30 32 33 30 33 30 39 31  39 30 31 00 00 00 00 00  |02303091901.....|
00000230  00 00 00 00 30 30 30 30  30 30 30 30 30 30 30 30  |....000000000000|
00000240  30 30 30 30 30 30 30 30  30 30 30 30 30 30 30 30  |0000000000000000|
*
00000270  30 30 30 30 00 00 00 00  00 00 00 00 00 00 00 00  |0000............|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002f0  00 00 00 00 54 4e 50 55  53 41 51 2d 32 38 30 39  |....TNPUSAQ-2809|
00000300  38 32 2d 4e 46 4c 5a 4b  00 00 00 00 00 00 00 00  |82-NFLZK........|
00000310  00 00 00 00 59 46 55 53  59 39 34 42 56 52 51 45  |....YFUSY94BVRQE|
00000320  XX XX XX XX XX XX XX XX  00 00 00 00 00 00 00 00  |XXXXXXXX........|
00000330  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000350  00 00 00 00 43 75 6c 5a  49 36 46 48 75 70 39 7a  |....CulZI6FHup9z|
00000360  31 6a 6f 66 00 00 00 00  00 00 00 00 00 00 00 00  |1jof............|
00000370  00 00 00 00 00 00 00 00  30 30 30 30 30 30 30 30  |........00000000|
00000380  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000390  00 00 00 00 00 00 00 00  46 46 46 46 46 46 46 46  |........FFFFFFFF|
000003a0  46 46 46 46 46 46 46 46  00 00 00 00 00 00 00 00  |FFFFFFFF........|
000003b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000003e0  00 00 00 00 00 00 53 37  59 52 52 63 6b 61 4a 4d  |......S7YRRckaJM|
000003f0  31 7a 49 67 6b 6c 49 38  56 74 64 76 67 45 34 4f  |1zIgklI8VtdvgE4O|
00000400  7a 75 38 66 31 47 00 00  00 00 00 00 00 00 00 00  |zu8f1G..........|
00000410  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000420  00 00 00 00 00 00 32 43  3a 43 33 3a 45 36 3a 35  |......2C:C3:E6:5|
00000430  42 3a 46 33 3a 35 46 00  00 00 00 00 00 00 00 00  |B:F3:5F.........|
00000440  00 00 00 00 00 00 36 36  3a 32 32 3a 33 32 3a 61  |......66:22:32:a|
00000450  65 3a 64 66 3a 36 32 00  00 00 00 00 00 00 00 00  |e:df:62.........|
00000460  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000480  00 00 00 00 00 00 31 39  32 2e 31 36 38 2e 33 30  |......192.168.30|
00000490  2e XX XX 00 00 00 00 00  00 00 00 00 00 00 00 00  |.IP.............|
000004a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000004e0  00 00 00 00 00 00 XX XX  XX XX XX XX XX XX XX XX  |......SIDXXXXXXX|
00000XXX  XX XX XX XX 00 00 00 00  00 00 00 00 00 00 00 00  |XXXX............|
00000500  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000520  00 00 00 00 00 00 XX XX  XX XX XX XX XX XX XX XX  |......PWXXXXXXXX|
00000530  XX XX XX XX XX XX XX XX  XX XX XX XX XX XX XX XX  |XXXXXXXXXXXXXXXX|
00000540  XX XX XX XX XX XX XX XX  XX XX XX XX XX XX XX XX  |XXXXXXXXXXXXXX..|
00000550  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000560  00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00  |................|
00000570  20 1c 00 00 20 1c 00 00  00 00 00 00 00 00 00 00  | ... ...........|
00000580  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000590  00 00 00 00 b4 00 00 00  00 00 00 00 64 00 00 00  |............d...|
[... 22 lines cut ...]
00000720  00 00 00 00 00 00 00 00  01 00 00 00 00 00 00 00  |................|
*
00000740  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000008a0  54 4e 50 55 53 41 51 2d  32 38 30 39 38 32 2d 4e  |TNPUSAQ-280982-N|
000008b0  46 4c 5a 4b 00 00 00 00  00 00 00 00 00 00 00 00  |FLZK............|
000008c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000009c0  00 00 00 00 00 00 02 50  10 27 00 00 00 00 00 00  |.......P.'......|
000009d0  31 62 35 33 34 64 33 30  33 30 33 30 33 30 33 30  |1b534d3030303030|
000009e0  31 30 33 39 61 35 32 61  66 31 30 30 65 63 63 35  |1039a52af100ecc5|
000009f0  0a 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000a00  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000a30  00 00 00 00 00 01 00 00  00 00 00 00 00 00 00 00  |................|
00000a40  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000a50  00 00 00 00 03 00 00 00  01 00 00 00 00 00 00 00  |................|
00000a60  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000a90  01 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00  |................|
00000aa0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000ac0  00 00 00 00 00 00 00 00  01 00 00 00              |............|
00000acc
```

roleoroleo, 2024-09-17:
us = mm/dd/yyyy
You could try to change us to eu.

Export33, 2024-09-18:
That works. Is there any way to make it permanent? Right now it´s reset when you reboot the camera.

[image]

roleoroleo, 2024-09-20:
I think you should change the serial number.
But you can try to write the command you used in the startup script:
/tmp/sd/yi-hack/startup.sh

Export33, 2024-09-30:
How can i change the serialnumber? and to what?

Neither in '/tmp/sd/yi-hack' nor in '/tmp/sd/yi-hack/scripts/script' is a startup.sh.
can i write my own one and it´s used?

roleoroleo, 2024-10-04:
startup.sh file doesn't exist.
You have to create it in `/tmp/sd/yi-hack/startup.sh` and the boot script will run it:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/82e537dfe307211af1a3bb8d8f1b89f6ec5d6ee8/src/static/static/yi-hack/script/system.sh#L474

dedors, 2025-01-31:
I don't understand what startup.sh should contain.
And is this function not worth to be impelemented in the UI?

roleoroleo, 2025-02-01:
See this example:

```
root@yi-hack-h52ga:~# hexdump -C -n 32 /tmp/mmap.info
00000000  01 00 00 00 10 00 00 00  00 00 00 00 68 74 74 70  |............http|
00000010  73 3a 2f 2f 61 70 69 2e  65 75 2e 78 69 61 6f 79  |s://api.eu.xiaoy|
00000020
root@yi-hack-h52ga:~# echo -n "us" | dd of=/tmp/mmap.info bs=1 seek=24 count=2 conv=notrunc
2+0 records in
2+0 records out
2 bytes (2B) copied, 0.000219 seconds, 8.9KB/s
root@yi-hack-h52ga:~# hexdump -C -n 32 /tmp/mmap.info
00000000  01 00 00 00 10 00 00 00  00 00 00 00 68 74 74 70  |............http|
00000010  73 3a 2f 2f 61 70 69 2e  75 73 2e 78 69 61 6f 79  |s://api.us.xiaoy|
00000020
```

closed 2025-06-02 by github-actions[bot]