roleoroleo/yi-hack-Allwinner-v2 #934: PTZ inverted?

tib-kpl, 2024-09-16:
Hello, 

I have h60ga cameras with firmware 0.3.2 and I notice that the up/down ptz is reversed. I think I saw a similar ticket closed on this subject https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/196. 
But the fix code seems to have disappeared with version 0.3.1, is this normal?

roleoroleo, 2024-09-18:
The code was moved to ipc_cmd, but h60ga is not in the list of "inverted" cam.
Try this:
[ipc_cmd.gz](https://github.com/user-attachments/files/17039523/ipc_cmd.gz)

tcaptalan, 2024-09-18:
@roleoroleo I just tried with the ipc_cmd you provided and it works fine. Thank you

tib-kpl, 2024-09-18:
@roleoroleo I just tried with the ipc_cmd you provided and it works fine. Thank you

closed 2025-01-18 by github-actions[bot]