roleoroleo/yi-hack-Allwinner-v2 #935: Question about update

oli737, 2024-09-17:
Can I update from hacked version via the Yi app?

Running rn on version:

YFUSY94P.....
12.0.51.01_202303091901

Do I need need patch the camera after again or is there a risk update the hacked cam to last version from app?

Is the last version running smooth or should I wait update the camera firmware?

Thanks.

roleoroleo, 2024-09-18:
There is no answer to this question.
It depends on the content of the update.
The hack changes the content of the file /backup/init.sh
If the update overwrites this file, you need to apply again the hack.

If you don't want to restore a backup before the update, at least
- restart the cam without the sd card (the cam starts with the original fw apart from /backup/init.sh)
- apply the update
- restart the cam with the sd card
- check if the cam is hacked, if it's not hacked apply again the hack.

oli737, 2024-09-18:
Thank you, it worked.

closed 2025-01-18 by github-actions[bot]