roleoroleo/yi-hack-Allwinner-v2 #937: ONVIF motion detection on HIK nvr

piot84zet, 2024-09-20:
Hello, thank you for the great job.
I got problem with my hik nvr. I want to start records based on motion detection. Is any way to make it work?
my yi home 2k pro (y623) is detected by nvr, got streams, can continuous record. Want to change the continuous record to record when motion is detected. When changes options in my nvr, always shows cam doesn't suport motion detection. 
I can resign lot from fw, just need to cam work with nvr and record when motion is detected. 
Thank you for any help.

piot84zet, 2024-09-20:
Thank you for replay.
What is strange, ODM does not see my cam. others cams (hik) are visiable. Will try to reformat sd and start everything from begining. 
Is it nessesary to start with official app? I just opened box after shippment arrived and inserted sd with your fw. 

Will be back soon.

piot84zet, 2024-09-20:
Ok, ODM can see yi hacked cam, but events shows very late, or not.
Well...no matter what I try to set on yi hack web, and after try to turn on motion detection for this cam in nvr, I just get "incorrect operation" from my nvr website.

roleoroleo, 2024-09-20:
I recently fixed some bugs.
Please try this version: 
[y623_0.3.2.tgz](https://github.com/user-attachments/files/17078115/y623_0.3.2.tgz)

piot84zet, 2024-09-20:
Thank you for new image. 
On this image webapi works faster I think so. 
For motion detection, I changed all options related to rtsp and onvif and unfortunately I still get the "incorrect operation" message.
I tried to change the RTSP server program to "alternative" or "go2rtc" but after reboot it returns to "standard".
In ODM when clik "Events" get "Error during events subscripion: Subscribe creation failed fault".

Any sugestion?
Should I turn on/off anything in yi app? i didnt even instaled it, just started with yi hack on brand new cam.

roleoroleo, 2024-09-21:
> For motion detection, I changed all options related to rtsp and onvif and unfortunately I still get the "incorrect operation" message.

RTSP is not related to onvif events.

> I tried to change the RTSP server program to "alternative" or "go2rtc" but after reboot it returns to "standard".

Maybe you need to clear the cache of the browser, I changed js code.

> In ODM when clik "Events" get "Error during events subscripion: Subscribe creation failed fault".

This is strange. Check the option, are you using PullPoint or WS-Base?
[image]

> Should I turn on/off anything in yi app? i didnt even instaled it, just started with yi hack on brand new cam.

No, if the wifi is running properly, the app is not necessary.

piot84zet, 2024-09-21:
I just installed ODM, nothing changed, so "application settings" looks like your screen.

After few hours of searching just clicked on my other cam with onvif where motion detection works well. And please look at this screen:
[image]
In "Profiles" section can find "Video Analytics Configuration" and some with "motion detection rule"
here is screen from yi hack cam:
[image]
there is no section with Analytics

here is my prtscreen from yi hack webapi config site onvif section:
[image]

roleoroleo, 2024-09-23:
VideoAnalytics is not supported in this simplified onvif server.
But I never find a client that needs this configuration.
Usually (client side) events handling is not related to VideoAnalyticsConfiguration.

About ODM, what do you see in Events page?
Doesc ODM receive event messages?

piot84zet, 2024-09-23:
Do I need to have like that in options?
[image]

In ODM got mesages after a long while when cover cam with hand. 
[image]

roleoroleo, 2024-09-23:
Yes, you need motion detection enabled in your cam.
Is the message late?
If yes, please compare the time with the mqtt messages, if you configured mqtt.
I need to understand if the delay depends on the cam or on the onvif implementation.
Another try, you could try configuring ODM with WS-Base instead of PullPoint.

piot84zet, 2024-09-23:
Yes, get late mesages. 
On Base Subscription  no messages at all.
but even got messages on Pullpoint in ODM, NVR doesnt recrod any video based on motion.

Must configure mqtt to check it deley, will check it later.

What else I find, when change RTSP server program to other than Standard, NVR cant connect with cam.

piot84zet, 2024-09-26:
Well, what is strange, when open ODM when cam is on, get error "Subcribe creation failed fault" with reading Events in ODM. after a while, cam reset itself. After reset cam and refresh ODM, get no error in Events section. 
On Base Subscription get late messages too or no messages at all. all firewalls turned off.
Noticed, after rebots cam (changing options) Evenst show "Subcribe creation failed fault". must restart ODM and cam.

Ok, tried to change to go2rtc/alternative and check with vlc, but after saving and rebot, its go back to "standard" rtsp server.

Is there any way to go back stock if i dont have my backup? just formated card with all files.... :)

piot84zet, 2024-09-26:
just upgraded to 12.0.51.08_202403081746 and started from beginning.
On standard  y623_0.3.2.tgz  on "high" profile in Onvife can't get motion records in nvr. after selectin "low" in "profile Onvife" nvr cant connect with cam.
Tried to change for ONVIF Network Interface to "Ethernet" but nvr cant connect with cam.

roleoroleo, 2024-10-04:
Please, update with this beta:
[y623_0.3.2.tgz](https://github.com/user-attachments/files/17257174/y623_0.3.2.tgz)

piot84zet, 2024-10-04:
upgraded from sd.
still cant start move detection with this cam in my nvr. 
tried to change Configurations->[ONVIF configuration]-> ONVIF profile -> both/low
after this I get "error of parametr" when trying to connect from nvr.
tried to change HTTPD Port to 4520, but still get "error of paramter".
After go back to 80 port, still "error of parametr". 
So Im going to factory reset it now.

piot84zet, 2024-10-04:
after factory reset still "error of paramtr" for this cam and cannt add to nvr. Everty time I do any chagne in opiotns in cam, I delete from nvr and add freshly. everytime get error.

i dont know what is going now.... will try to reformat sd and try again from the beginning.

piot84zet, 2024-10-08:
Well, after hard work to find any log, can say it has just main logs on nvr's lokal website like starting recording from movement, disconected or smth like that. 
Find option to log on usb drive, will tray soon, and will be back with this log.
Probably it will be better to turn off others cam to have clear logs, am I right?

roleoroleo, 2024-10-11:
> Probably it will be better to turn off others cam to have clear logs, am I right?

You could also capture network traffic with wireshark or tcpdump.

closed 2025-02-10 by github-actions[bot]

megadimich, 2025-04-27:
Firmware Version 0.3.3, Base Version 12.0.51.08_202411131107 - not work ONVIF motion detection.