roleoroleo/yi-hack-Allwinner-v2 #939: y623 Artefacts

oli737, 2024-09-22:
Hi there,
I'm hoping someone can help me with a little problem I'm having. I'm running 5 cameras and have artefacts randomly every couple of seconds. I'm running the last firmware and the last 0.32 hack version. I've tried TRSPtoWEB, go2rtc and a generic camera in home assistant, but the only way I have no artefacts is using VLC media player. Is there something I can try to fix this issue? Thank you very much.

roleoroleo, 2024-09-22:
Please, send me a copy of the video buffer.
/dev/shm/fshare_frame_buf

oli737, 2024-09-22:
[fshare_frame_buf.zip](https://github.com/user-attachments/files/17088987/fshare_frame_buf.zip)

roleoroleo, 2024-09-24:
Something strange at the end of the buffer.
Please send me 3 files:
```
cp /dev/shm/fshare_frame_buf /tmp/buf1.bin
cp /dev/shm/fshare_frame_buf /tmp/buf2.bin
cp /dev/shm/fshare_frame_buf /tmp/buf3.bin
```
I need to compare them.

oli737, 2024-09-24:
I did fix this problem with deleting RTSPtoWEB and adding the streams manually again. Since than I never had that issue after again. I think it can be closed 🔒.
Idk this fix make sense but it worked for all cameras.

oli737, 2024-09-25:
Problem came back.
[buffer.zip](https://github.com/user-attachments/files/17134319/buffer.zip)

oli737, 2024-09-26:
RN it's working again.

oli737, 2024-10-04:
Yes RN it does work.

closed 2025-02-03 by github-actions[bot]