roleoroleo/yi-hack-Allwinner-v2 #942: YI Dome U 2K (h51ga) reboot frequently

Plop, 2024-09-23:
Hi,

I've got an issue with my h51ga camera, firmware version 0.3.2, base version 9.0.21.13_202109101104. It frequently (once or twice an hour) reboots.

Here are some info : 
[image]

The following services are enabled : 
- RTSP server
- ONVIF
- SSH
- FTP
- NTPD
- HTTPD
All the others are disabled.

The swap file is enabled, and the swappiness value is at 15.
The kernel tuning is enabled.

No detection is enabled.

Is there a way to fix this issue ?

Thanks.

roleoroleo, 2024-09-24:
Try to disable kernel tuning.
For some models these settings are too "strong".

Gifford47, 2024-12-18:
I've the same issue. Disabling kernel optimization didn't help. After a reboot all camera settings are reverted.

@roleoroleo do you know any workaround?

roleoroleo, 2024-12-22:
Please try another power supply or cable.
II have seen this problem many times.

Gifford47, 2025-01-14:
ok, thanks! A new power supply solved my issue!

closed 2025-05-15 by github-actions[bot]