roleoroleo/yi-hack-Allwinner-v2 #943: Upgrade from yi-hack-Allwinner v1 Not Applying

Garywoo, 2024-09-25:
I have a BFUS camera which was running an old version of [yi-hack-Allwinner](https://github.com/roleoroleo/yi-hack-Allwinner) 0.2.1.

I was unable to upgrade its firmware in the web UI (it think it was too old), so I applied the hack manually from scratch using the latest release which was 0.3.9. This worked fine.

I then realised that this v2 repo of the hack exited, so wanting to keep up-to-date, I tried to install the latest version from this repo (0.3.2) to the camera by formatting the SD and copying over the files. After the camera boots, the version is still listed as 0.3.9 in the web UI, and the GitHub project still shows `yi-hack-Allwinner`.

I'm assuming that the firmware update did not work, as I would expect to see the GitHub project say `yi-hack-Allwinner-v2` and the version say 0.3.2. Is my assumption correct?

roleoroleo, 2024-10-04:
Allwinner-v2 is not an upgrade for Allwinner.
It's a version dedicated to a different "family".
Same cpu but different flash layout.

Garywoo, 2024-10-05:
Oh ok. I saw that my camera was supported on both, so assumed v2 was an upgrade.

Can I suggest adding a "_What's the difference between v1 and v2_" section to the readme or wiki, so that others don't also come to the same conclusion I did?

roleoroleo, 2024-10-05:
Yes, I will add a note in the readme.

closed 2025-02-05 by github-actions[bot]