roleoroleo/yi-hack-Allwinner-v2 #944: Xiaomi Outdoor AW300

rubenleon, 2024-09-27:
Hello, is this camera compatible with the project?

https://www.amazon.es/Xiaomi-vigilancia-rotaci%C3%B3n-detecci%C3%B3n-Movimiento/dp/B0BW48J4Z3/ref=sr_1_13?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&sr=8-13

Thx 😊

roleoroleo, 2024-10-04:
AFAIK it's not compatible.

rubenleon, 2024-10-12:
> AFAIK it's not compatible.

"Do you know where the 'Yi Outdoor 1080p' model or a similar one that is compatible with the firmware is sold? Can you share the link to the store?"

roleoroleo, 2024-10-14:
If you are looking for an outdoor cam it's not simple.
But usually you can buy Yi cams on Aliexpress.

rubenleon, 2024-10-16:
Thank you very much for your response.

I'm having a hard time finding an outdoor camera compatible with this project.
The Yi Home camera is easy to find, but the Yi Outdoor seems impossible to get.

Best regards.

closed 2025-02-14 by github-actions[bot]