roleoroleo/yi-hack-Allwinner-v2 #946: No support for Y29GA_MB_V2.1?

graham30rad, 2024-10-05:
The firmware is 12.1.20 but I only see support for 12.1.119, which isn't working.

The board is Y29GA_MB_V2.1 2020/04/07

The chip is QG2101A

Any help would be appreciated! :-)

roleoroleo, 2024-10-06:
I need a dump of the flash to check if this version is compatible,
Could you send me a copy?

graham30rad, 2024-10-07:
Thanks for the reply . . Yes, I will get it for you, I just need a few days as I am away at the moment, please bare with me and I will upload when home :-) 👍

kumdzio, 2025-01-14:
Hi @roleoroleo 
Did you receive the dump from @graham30rad?
If not and if you want it then i can upload it for you as i have exact same motherboard but with firmware version 9.0.19.02_202007141512

graham30rad, 2025-02-09:
Terribly terribly sorry for the long delay in getting back, I had something come up that demanded my attention and I lost track of this project. I will attach a copy of the firmware.bin file that I manage to get off the camera.

graham30rad, 2025-02-09:
Whoops, .bin is not allowed . . is there another way to share this? (I tried binwalking it but I'm not getting anything out) here is the first 8 lines of hexdump, if it helps to know if I'm in the right area?  hexdump -C firmware.bin | head -n 8
00000000  cc 00 00 ea 65 47 4f 4e  2e 42 54 30 4c 50 0e 44  |....eGON.BT0LP.D|
00000010  00 80 00 00 30 00 00 00  00 00 00 00 00 00 02 00  |....0...........|
00000020  00 00 02 00 00 00 00 00  00 00 00 00 34 2e 30 00  |............4.0.|
00000030  00 00 00 00 08 00 00 00  10 02 00 00 02 00 00 00  |................|
00000040  f9 0b 00 00 00 00 00 00  d2 00 00 00 00 00 00 00  |................|
00000050  73 0e 00 00 02 00 00 00  00 00 00 00 00 00 00 00  |s...............|
00000060  92 19 47 00 0c a1 31 01  41 70 05 00 96 78 78 b4  |..G...1.Ap...xx.|
00000070  00 00 00 00 48 48 48 48  48 00 00 00 1e 12 21 16  |....HHHHH.....!.|

graham30rad, 2025-02-09:
[image]

graham30rad, 2025-02-09:
[38]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[43]BOOT0 commit : 81c18ee
[46]board init start
[48]set pll start
[50]set pll end
[52][pmu]: bus read error
[54]board init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[61]DRAM CLK = 528 MHz
[63]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT off.
[71]DRAM SIZE =64 M
[78]DRAM simple test OK.
[80]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[88]spinor id is: ef 40 17, read cmd: 03
[92]Succeed in reading toc file head.
[96]The size of toc is 4c000.
[148]Entry_name        = optee
[152]Entry_name        = u-boot
[156]Entry_name        = dtb
[159]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年                                                                                              11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00007-g8b9203a (Jul 06 2020 - 18:40:27 +0800) Allwinner Technolog                                                                                             y

[00.204]DRAM:  64 MiB
[00.207]Relocation Offset is: f9f94000
[00.218]secure enable bit: 0
[00.221]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.227]gic: sec monitor mode
[00.230]flash init start
[00.232]workmode = 0,storage type = 3
SF: Detected w25q64cv with page size 256 Bytes, erase size 64 KiB, total 8 MiB
[00.248]sunxi flash init ok
[00.252]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.312]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.320]sprite_led_gpio start
[00.323](weak)update dtb dram start
[00.327]update dtb dram  end
[00.330]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
[... 199 lines cut ...]
dispatch.c(get_config-2538)[00:01:32.476]:def_x:4040,def_y:637
>>>>>>>>>>>>>#########is_sd_exit chg to: 10012<<<<<<<<<<<<<<<<<<<>>>>>>>>>>>>>is                                                                                             _sd_exit chg to: 0<<<<<<<<<<<<<<<<<<<
dispatch.c(choose_server-942)[00:01:32.476]:in choose_server, region_id = 17, ap                                                                                             i_server = https://api.us.xiaoyi.com, log_server = http://log.us.xiaoyi.com, sna                                                                                             me = familymonitor-y28ga, dlproto = mius

dispatch.c(choose_server-942)[00:01:32.476]:in choose_server, region_id = 16, ap                                                                                             i_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sna                                                                                             me = familymonitor-y28ga, dlproto = mieu

system_cmd_withret_timeout:195, exited, status=0

dispatch.c(main-6394)[00:01:32.499]:init ok, cost time(2567 ms)

ptz_ctl.c(p_ptz-2482)[00:01:32.501]:ptz_ctl.c
g_dispatch_info.mmap_info->hw_type =1
now ptz_info----x:0,y:0

ptz_ctl.c(ptz_stop-162)[00:01:32.503]:PTZ stop OK

~~~~~~~dir = 3, speed = 10, run time = 80800~~~~~~

-15

rmm.c(init_font-931)[00:01:33.129]:font:/home/base/iso8859-1-24x48.bin

rmm.c(init_font-943)[00:01:33.129]:filesize=36864

rmm.c(msg_proc-2142)[00:01:33.134]:pid[648]

graham30rad, 2025-02-09:
[38]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[43]BOOT0 commit : 81c18ee
[46]board init start
[48]set pll start
[50]set pll end
[52][pmu]: bus read error
[54]board init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[61]DRAM CLK = 528 MHz
[63]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT off.
[71]DRAM SIZE =64 M
[78]DRAM simple test OK.
[80]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[88]spinor id is: ef 40 17, read cmd: 03
[92]Succeed in reading toc file head.
[96]The size of toc is 4c000.
[148]Entry_name        = optee
[152]Entry_name        = u-boot
[156]Entry_name        = dtb
[159]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年                                                                                              11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00007-g8b9203a (Jul 06 2020 - 18:40:27 +0800) Allwinner Technolog                                                                                             y

[00.204]DRAM:  64 MiB
[00.207]Relocation Offset is: f9f94000
[00.218]secure enable bit: 0
[00.221]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.227]gic: sec monitor mode
[00.230]flash init start
[00.232]workmode = 0,storage type = 3
SF: Detected w25q64cv with page size 256 Bytes, erase size 64 KiB, total 8 MiB
[00.248]sunxi flash init ok
[00.252]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.312]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.320]sprite_led_gpio start
[00.323](weak)update dtb dram start
[00.327]update dtb dram  end
[00.330]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
[... 199 lines cut ...]
dispatch.c(get_config-2538)[00:01:32.476]:def_x:4040,def_y:637
>>>>>>>>>>>>>#########is_sd_exit chg to: 10012<<<<<<<<<<<<<<<<<<<>>>>>>>>>>>>>is                                                                                             _sd_exit chg to: 0<<<<<<<<<<<<<<<<<<<
dispatch.c(choose_server-942)[00:01:32.476]:in choose_server, region_id = 17, ap                                                                                             i_server = https://api.us.xiaoyi.com, log_server = http://log.us.xiaoyi.com, sna                                                                                             me = familymonitor-y28ga, dlproto = mius

dispatch.c(choose_server-942)[00:01:32.476]:in choose_server, region_id = 16, ap                                                                                             i_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sna                                                                                             me = familymonitor-y28ga, dlproto = mieu

system_cmd_withret_timeout:195, exited, status=0

dispatch.c(main-6394)[00:01:32.499]:init ok, cost time(2567 ms)

ptz_ctl.c(p_ptz-2482)[00:01:32.501]:ptz_ctl.c
g_dispatch_info.mmap_info->hw_type =1
now ptz_info----x:0,y:0

ptz_ctl.c(ptz_stop-162)[00:01:32.503]:PTZ stop OK

~~~~~~~dir = 3, speed = 10, run time = 80800~~~~~~

-15

rmm.c(init_font-931)[00:01:33.129]:font:/home/base/iso8859-1-24x48.bin

rmm.c(init_font-943)[00:01:33.129]:filesize=36864

rmm.c(msg_proc-2142)[00:01:33.134]:pid[648]

graham30rad, 2025-02-09:
I noticed in the output that it's looking for a y28ga and not the y29 thats printed on the board, could this be the issue?

roleoroleo, 2025-02-16:
Check the first lines of /home/base/tools/extpkg.sh to get the model.
And the file /home/homever to get the version.

graham30rad, 2025-02-16:
[firmware.zip](https://github.com/user-attachments/files/18815549/firmware.zip)

I didn't know about zipping the file here, I learn something new everyday lol I hope that file works for you as I cannot get it to output 
 anything in binwalk, and I'm guessing thats where the files that you mentioned above are located?

roleoroleo, 2025-02-22:
I think this image is not valid.
The size ix 0x400000 and it should be 0x800000.
And I'm unable to split it in partitions.
Did you create it from the shell?
Try to dump one partition at a time, from 0 to 7:
`dd if=/dev/mtdblock0 of=/tmp/sd/mtdblock0.bin`

graham30rad, 2025-02-22:
Sorry, I must have fat fingered the query when typing it, I managed to get the full 8mb bin, I also checked the model and version as you requested: 

head -n 10 ./squashfs-root-0/base/tools/extpkg.sh
echo "------------------------------------home base tools extpkg.sh enter------------------------------------"
SUFFIX=y28ga
cd /tmp/update

cp /home/base/tools/rsa_pub_dec /tmp/update

dd if=home_${SUFFIX}m of=verout bs=23 count=1
dd if=home_${SUFFIX}m of=home1 bs=23 skip=1 count=999999999999999
rm home_${SUFFIX}m
dd if=home1 of=enc_key bs=1024 count=1

and:

cat ./squashfs-root-0/homever
12.1.20.1_202302201734

The model and version seem to line up with what I initially posted, it's the version that seems to be newer?

If you still need the .bin I can send it. 

I really appreciate your time and help.

Thanks.

roleoroleo, 2025-02-24:
It's a y28ga but with the new 12.x firmware.
It's new for me, now this hack supports only 9.x.
I can try to support it if you share your backup firmware.

graham30rad, 2025-02-25:
Ah great, thanks! I factory reset the camera so my wifi credentials are not present, here is the complete firmware.bin and also the seperate mtd's you requested, to save you a few minutes :-) Will look forward to your reply! Thanks again! 

[firmware.zip](https://github.com/user-attachments/files/18964422/firmware.zip)

[mtd2.zip](https://github.com/user-attachments/files/18964402/mtd2.zip)

[mtd3.zip](https://github.com/user-attachments/files/18964403/mtd3.zip)

[mtd4.zip](https://github.com/user-attachments/files/18964405/mtd4.zip)

roleoroleo, 2025-02-28:
Try this beta version:

[y28ga_0.3.3.tgz](https://github.com/user-attachments/files/19026763/y28ga_0.3.3.tgz)

graham30rad, 2025-02-28:
I'll try it once I get home in a few hours and report back! thanks! stand by...

graham30rad, 2025-02-28:
Yep, all seems as it should! Perfect! Great job! Thanks a mill Roleoroleo! :-)

roleoroleo, 2025-03-01:
Good news.
I will add it to the readme.

closed 2025-06-30 by github-actions[bot]