roleoroleo/yi-hack-Allwinner-v2 #948: Several networks in wpa_supplicant

ivanfor, 2024-10-12:
Hi @roleoroleo.
Is it possible to add several networks to wpa_supplicant.conf file? I guess it's a temporary file created on boot.
I have several APs with the same SSID and I would like to establish priorities as the camera frequently connects to one with low signal quality.
Would it be possible?
Thanks in advance!

roleoroleo, 2024-10-14:
wpa_supplicant.conf is created by Yi processes.
I don't know how and when it's created.
Sorry...

ivanfor, 2024-10-14:
Do YI processes have some sort of log file that can help hint how it is created?
Anyway, I guess it could be changed with startup.sh script. I'll try to investigate a bit, do some tests and share the results back here.

floronxt, 2024-10-17:
Maybe a secondary "/Factory/wpa_supplicant.conf" could help?

ivanfor, 2024-10-18:
Hi. 
I have finally done it by using the following script (done quickly and probably with mistakes) in startup.sh.
Maybe it can help somebody. Seems to be working so far ....

#!/bin/sh
CFG="/tmp/wpa_supplicant.conf"
bssid1="xx:xx:xx:xx:xx:xx"
cat $CFG | grep update_config >/dev/null || echo "update_config=1" >> $CFG
wpa_pid=$(ps | grep wpa_supplicant | grep -v grep | awk '{print $1}')
[ -z $wpa_pid ] && echo "Error al obtener el PID" && exit 1
kill -HUP $wpa_pid
sleep 5

cat << EOF | wpa_cli
remove_network 1
remove_network 2
remove_network 3
set_network 0 bssid $bssid1
set_network 0 priority 2
add_network
set_network 1 ssid "WiOT"
set_network 1 key_mgmt WPA-PSK                                                                                                                              
set_network 1 psk "password"
set_network 1 scan_ssid 1
set_network 1 priority 0
enable_network 1
save_config
EOF

floronxt, 2024-10-19:
Brilliant, thank you so much for sharing! I've always wished for a function like this.

closed 2025-02-17 by github-actions[bot]