roleoroleo/yi-hack-Allwinner-v2 #951: Wifi deconnection ; time

gtrhayrug, 2024-10-17:
Hi, 
I open a new issue to don't "troll" the other.

I have a lot of wifi disconnection ; once per day approximativelly.
A simple power off, on and after it is okay.
Wifi strength is 97%... the router wifi is just at 50cm...
[image]

I've tried to change the RTSP server but i loose motioneye support...
I've changed the SD card ; new one.

I've tried to minimize the services : disable cloud, recording without cloud, no onvif, just ssh+no ftp+ntpd+httpd

[image]

Even if I have configure Europe/Paris ; the clock is 2h before the real hour... any idea ? (NTP is configured).

roleoroleo, 2024-10-19:
Does the cam reboot when it disconnect from the network?
Check the uptime.

gtrhayrug, 2024-10-19:
I've no idea because i can't access to the html page when the cam is disconnected.
I need to unplug/plug and after the camera is available. So the uptime will be at 0.

What about my other question about NTP false time ?
`Even if I have configure Europe/Paris ; the clock is 2h before the real hour... any idea ? (NTP is configured).`

roleoroleo, 2024-10-20:
Where do you read "2h before the real hour"?

gtrhayrug, 2024-10-20:
If it was 6:52pm now, the camera shows 4:52 even if I have selected the good timezone.

roleoroleo, 2024-10-22:
> If it was 6:52pm now, the camera shows 4:52 even if I have selected the good timezone.

Where?
Web gui, ssh session, osd...

gtrhayrug, 2024-10-24:
Some news. I've restarted the router, no deconnection issue since it.

About the hour, in the yi-hack web page :
[image]
even if the timezone is correct : 
[image]
and NTPd is active : 
[image]
same in the snapshot (osd + hack) : 
[image]
[image]

even if it is at the same time, really : 
[image]

roleoroleo, 2024-10-25:
Please check if the time is properly saved in system.conf
`cat /tmp/sd/yi-hack/etc/system.conf | grep TIMEZONE`

Is the camera protected by a firewall?

gtrhayrug, 2024-10-27:
Seems to be Ok :
[image]

Now it is 
[image]
date reports : 
[image]
Same in the webpage : 
[image]

roleoroleo, 2024-11-04:
The time is ok.
The problem is the timezone.
I'll check my y623.

roleoroleo, 2024-11-11:
My y623 is ok.
[image]

Please, try Europe/Rome and check if it works.

closed 2025-03-12 by github-actions[bot]