roleoroleo/yi-hack-Allwinner-v2 #954: Yi Outdoor 1080p IFUSY59 no longer connecting to wifi

SebLz, 2024-10-26:
Hello,
Used this camera with this hack for many years and it stopped connecting to wifi at some point. I tried to reset it, change sd card, and even followed the unbrick procedure with a backup I had and using the .bin from another twin camera.
Each time I boot, it says "waiting to connect" undefinately.
When I try to pair it with the Yi Home app, it always says "Wifi password error" then proceeds back to "waiting to connect". The wifi pw is obviously correct and another cam from the same pack connects to it.
Anything I could still try at this point ?

roleoroleo, 2024-11-04:
If the fw is original (I mean non hack) you need the app to pair the cam.
Or you can try to "force" the connection using the hack procedure, as described in the README (Install procedure step 4).
_If you want to set wifi credentials, rename the file Factory/configure_wifi.cfg.ori to Factory/configure_wifi.cfg and edit the file with your username and password._

roleoroleo, 2024-11-06:
I recently changed the procedure: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/8a971b991520b7cb7f25d6a3fbcd84b09a4e1ba2
but it's not released.
Before this change the wifi procedure started only when the cam was not hacked.
Try changing the files manually.

SebLz, 2024-11-08:
Thnks @roleoroleo, i'm currently abroad but will give it a try when i'm back

SebLz, 2024-11-22:
Today I've tried again using the new h30ga_0.3.3 version (with the corrected config.sh as it seems).
Now when the camera starts, it quickly ends up with the led blinking blue and nothing else happens (no connection to wifi, no voice, similar to [this issue](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/716)). I tried to reset the camera then try again with another SD card, but I get the same result.

weibell, 2024-12-28:
Hi @SebLz, I was experiencing the same issue with the y623 model and was able resolve it, here are the steps I took:

- Extract version 0.3.3 to the SD card and edit Factory/configure_wifi.cfg as described in the [install procedure, step 4](https://github.com/roleoroleo/yi-hack-Allwinner-v2?tab=readme-ov-file#install-procedure)
- When powering on the camera, observe that the blue light keeps on blinking, even when powering it on again
- Hold the reset button for a few seconds, the camera now says "waiting to connect" when powering on
- Follow the steps described in [Change WiFi credentials (deprecated)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)):
> Download the following archive [sdwifi.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15275311/sdwifi.tar.gz) and decompress it to a FAT32 formatted SD card in the folder "yi-hack"; don't clean the SD card, it must contain both the hack (already applied) and this decompressed archive.
> Edit the file configure_wifi.cfg according to your needs. Pay attention to the special chars:
> ...
> When you open the SD card you will find these new files in the yi-hack folder.
> / --- yi-hack --- configure_wifi.cfg
>               |-- configure_wifi.sh
>               |-- startup.sh
> Insert the SD into the cam and power on it.
> The cam will reboot once.
> Check the wifi connection.
> Delete the files listed above (configure_wifi.cfg, configure_wifi.sh and startup.sh).
- The camera now successfully connects to the network

SebLz, 2024-12-29:
> Hi @SebLz, I was experiencing the same issue with the y623 model and was able resolve it, here are the steps I took:
>
> * Extract version 0.3.3 to the SD card and edit Factory/configure_wifi.cfg as [...]

Hi @weibell,

Thanks for your help. I've followed exactly your step and at the last step, when I power the camera on just after having pasted and edited the files into the yi-hack folder, this is what happens:
- camera boots and orange light blinks for a moment while I hear the "waiting to connect" voice exactly once
- the orange light suddenly stops blinking and stays orange still
- the camera reboots
- the camera goes back into a state where the blue light blinks undefinately.

During the whole process, no connection was made to the router although it is well in range and reachable.

closed 2025-04-29 by github-actions[bot]