roleoroleo/yi-hack-Allwinner-v2 #960: Home 2k Pro

InvisibleRasta, 2024-11-09:
Hello, will this work for the Yi camara 2K Pro serial YY6230

roleoroleo, 2024-11-11:
I need to know the Yi fw version.

jorge-msp, 2024-11-14:
The hack worked with my Yi camera 2K Pro - YFUSY94D5X.
Before the hack, the camera had the latest firmware version, 12.0.51.08.

Overall, it is working fine but I'm experiencing some delays when connected with the Home assistant.

roleoroleo, 2024-11-15:
> Overall, it is working fine but I'm experiencing some delays when connected with the Home assistant.

Try to install an integration that uses go2rtc. It should decrease the delay.

luiplapin1, 2024-11-19:
> The hack worked with my Yi camera 2K Pro - YFUSY94D5X. Before the hack, the camera had the latest firmware version, 12.0.51.08.
> 
> Overall, it is working fine but I'm experiencing some delays when connected with the Home assistant.

Hi! How do you do this work in that model? I'm not getting the cam hacked! :(

closed 2025-03-20 by github-actions[bot]