roleoroleo/yi-hack-Allwinner-v2 #965: Camera settings are not saved!

stefankoehne, 2024-11-26:
Hello, when I make settings on the camera_settings page in the web interface, the changes are not applied after saving. Reloading the page shows the old settings.
I tried turning off the status and IR LEDs, but no success.

Model y21ga, Firmware 0.3.3

stefankoehne, 2024-11-26:
Update:
The only way for me to change the settings is to export the config file, change it and import it again. Then the settings are displayed correctly in the web interface.

roleoroleo, 2024-11-27:
Try to clean the cache of your browser.

stefankoehne, 2025-01-19:
Same problem with deleted cache and different browsers.

roleoroleo, 2025-01-21:
Maybe something is broken in the config file.
Please try "Reset to yi-hack default" and reconfigure the cam.

closed 2025-05-22 by github-actions[bot]