roleoroleo/yi-hack-Allwinner-v2 #974: YI DOME GUARD 2K - RTSP not working

sam-semillia, 2024-12-10:
Hi,
I can not get RTSP working on the following model:

YI DOME GUARD
Model: YRS.3521
RFUSY95E13

Tested FW:

- r30gb: Not starting at all.
- r35 rgb: Starting, WEB UI available. RTSP not working.

Any ideas or help would be much appreiated.

Thanks, BR

dberrocal-git, 2024-12-11:
This camera is an r37gb, i have recently provided a backup and two copies of fshare_frame_buf to attempt the integration.

https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/757

sam-semillia, 2024-12-12:
Many thanks, fingers crossed integration works.

vgisbert, 2025-02-24:
Yi Dome Guard YRS 12.1.47.05_202411081421 yi-hack-Allwinner-v2 0.3.3 r35gb port 554 is closed.

I have tried 6 versions, from 0.2.9 to 0.3.3 and none of them work. It seems that Yi has blocked port 554.

vgisbert, 2025-04-28:
Hello. Any news on this camera? I bought a batch of two, and they're here, waiting for help. Thank you very much in advance.

roleoroleo, 2025-05-05:
Your cam is a r37gb.
It's supported by a beta release:
[r37gb_0.3.3.tgz](https://github.com/user-attachments/files/20041667/r37gb_0.3.3.tgz)

vgisbert, 2025-05-05:
> Your cam is a r37gb. It's supported by a beta release: [r37gb_0.3.3.tgz](https://github.com/user-attachments/files/20041667/r37gb_0.3.3.tgz)

Thanks for the reply. I've provided my information in the following thread:

https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/1013#issuecomment-2852190445

kasskas, 2025-05-06:
Thanks for the support @roleoroleo, happy to have a nice cameras which I know wont home back to China. I am testing build which you provided yesterday, but it looks like camera is rebooting every minute or two. I know it is beta, so not sure if that is expected? If I can help you out with some logging, please let me know.

olanwe, 2025-05-08:
> Thanks for the support [@roleoroleo](https://github.com/roleoroleo), happy to have a nice cameras which I know wont home back to China. I am testing build which you provided yesterday, but it looks like camera is rebooting every minute or two. I know it is beta, so not sure if that is expected? If I can help you out with some logging, please let me know. Switching to go2rtc seems to have fixed it.

I had exactly the same problem with the new version here with the cam rebooting every minute or so. I reverted the cam to the version from december (replacing the imggrabber, leading to this version: [r37gb_0.3.3_mod.tar.gz](https://github.com/user-attachments/files/20099232/r37gb_0.3.3_mod.tar.gz)) and it worked again.

roleoroleo, 2025-05-10:
I will check it.

closed 2026-01-01 by github-actions[bot]