roleoroleo/yi-hack-Allwinner-v2 #976: Bricked r30gb - no backup but RS232 working

Reni2005, 2024-12-11:
Hello, during a storm my camera stopped working, probably due a not correct firmware upgrade. When I connect a RS232 to the camera I get this:

`
▒▒0▒▒▒[38]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[43]BOOT0 commit : 81c18ee
[46]board init start
[48]set pll start
[50]set pll end
[52][pmu]: bus read error
[54]board init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[61]DRAM CLK = 528 MHz
[63]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT off.
[71]DRAM SIZE =64 M
[78]DRAM simple test OK.
[80]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[88]spinor id is: ef 40 17, read cmd: 03
[92]Succeed in reading toc file head.
[96]The size of toc is 4c000.
[149]Entry_name        = optee
[152]Entry_name        = u-boot
[156]Entry_name        = dtb
[159]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年 11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00006-g8e30467 (Sep 25 2020 - 18:47:17 +0800) Allwinner Technology

[00.204]DRAM:  64 MiB
[00.207]Relocation Offset is: f9f94000
[00.218]secure enable bit: 0
[00.221]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.227]gic: sec monitor mode
[00.230]flash init start
[00.232]workmode = 0,storage type = 3
SF: Detected w25q64cv with page size 256 Bytes, erase size 64 KiB, total 8 MiB
[00.249]sunxi flash init ok
[00.252]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.313]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.320]sprite_led_gpio start
[00.323](weak)update dtb dram start
[00.328]update dtb dram  end
[00.330]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
set root to /dev/mtdblock2
[00.393]update part info
[00.395]update bootcmd
mmc driver ver uboot2018:2019-8-15 16:34:00
get mem for descripter OK !
[00.408]get sdc_wipe fail.
[00.410]get sdc0 sdc_erase fail.
[00.413]get sdc0 sdc_boot fail.
[00.416]get sdc0 sdc_odly_50M fail.
[00.419]get sdc0 sdc_sdly_50M fail.
[00.422]get sdc0 sdc_odly_50M_ddr fail.
[00.426]get sdc0 sdc_sdly_50M_ddr fail.
[00.429]get sdc0 sdc_freq fail.
[00.432]get sdc0 sdc_b0p fail.
[00.435]get card0_boot_para:sdc_ex_dly_used fail
[00.440]get card-pwr-gpios handler:1119307304
[00.444]get card0_boot_para:time_pwroff:200ms
[00.450]Using default timing para
[00.652]init mmc 0 clock and io
devnum 0, prv 43fbd410, bdesc 42b7476c
SUNXI SD/MMC: 0
[00.660]==================== work mode: 0 0, sample_mode:0
[00.665]=============== start mmc_init_boot...
[00.684]Card did not respond to voltage select!
[00.688]mmc_init: -95, time 28
[00.690]mmc_init: mmc init fail, err -95
card0 init failed
No sd-card insert,will normal boot
Hit any key to stop autoboot:  0
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
[01.713]partinfo: name boot, start 0x20, size 0xe80
start: 0x300, len: 0x40
start: 0x340, len: 0xe11
[02.162]android.hardware = sun8iw19p1
## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1875936 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
[02.229]Starting kernel ...
`

And doesn't continue. Can I try to flash a backup?

roleoroleo, 2024-12-13:
The kernel is probably broken.
You could try the unbrick procedure adding the kernel partition (mtdblock1).
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam

Insert a sd card and try again the boot, you should read a log where the cam tries to find some file inside the card.
I don't remember if the kernel partition can be restored by the bootloader. If not, you should create xiaoyi.onebin

Otherwise, considering that you already have the serial port connected, try the hard way.
But using mtdblock1.

Reni2005, 2024-12-13:
> The kernel is probably broken. You could try the unbrick procedure adding the kernel partition (mtdblock1). https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam
>
> Insert a sd card [...]

Thank you for the answer. Should I try something like this before?: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/270#issuecomment-937764772

Or in this kind of model I should see directly the kernel? 
The camera stopped working after a storm, however if I get a RS232 comunication should it be still recoverable? I mean it dosen't seem "braind3ad".

I'm gonna wait some days before "putting hands on it" because I need another person to keep the "tx" and "rx" cable pressed because there's no screw to open it completely and because of that I cannot solder.

One last question: where should I find a working backup?

roleoroleo, 2024-12-14:
> Thank you for the answer. Should I try something like this before?: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/270#issuecomment-937764772

You could but probably it's not necessary.

> One last question: where should I find a working backup?

When you apply the hack the 1st time, the hack procedure creates a full backup.
You should find it in the backup folder of the sd card. Otherwise, if you know the version of your current fw, I cand send you.
Available versions:
- 9.0.05.32_202004101552
- 9.0.05.33_202009251856
- 9.0.05.99_202011091624
- 12.1.05.1_202302091435

> So If I get a backup and make a file with the unbriking tool I can put the created file in the root of a FAT32 sd card and the camera should be able to read it and flash it?

For the other paritions it works, I never tried the kernel.
Otherwise the hard way: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam#restore-backup-partition

```
=> sunxi_card0_probe
=> fatload mmc 0 45000000 mtdblock1.bin
=> sf probe
=> sf erase 60000 1d0000
=> sf write 45000000 60000 1d0000
```

Where:
- 45000000 is the address of the location where you copy termporarily the file mtdblock1.bin.
- 60000 is the offset of the kernel from the flash base
- 1d0000 is the size of mtdblock1.bin

Reni2005, 2024-12-14:
The camera stopped working before February 9, 2023. I would greatly appreciate it if you could send me the "9.0.05.99_202011091624" backup. However, I am not entirely sure about this version, as it has been a long time since the camera stopped functioning. What might happen if the version is incorrect? Could it result in the kernel working but the boot failing because the "home" partition might have the wrong version?

roleoroleo, 2024-12-14:
If you don't know exactly the version you must flash mtd1, 2, 3, 4.
[archive.tar.gz](https://github.com/user-attachments/files/18136855/archive.tar.gz)

Reni2005, 2024-12-15:
> If you don't know exactly the version you must flash mtd1, 2, 3, 4. [archive.tar.gz](https://github.com/user-attachments/files/18136855/archive.tar.gz)

Thank you very much, in these days I'll try.

In short the steps I'm gonna follow if I understood correctly

```
1) Flash mtd4 via hard way
2) Reset 
3) Insert micro-SD with mtd2,3,4 processed by unbrick tool and read on serial what's happening
```

I think I'm going to do only mtblock4 the hard way because the wiring is not soldered and the hand goes dizzy 🥴. 
If the bootloader is okay the device shouldn't have any hardware damage that would prevent booting?  
Gvien that every procedure of flashing has a risk, the bootloader cannot be bricked by flashing these .bin files right?

roleoroleo, 2024-12-18:
> In short the steps I'm gonna follow if I understood correctly

Yes.

> If the bootloader is okay the device shouldn't have any hardware damage that would prevent booting?

Yes, if the bootloader is ok you can flash all other partitions.

> Gvien that every procedure of flashing has a risk, the bootloader cannot be bricked by flashing these .bin files right?

It depends: with the hard way, if you use wrong addresses, you can overwrite the bootloader.

Reni2005, 2024-12-18:
> Yes.
>
>
> Yes, if the bootloader is ok you can flash all other partitions.
>
>
> It depends: with the hard way, if you use wrong addresses, you can overwrite the bootloader.

Hello, I'm having difficulties connecting both tx and rx doing it the "hard way". How can I try to flash bootloader via soft way? The ./unbrick.sh factory makes only three files and (backup, rootfs and home) but not the kernel one. Can I try to make it manually?

roleoroleo, 2024-12-22:
The problem is: "bad data checksum" but I don't know why it happens.
Are you sure that the files are created without errors when you run build.sh?

> I was thinking, can I build a .sh script in the micro sd root to run it via RS232? in this way I'd sure to not miss characters while writing commands on an unstable channel

You can run a shell command only if kernel and rootfs are running properly.

Please try my files: 
[archive2.tar.gz](https://github.com/user-attachments/files/18222107/archive2.tar.gz)

Reni2005, 2024-12-23:
Hello, I tried both ways (soft and hard).

`U-Boot 2018.05-00006-g8e30467 (Sep 25 2020 - 18:47:17 +0800) Allwinner Technology

[00.295]DRAM:  64 MiB
[00.298]Relocation Offset is: f9f94000
[00.309]secure enable bit: 0
[00.312]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.318]gic: sec monitor mode
[00.321]flash init start
[00.323]workmode = 0,storage type = 3
SF: Detected w25q64cv with page size 256 Bytes, erase size 64 KiB, total 8 MiB
[00.339]sunxi flash init ok
[00.343]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.404]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.411]sprite_led_gpio start
[00.414](weak)update dtb dram start
[00.419]update dtb dram  end
[00.421]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
set root to /dev/mtdblock2
[00.484]update part info
[00.486]update bootcmd
mmc driver ver uboot2018:2019-8-15 16:34:00
get mem for descripter OK !
[00.499]get sdc_wipe fail.
[... 257 lines cut ...]
Hit any key to stop autoboot:  0
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
[02.696]partinfo: name boot, start 0x20, size 0xe80
start: 0x300, len: 0x40
start: 0x340, len: 0xe12
[03.173]android.hardware = sun8iw19p1
## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1876616 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
[03.240]Starting kernel ...

`

I don’t understand why I had to manually write "sunxi_card0_probe". However, rebooting didn’t resolve the issue; it’s still stuck at "Starting Kernel." The camera doesn’t even flash the partitions you sent me giving me the same error (except with the kernel where i get a "image[1] wrong type").

Could I have provided the wrong model? Here is a picture of the external board:
[image]

In case it’s helpful, here is a picture of my SD card's root folder:
[image]

roleoroleo, 2024-12-23:
I'm not sure about the kernel, is the only partition I never restored.
The model is ok, it looks for *_r30gb files.
I think the only way is the hard way.

Reni2005, 2024-12-23:
Did I filash mtd 1 correctly? It stil doesn't boot after that

roleoroleo, 2024-12-24:
Try to flash other partitions: rootfs, home and backup.

Reni2005, 2025-03-25:
Hello,
I'm currently busy with my academic studies, but in the coming days, I plan to take another shot at fixing the camera. If I manage to get it working, there's still hope!

Just wanted to keep this issue active. Thanks!

closed 2025-07-24 by github-actions[bot]

Reni2005, 2025-07-31:
> Yes, it could be an hardware issue that crashes the kernel when it starts. Some hardware device that doesn't reply properly to the cpu.

Is there a way to check it via software or do I have to try a visual inspection? The camera doesn't seem get warm unless I write commands via RS232. AliExpress already provided me a replacement but I still wanted to try to "resuscitate" it

Reni2005, 2025-08-13:
> It depends on the issue. But not all problems can be checked.

The device was probably struck by a power surge, however the hardware seems intact. Might be a particular transistor that got burnt ? Can I buy just the board or it's gonna cost me more than the camera when I bought it?

roleoroleo, 2025-09-03:
I think the cost of the board is greater than the cost of the whole cam.