roleoroleo/yi-hack-Allwinner-v2 #977: PLAYBACK ISSUES WITH CHROME AND EDGE

vgisbert, 2024-12-12:
Hello.
I have 6 Yi cameras from various generations and models. In general, they work well. But in 2 of them I have problems with the display in Chrome and Edge browsers. But not always. Sometimes, for a short time, they randomly look good.
In the APP, in Firefox and VLC the images are always seen well.
The integration in Homeassistant is done through go2rtc and Webrtc.
The cameras are:
-	Yi Home Camera 1080P - RFUS
-	Yi Dome Camera U (Full HD) - SFUS
They have the same settings in the hack:
Links
Low res stream	rtsp://192.168.xx.xx/ch0_1.h264
Only audio stream	rtsp://192.168.xx.xx /ch0_2.h264
High res snapshot	http://192.168.xx.xx:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
Low res snapshot	http://192.168.xx.xx:8080/cgi-bin/snapshot.sh?res=low&watermark=yes
GitHub project	yi-hack-Allwinner-v2
System
Hostname	yi_hack9
Firmware Version	0.3.3
Base Version	9.0.19.12_202102241808
Model Suffix	y21ga
Serial Number	RFUSxxxxxxxxxx

And in go2rtc:
Streams:
        yi_hack9:
          - rtsp://user:pass@192.168.xx.xx/ch0_1.h264

I don't understand why it only happens with these browsers and why the image rarely looks good.
I attach the image of one that looks bad and one good at the same time.
Any idea?
[image]

roleoroleo, 2024-12-13:
No idea...
Maybe some incompatibility between the stream and the browser.
Did you try to change the rtsp server inside the cam?
For example setting go2rtc.

vgisbert, 2024-12-15:
> No idea... Maybe some incompatibility between the stream and the browser. Did you try to change the rtsp server inside the cam? For example setting go2rtc.

Yes, of course. I have tried three methods of integrating Yi cameras, and they all give me the same problem with the same cameras.

roleoroleo, 2024-12-18:
So the problem could be the codec used by Yi.
But I can't change it.

vgisbert, 2024-12-18:
> Please, try to change `RTSP_STI=yes` to `RTSP_STI=no` in /tmp/sd/yi-hack/etc/system.conf

It worked perfectly on one camera!!! Thank you very much!!!
I'm going to modify it on the cameras that give me the error. I'll let you know if it's fixed on all of them. You're very kind.

closed 2024-12-19 by vgisbert